matthewwall/weewx-sdr #1: error

atarist, 2016-07-05:
I get this eeror when i try to run sdr driver:
 weewxd /etc/weewx/weewx.conf
Traceback (most recent call last):
  File "/usr/bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/usr/share/weewx/weewx/engine.py", line 859, in main
    engine.run()
  File "/usr/share/weewx/weewx/engine.py", line 185, in run
    self.dispatchEvent(weewx.Event(weewx.NEW_LOOP_PACKET, packet=packet))
  File "/usr/share/weewx/weewx/engine.py", line 218, in dispatchEvent
    callback(event)
  File "/usr/share/weewx/weewx/engine.py", line 313, in new_loop_packet
    if event.packet['usUnits'] == self.target_unit: return
KeyError: 'usUnits'

matthewwall, 2016-07-05:
unfortunately the sdr driver is not working yet.  i started writing the code to parse the traffic my sdr was reporting (there is a lot!), then i had an sd card failure on the system running the sdr and lost it all before i could push it to the weewx-sdr repo.

the failure you see is what happens when the driver presents an empty packet to the weewx engine.

closed 2016-07-05 by matthewwall

atarist, 2016-07-05:
thanx for info, I hope you have time to finish it soon.