matthewwall/weewx-sdr #3: Oregon Science sensors

MartinSwanepoel, 2016-11-03:
Hi,

Anyone willing to add these 2 sensors to sdr.py? Unfortunately my knowledge is limited so I might cause more damage than good if I try.

Nov  3 06:36:26 Weather weewx[16727]: sdr: MainThread: lines=['2016-11-03 04:36:23 \t:\tOS :\tPCR800\n', '\tHouse Code:\t 93\n', '\tChannel:\t 0\n', '\tBattery:\t OK\n', '\tRain Rate:\t 0.0 in/hr\n', '\tTotal Rain:\t 41.0 in\n']
Nov  3 06:36:26 Weather weewx[16727]: sdr: MainThread: ts=1478147783 payload=OS :#011PCR800
Nov  3 06:36:26 Weather weewx[16727]: sdr: MainThread: unhandled message format: ts=1478147783 payload=OS :#011PCR800
Nov  3 06:36:29 Weather weewx[16727]: sdr: MainThread: lines=[]
Nov  3 06:36:32 Weather weewx[16727]: sdr: MainThread: lines=[]
Nov  3 06:36:37 Weather weewx[16727]: sdr: MainThread: lines=['2016-11-03 04:36:34 \t:\tOS :\tWGR800\n', '\tHouse Code:\t 85\n', '\tChannel:\t 0\n', '\tBattery:\t OK\n', '\tGust:\t 1.1 m/s\n', '\tAverage:\t 1.1 m/s\n', '\tDirection:\t 22.5 degrees\n']
Nov  3 06:36:37 Weather weewx[16727]: sdr: MainThread: ts=1478147794 payload=OS :#011WGR800
Nov  3 06:36:37 Weather weewx[16727]: sdr: MainThread: unhandled message format: ts=1478147794 payload=OS :#011WGR800

Thanks

matthewwall, 2016-11-03:
hi martin,

please try the sdr driver as of commit 1b76c83da4d2824ab53e5097299920f2b86a61d3 or later.  it should parse PCR800 and WGR800 sensors.

m

closed 2016-11-03 by matthewwall

MartinSwanepoel, 2016-11-03:
Hi Matthew,

Thanks that was quick !!

For some reason it seams as if the wind in not updated. I tried to capture
as much as possible. If you can perhaps point me where to look?

Wind N/A from N/A (N/A)

SDR.PY
out: ['2016-11-03 13:15:45 \t:\tOS :\tPCR800\n', '\tHouse Code:\t 93\n',
'\tChannel:\t 0\n', '\tBattery:\t OK\n', '\tRain Rate:\t 0.0 in/hr\n',
'\tTotal Rain:\t 41.0 in\n']
parsed: {'battery.0:93.OSPCR800Packet': 0,
'rain_total.0:93.OSPCR800Packet': 41.0, 'rain_rate.0:93.OSPCR800Packet':
0.0, 'usUnits': 1, 'dateTime': 1478178945}

out: ['2016-11-03 13:16:26 \t:\tOS :\tWGR800\n', '\tHouse Code:\t 85\n',
'\tChannel:\t 0\n', '\tBattery:\t OK\n', '\tGust:\t 1.4 m/s\n',
'\tAverage:\t 1.5 m/s\n', '\tDirection:\t 292.5 degrees\n']
parsed: {'dateTime': 1478178986, 'battery.0:85.OSWGR800Packet': 0,
'wind_gust.0:85.OSWGR800Packet': 1.4, 'wind_speed.0:85.OSWGR800Packet':
1.5, 'wind_dir.0:85.OSWGR800Packet': 292.5, 'usUnits': 17}

out: ['2016-11-03 13:18:54 \t:\tOS :\tTHGR810\n', '\tHouse Code:\t 176\n',
'\tChannel:\t 1\n', '\tBattery:\t OK\n', '\tCelcius:\t 27.20 C\n',
'\tFahrenheit:\t 80.96 F\n', '\tHumidity:\t 43 %\n']
parsed: {'temperature.1:176.OSTHGR810Packet': 27.2,
'humidity.1:176.OSTHGR810Packet': 43.0,
'temperature_F.1:176.OSTHGR810Packet': 80.96,
'battery.1:176.OSTHGR810Packet': 0, 'dateTime': 1478179134, 'usUnits': 16}

WEEWX
LOOP:   2016-11-03 16:18:16 SAST (1478182696) altimeter: None, appTemp:
None, barometer: None, cloudbase: None, dateTime: 1478182696, dewpoint:
None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None,
pressure: None, rainRate: 0, usUnits: 1, windBatteryStatus: 0, windchill:
None, windDir: 225.0, windGust: 2.23694185194
int: None, maxSolarRad: None, pressure: None, rainRate: 0, usUnits: 1,
windBatteryStatus: 0, windchill: None, windDir: 225.0, windGust:
2.23694185194

REC:    2016-11-03 16:20:00 SAST (1478182800) altimeter: None, appTemp:
None, barometer: None, cloudbase: 10439.4765972, dateTime: 1478182800.0,
dewpoint: 55.5482703627, ET: None, heatindex: 80.81, humidex:
85.9252847524, inDewpoint: None, interval: 5, maxSolarRad: None,
outBatteryStatus: 0.0, outHumidity: 42.0, outTemp: 80.81, pressure: None,
rain: 287.0, rainBatteryStatus: 0.0, rainRate: 0.0, usUnits: 1,
windBatteryStatus: 0.0, windchill: None, windrun: 0.0

SYSLOG
Nov  3 16:22:03 Weather weewx[15585]: sdr: MainThread: lines=['2016-11-03
14:22:00 \t:\tOS :\tWGR800\n', '\tHouse Code:\t 85\n', '\tChannel:\t 0\n',
'\tBattery:\t OK\n', '\tGust:\t 1.8 m/s\n', '\tAverage:\t 1.6 m/s\n',
'\tDirection:\t 270.0 degrees\n']
Nov  3 16:22:03 Weather weewx[15585]: sdr: MainThread: ts=1478182920
payload=OS :#011WGR800
Nov  3 16:22:03 Weather weewx[15585]: sdr: MainThread: pkt={'dateTime':
1478182920, 'battery.0:85.OSWGR800Packet': 0,
'wind_gust.0:85.OSWGR800Packet': 1.8, 'wind_speed.0:85.OSWGR800Packet':
1.6, 'wind_dir.0:85.OSWGR800Packet': 270.0, 'usUnits': 17}
Nov  3 16:22:03 Weather weewx[15585]: sdr: MainThread: packet={'windDir':
270.0, 'windBatteryStatus': 0, 'dateTime': 1478182920, 'usUnits': 17,
'windGust': 1.8}

WEEWX.CONF
[SDR]
    # This section is for the software-defined radio driver.

    # The driver to use
    driver = user.sdr
    [[sensor_map]]
        outBatteryStatus = battery.1:176.OSTHGR810Packet
        outHumidity = humidity.1:176.OSTHGR810Packet
        outTemp = temperature.1:176.OSTHGR810Packet
        windBatteryStatus = battery.0:85.OSWGR800Packet
        windDir = wind_dir.0:85.OSWGR800Packet
        windGust = wind_gust.0:85.OSWGR800Packet
        rainBatteryStatus = battery.0:93.OSPCR800Packet
        rainRate = rain_rate.0:93.OSPCR800Packet
        rain = rain_total.0:93.OSPCR800Packet

        log_unknown_sensors = True
        log_unmapped_sensors = True

matthewwall, 2016-11-03:
hi martin,

you need a windSpeed mapping in the sensor_map.  for example:

`windSpeed = wind_speed.0:85.OSWGR800Packet`

also, the log_* directives must go before the [[sensor_map]], otherwise they are treated as sensor mappings.

m

MartinSwanepoel, 2016-11-04:
Thanks !!! Must be the age that's causing the eyesight to go south ;-)

Regards,
Martin