matthewwall/weewx-sdr #6: Parsing for Accurite 06044M temp/humidity sensors

thespitz, 2017-01-12:
Hello,

I recently purchased AcuRite 06044M and would be super appreciative if someone with the skills could write a parser :)

sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd "rtl_433 -q -U -G" --hide parsed,out,empty

# Channel C:
unparsed: ['2017-01-12 01:35:20 :\tAcurite tower sensor \t:\t5585 \t:\tC\n', '\tTemperature:\t 19.2 C\n', '\tHumidity:\t 40\n', '\tBattery:\t 0\n', '\t:\t68\n']

# Channel A:
unparsed: ['2017-01-12 01:35:22 :\tAcurite tower sensor \t:\t521 \t:\tA\n', '\tTemperature:\t 1.9 C\n', '\tHumidity:\t 61\n', '\tBattery:\t 0\n', '\t:\t68\n']

Thanks!

matthewwall, 2017-01-12:
which version of rtl_433 are you running?  please post the output from this:

rtl_433 --help

the output you posted is for tower sensors (already recognized by weewx-sdr), but the formatting is multi-line instead of single line (which would be why weewx-sdr is not recognizing them).

thespitz, 2017-01-12:
I complied from source maybe a week ago.

weewx@weewx:~ $ rtl_433 --help
rtl_433: invalid option -- '-'
rtl_433, an ISM band generic data receiver for RTL2832 based DVB-T receivers

Usage:  = Tuner options =
        [-d <RTL-SDR USB device index>] (default: 0)
        [-g <gain>] (default: 0 for auto)
        [-f <frequency>] [-f...] Receive frequency(s) (default: 433920000 Hz)
        [-p <ppm_error] Correct rtl-sdr tuner frequency offset error (default: 0)
        [-s <sample rate>] Set sample rate (default: 250000 Hz)
        [-S] Force sync output (default: async)
        = Demodulator options =
        [-R <device>] Enable only the specified device decoding protocol (can be used multiple times)
        [-G] Enable all device protocols, included those disabled by default
        [-l <level>] Change detection level used to determine pulses [0-16384] (0 = auto) (default: 0)
        [-z <value>] Override short value in data decoder
        [-x <value>] Override long value in data decoder
        [-n <value>] Specify number of samples to take (each sample is 2 bytes: 1 each of I & Q)
        = Analyze/Debug options =
        [-a] Analyze mode. Print a textual description of the signal. Disables decoding
        [-A] Pulse Analyzer. Enable pulse analyzis and decode attempt
        [-D] Print debug info on event (repeat for more info)
        [-q] Quiet mode, suppress non-data messages
        [-W] Overwrite mode, disable checks to prevent files from being overwritten
        = File I/O options =
        [-t] Test signal auto save. Use it together with analyze mode (-a -t). Creates one file per signal
                 Note: Saves raw I/Q samples (uint8 pcm, 2 channel). Preferred mode for generating test files
        [-r <filename>] Read data from input file instead of a receiver
        [-m <mode>] Data file mode for input / output file (default: 0)
                 0 = Raw I/Q samples (uint8, 2 channel)
                 1 = AM demodulated samples (int16 pcm, 1 channel)
                 2 = FM demodulated samples (int16) (experimental)
                 3 = Raw I/Q samples (cf32, 2 channel)
                 Note: If output file is specified, input will always be I/Q
        [-F] kv|json|csv Produce decoded output in given format. Not yet supported by all drivers.
        [-C] native|si|customary Convert units in decoded output.
        [-T] specify number of seconds to run
        [-U] Print timestamps in UTC (this may also be accomplished by invocation with TZ environment variable set).
        [<filename>] Save data stream to output file (a '-' dumps samples to stdout)

Supported device protocols:
    [01]* Silvercrest Remote Control
    [02]  Rubicson Temperature Sensor
    [03]  Prologue Temperature Sensor
    [04]  Waveman Switch Transmitter
    [05]* Steffen Switch Transmitter
    [06]* ELV EM 1000
    [07]* ELV WS 2000
    [08]  LaCrosse TX Temperature / Humidity Sensor
    [09]* Acurite 5n1 Weather Station
    [10]* Acurite 896 Rain Gauge
    [11]  Acurite 609TXC Temperature and Humidity Sensor
    [12]  Oregon Scientific Weather Sensor
    [13]* Mebus 433
    [14]* Intertechno 433
    [15]  KlikAanKlikUit Wireless Switch
    [16]  AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)
    [17]* Cardin S466-TX2
    [18]  Fine Offset Electronics, WH-2 Sensor
    [19]  Nexus Temperature & Humidity Sensor
    [20]  Ambient Weather Temperature Sensor
    [21]  Calibeur RF-104 Sensor
    [22]* X10 RF
    [23]* DSC Security Contact
    [24]* Brennenstuhl RCS 2044
    [25]* GT-WT-02 Sensor
    [26]  Danfoss CFR Thermostat
    [27]* Energy Count 3000 (868.3 MHz)
    [28]* Valeo Car Key
    [29]  Chuango Security Technology
    [30]  Generic Remote SC226x EV1527
    [31]  TFA-Twin-Plus-30.3049 and Ea2 BL999
    [32]  Fine Offset WH1080 Weather Station
    [33]  WT450
    [34]  LaCrosse WS-2310 Weather Station
    [35]  Esperanza EWS
    [36]* Efergy e2 classic
    [37]* Inovalley kw9015b rain and Temperature weather station
    [38]  Generic temperature sensor 1
    [39]* Acurite 592TXR Temp/Humidity, 5n1 Weather Station, 6045 Lightning
    [40]* Acurite 986 Refrigerator / Freezer Thermometer
    [41]  HIDEKI TS04 Temperature, Humidity, Wind and Rain Sensor
    [42]  Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor
    [43]  CurrentCost Current Sensor
    [44]  emonTx OpenEnergyMonitor
    [45]  HT680 Remote control
    [46]  S3318P Temperature & Humidity Sensor
    [47]  Akhan 100F14 remote keyless entry
    [48]  Quhwa
    [49]  OSv1 Temperature Sensor
    [50]  Proove
    [51]  Bresser Thermo-/Hygro-Sensor 3CH
    [52]  Springfield Temperature and Soil Moisture
    [53]  Oregon Scientific SL109H Remote Thermal Hygro Sensor
    [54]  Acurite 606TX Temperature Sensor
    [55]  TFA pool temperature sensor
    [56]  Kedsum Temperature & Humidity Sensor
    [57]  blyss DC5-UK-WH (433.92 MHz)
    [58]  Steelmate TPMS
    [59]  Schraeder TPMS
    [60]* LightwaveRF
    [61]  Elro DB286A Doorbell
    [62]  Efergy Optical
    [63]  Honda Car Key
    [64]* Template decoder
    [65]  Fine Offset Electronics, XC0400
    [66]  Radiohead ASK
    [67]  Kerui PIR Sensor
    [68]  Fine Offset WH1050 Weather Station
    [69]  Honeywell Door/Window Sensor
    [70]  Maverick ET-732/733 BBQ Sensor

* Disabled by default, use -R n or -G

matthewwall, 2017-01-12:
try a specific protocol instead of -G.  for example:

rtl_433 -q -U -R 11

then 39 then 54.  see which protocol is actually being used, and see if your sensors are picked up by a protocol that spits out readings on a single line.

also, could you look at the back/inside of one of your sensors and find a model number?

thespitz, 2017-01-12:
Protocol is 39.  The model number on the back of the sensors is 06044-TX.  FCC ID RNE06044TX

andylittle, 2017-01-12:
This is with a 592TXR sensor. 
###This is with the rtl_433 that I downloaded on Jan 11th. 
root@raspberrypi2-2:/home/weewx# rtl_433 -q -U -R 39
2017-01-12 02:55:10 :	Acurite tower sensor 	:	12391 	:	B
	Temperature:	 18.0 C
	Humidity:	 68
	Battery:	 0
	:	68
###This is with the rtl_433 that I downloaded on October 20th. 
root@chip1-2:~# rtl_433 -q -U -R 39 
2017-01-12 03:14:18 Acurite tower sensor 0x3067 Ch B: 18.2 C 64.8 F 68 % RH

matthewwall, 2017-01-12:
what happens if you specify json output?  i'd rather write a json parser than having to keep changing the plain text parser...

thespitz, 2017-01-12:
$ rtl_433 -q -U -R 39 -F json

{"time" : "2017-01-12 03:43:05", "model" : "Acurite tower sensor", "id" : 521, "channel" : "A", "temperature_C" : 0.800, "humidity" : 68, "battery" : 0, "status" : 68}
{"time" : "2017-01-12 03:43:11", "model" : "Acurite tower sensor", "id" : 5585, "channel" : "C", "temperature_C" : 21.100, "humidity" : 32, "battery" : 0, "status" : 68}

matthewwall, 2017-01-12:
cool.  so they actually implemented json.  i'll add that for the tower sensor so we can deprecate all the unstructured plain text rubbish.

matthewwall, 2017-01-12:
if you have lightning sensors or 5n1 or any other acurite hardware, please post json for that too.  it will be a few days before i can update rtl-433 on the hardware i have for testing acurite hardware.

thespitz, 2017-01-12:
Thanks, I was just about to say, I'm adding these sensors in addition to a 5n1 that I started with.

Neither protocols 09 nor 39 seem to format to JSON.  This is after installing today's snapshot of rtl_433:

 $ rtl_433 -q -U -R 09 -F json
wind speed: 0 kph, temp: 14.4° F, humidity: 95% RH
wind speed: 0 kph, wind direction: 90.0°, rain gauge: 0.00 in.

$ rtl_433 -q -U -R 39 -F json
2017-01-12 03:53:33 Acurite 5n1 sensor 0x0CD7 Ch B, Total rain fall since last reset: 4.97
2017-01-12 03:53:33 Acurite 5n1 sensor 0x0CD7 Ch B, Msg 31, Wind 0 kmph / 0.0 mph 337.5° NNW (4), rain gauge 0.00 in.
2017-01-12 03:53:51 Acurite 5n1 sensor 0x0CD7 Ch B, Msg 38, Wind 0 kmph / 0.0 mph, -9.8 C 14.4 F 96 % RH

matthewwall, 2017-01-12:
json support for tower sensors added at commit dc44bf3 (driver version 0.13)

matthewwall, 2017-01-12:
there is a pr for a bunch of changes to acurite.c in rtl_433.  i think that is intended to consolidate the various acurite decoders.

matthewwall, 2017-01-12:
the weewx-sdr driver *should* handle a mix of json and non-json output, as long as they don't keep changing the non-json structure.

thespitz, 2017-01-12:
Aaaand I'm in business!  Its working beautifully.  Thanks - I owe you a beer!

closed 2017-01-12 by thespitz