matthewwall/weewx-sdr #7: rtl_433 lingers on while restarting weewx

kathode, 2017-01-12:
Hi,

First of all thanks a lot for writing this driver! I'm currently using this nearly flawlessly on a RPi2 to process data with an SDR dongle coming off an Acurite 5-in-1 weather station. 

I'm experiencing problems with restarting weewx, where the rtl_433 process stays on, thus preventing the weewx-sdr driver from being properly relaunched. For example, I end up with entries like the following on the weewx log:

`
Jan 12 18:38:12 framboesa weewx[6866]: sdr: MainThread: err: ['usb_claim_interface error -6\n', 'Failed to open rtlsdr device #0.\n']
Jan 12 18:38:12 framboesa weewx[6866]: sdr: MainThread: shutdown process /usr/local/bin/rtl_433 -q -U -R 9 -R 39 -R 12
Jan 12 18:38:12 framboesa weewx[6866]: engine: Caught WeeWxIOError: rtl_433 process is not running
Jan 12 18:38:12 framboesa weewx[6866]:     ****  Waiting 60 seconds then retrying...
`

Instead, I usually have to stop weewx before starting it again. This way the driver initializes rtl_433 properly with only one instance running. It seems like there is some race condition of sorts. It most recently happened when the weewx engine for some reason restarted itself.

Let me know if there are any other details which would be helpful to identify the problem.

Thanks

matthewwall, 2017-01-12:
could you explain a bit more about when this happens?

how would i duplicate the behavior you are seeing?

when i do this:

sudo /etc/init.d/weewx start
sudo /etc/init.d/weewx stop

everything works properly.  are you starting/stopping some other way?  faster?  slower?

do you see any log messages about "unable to join thread"?

kathode, 2017-01-12:
Indeed `/etc/init.d/weewx stop` followed by `/etc/init.d/weewx start` works fine. But if I do `/etc/init.d/weewx restart` instead, then I start seeing problems with rtl_433. As I mentioned earlier, the problem most recently happened a few hours ago when weewx decided to restart itself for some strange reason. Without further manual intervention (ie, stop followed by start) weewx-sdr couldn't recover for more than an hour.

I don't currently see any messages about "unable to join thread".

Thanks

matthewwall, 2017-01-12:
aha!  the bug is in `/etc/init.d/weewx restart`

the init script is starting weewx before the previous instance has actually stopped.

are you using an old init script for weewx?

or perhaps i need to do something in the driver to prevent the primary thread from returning before the rtl_433 process has finished shutting down...

kathode, 2017-01-12:
Well, I'm not sure how 'old' the init script is. I installed weewx about 1 week and a half ago using the .deb package, followed by your driver. 

I suspected the problem was to do with the way weewx is started, as waiting for rtl_433 to finish before restarting weewx would make it work every single time. Interestingly, when I do `/etc/init.d/weewx stop` it takes a while to return, which I assume was due to it waiting for rtl_433 to terminate, but as you rightly said there is still a problem. Weewx is unable to recover if it ever gets into that state, though.

Let me know if there is a fix, no rush :-)

matthewwall, 2017-01-12:
please try 0.14rc1 of weewx-sdr

i added explicit thread termination, and also extending debug logging while shutting down.  i suspect the explicit termination will solve the problem, but if not, set debug=1 then post the log from shutdown.

kathode, 2017-01-13:
Thanks for the quick fix! 👍 

I've installed the new version and it seems to be fixed. Restarting now works correctly.

Obviously I can't test the scenario where weewx's engine decides to randomly restart by itself, but suppose your new logic will handle that gracefully as well. If I do run into problems I'll open a new issue again.

Thanks

closed 2017-01-13 by matthewwall

kathode, 2017-01-25:
So after 13 days it finally hit a problem again. Following is an excerpt from the log.

`Jan 25 13:00:51 framboesa weewx[28261]: genimages: Generated 29 images for StandardReport in 22.72 seconds
Jan 25 13:00:51 framboesa weewx[28261]: reportengine: copied 0 files to /var/weewx/reports
Jan 25 13:00:51 framboesa weewx[28261]: cheetahgenerator: Generated 1 files for report cmon in 0.07 seconds
Jan 25 13:01:35 framboesa weewx[28261]: engine: Shutting down StdReport thread
Jan 25 13:01:55 framboesa weewx[28261]: engine: Unable to shut down StdReport thread
Jan 25 13:01:55 framboesa weewx[28261]: sdr: MainThread: shutdown process /usr/local/bin/rtl_433 -q -U -R 9 -R 39 -R 12
Jan 25 13:02:05 framboesa weewx[28261]: sdr: MainThread: timed out waiting for stdout-thread
Jan 25 13:02:15 framboesa weewx[28261]: sdr: MainThread: timed out waiting for stderr-thread
Jan 25 13:02:15 framboesa weewx[28261]: engine: Caught database OperationalError: database is locked
Jan 25 13:02:15 framboesa weewx[28261]:     ****  Waiting 2 minutes then retrying...
Jan 25 13:02:17 framboesa weewx[28261]: genimages: Generated 32 images for cmon in 208.17 seconds
Jan 25 13:03:07 framboesa weewx[28261]: genimages: Generated 32 images for cmon in 136.20 seconds
Jan 25 13:04:15 framboesa weewx[28261]: engine: retrying...
Jan 25 13:04:15 framboesa weewx[28261]: engine: Using configuration file /etc/weewx/weewx.conf
Jan 25 13:04:15 framboesa weewx[28261]: engine: Loading station type SDR (user.sdr)
Jan 25 13:04:15 framboesa weewx[28261]: sdr: MainThread: driver version is 0.14rc1
Jan 25 13:04:15 framboesa weewx[28261]: sdr: MainThread: sensor map is {'windSpeed': 'wind_speed.03BB.Acurite5n1Packet', 'windDir': 'wind_dir.03BB.Acurite5n1Packet', 'outHumidity': 'humidity.03BB.Acurite5n1Packet', 'outTemp': 'temperature.03BB.Acurite5n1Packet', 'rain_total': 'rain_total.03BB.Acurite5n1Packet', 'extraTemp1': 'temperature.2:82.OSTHGR122NPacket', 'extraHumid1': 'humidity.2:82.OSTHGR122NPacket', 'extraBatteryStatus1': 'battery.2:82.OSTHGR122NPacket'}
Jan 25 13:04:15 framboesa weewx[28261]: sdr: MainThread: deltas is {'strikes': 'strikes_total', 'rain': 'rain_total'}
Jan 25 13:04:15 framboesa weewx[28261]: sdr: MainThread: startup process '/usr/local/bin/rtl_433 -q -U -R 9 -R 39 -R 12'
Jan 25 13:04:15 framboesa weewx[28261]: engine: StdConvert target unit is 0x1
Jan 25 13:04:15 framboesa weewx[28261]: wxcalculate: The following values will be calculated: barometer=prefer_hardware, windchill=prefer_hardware, dewpoint=prefer_hardware, appTemp=prefer_hardware, rainRate=prefer_hardware, windrun=prefer_hardware, heatindex=prefer_hardware, maxSolarRad=prefer_hardware, humidex=prefer_hardware, pressure=prefer_hardware, inDewpoint=prefer_hardware, ET=prefer_hardware, altimeter=prefer_hardware, cloudbase=prefer_hardware
Jan 25 13:04:15 framboesa weewx[28261]: wxcalculate: The following algorithms will be used for calculations: altimeter=aaNOAA, maxSolarRad=RS
(...)
Jan 25 13:04:16 framboesa weewx[28261]: engine: Starting up weewx version 3.6.2
Jan 25 13:04:16 framboesa weewx[28261]: engine: Starting main packet loop.
Jan 25 13:04:16 framboesa weewx[28261]: sdr: MainThread: err: ['usb_claim_interface error -6\n', 'Failed to open rtlsdr device #0.\n']
Jan 25 13:04:16 framboesa weewx[28261]: sdr: MainThread: shutdown process /usr/local/bin/rtl_433 -q -U -R 9 -R 39 -R 12
Jan 25 13:04:16 framboesa weewx[28261]: engine: Caught WeeWxIOError: rtl_433 process is not running
Jan 25 13:04:16 framboesa weewx[28261]:     ****  Waiting 60 seconds then retrying...
Jan 25 13:05:16 framboesa weewx[28261]: engine: retrying...
Jan 25 13:05:16 framboesa weewx[28261]: engine: Using configuration file /etc/weewx/weewx.conf
Jan 25 13:05:16 framboesa weewx[28261]: engine: Loading station type SDR (user.sdr)
Jan 25 13:05:16 framboesa weewx[28261]: sdr: MainThread: driver version is 0.14rc1
Jan 25 13:05:16 framboesa weewx[28261]: sdr: MainThread: sensor map is {'windSpeed': 'wind_speed.03BB.Acurite5n1Packet', 'windDir': 'wind_dir.03BB.Acurite5n1Packet', 'outHumidity': 'humidity.03BB.Acurite5n1Packet', 'outTemp': 'temperature.03BB.Acurite5n1Packet', 'rain_total': 'rain_total.03BB.Acurite5n1Packet', 'extraTemp1': 'temperature.2:82.OSTHGR122NPacket', 'extraHumid1': 'humidity.2:82.OSTHGR122NPacket', 'extraBatteryStatus1': 'battery.2:82.OSTHGR122NPacket'}
Jan 25 13:05:16 framboesa weewx[28261]: sdr: MainThread: deltas is {'strikes': 'strikes_total', 'rain': 'rain_total'}
Jan 25 13:05:16 framboesa weewx[28261]: sdr: MainThread: startup process '/usr/local/bin/rtl_433 -q -U -R 9 -R 39 -R 12'
(...)
Jan 25 13:05:16 framboesa weewx[28261]: engine: Starting up weewx version 3.6.2
Jan 25 13:05:16 framboesa weewx[28261]: engine: Starting main packet loop.
Jan 25 13:05:16 framboesa weewx[28261]: sdr: MainThread: err: ['usb_claim_interface error -6\n', 'Failed to open rtlsdr device #0.\n']
Jan 25 13:05:16 framboesa weewx[28261]: sdr: MainThread: shutdown process /usr/local/bin/rtl_433 -q -U -R 9 -R 39 -R 12
Jan 25 13:05:16 framboesa weewx[28261]: engine: Caught WeeWxIOError: rtl_433 process is not running
Jan 25 13:05:16 framboesa weewx[28261]:     ****  Waiting 60 seconds then retrying...
`

It seems these are new:

```
Jan 25 13:02:05 framboesa weewx[28261]: sdr: MainThread: timed out waiting for stdout-thread
Jan 25 13:02:15 framboesa weewx[28261]: sdr: MainThread: timed out waiting for stderr-thread
```

It's now 16:30 local time and Wunderground finally told me the station is no longer reporting. Weewx-sdr could not recover.

The problem seems to be similar but not sure where exactly why.

matthewwall, 2017-01-27:
you might want to update to the latest sdr driver - it is now at v0.20

one of the things that changed along the way is that it goes right for the 'kill' instead of trying to 'terminate' first.  see the changelog for other changes (mostly just support for additional sensor types).

the 'database is locked' might be your root cause.  that happens if you have multiple threads/processes trying to write to the sqlite database at the same time.  perhaps you have a report that is taking a very long time to run?  (check the report timings in your logs)

another cause is when the i/o is really bad, such as when an sd card is about to die.

kathode, 2017-02-01:
Hi Matthew,

Thanks for the quick update and the accompanying suggestions. 

I've updated the sdr driver to version 0.21. I was not immediately aware, but in the [weewx documentation](http://weewx.com/docs/usersguide.htm#spikes) it is stated that SQlite on RPi's has this 'database is locked' problem. I have also finally moved all databases over to MySQL (with a replicated setup for backups).

I have not checked the timings of my logs yet. But I will keep an eye on the performance of weewx now that I have made those important changes.

It's good that you mention SD cards, but this is a fairly new installation of Jessie on a recently bought sdcard. I've burnt a few cards before with lots of writes, specially non-SanDisk cards!