matthewwall/weewx-sdr #8: New parser Rubicson temperature sensor

sbsrouteur, 2017-01-16:
Hi, 

I started using sdr on my pi over the week-end. I found a new hardware.

Below is the parser:

`class RubicsonTempPacket(Packet):
    # 2017-01-15 14:49:03 :   Rubicson Temperature Sensor
    #   House Code:      14
    #   Channel:         1
    #   Battery:         OK
    #   Temperature:     4.5 C
    #   CRC:     OK

    IDENTIFIER = "Rubicson Temperature Sensor"
    PARSEINFO = {
        'House Code': ['house_code', None, lambda x: int(x)],
        'Channel': ['channel', None, lambda x: int(x)],
        'Battery': ['battery', None, lambda x: 0 if x == 'OK' else 1],
        'Temperature': ['temperature', re.compile('([\d.]+) C'), lambda x: float(x)]}

    @staticmethod
    def parse_text(ts, payload, lines):
        pkt = dict()
        pkt['dateTime'] = ts
        pkt['usUnits'] = weewx.METRICWX
        pkt.update(Packet.parse_lines(lines, RubicsonTempPacket.PARSEINFO))
        channel = pkt.pop('channel', 0)
        code = pkt.pop('house_code', 0)
        sensor_id = "%s:%s" % (channel, code)
        pkt = Packet.add_identifiers(pkt, sensor_id, RubicsonTempPacket.__name__)
        return pkt
`

matthewwall, 2017-01-16:
thanks for that!  could you post a couple samples of json output for that sensor?

sbsrouteur, 2017-01-17:
Hi, 

not sure wha yo mean by json output. I have put in the class definition the rtl_433 output (but github does not seem to accept this as code for some obscure reason.
Below is the json ofun din syslog as output by weewx-sdr driver : 

`
Jan 17 07:30:39 raspberrypi weewx[20122]: sdr: MainThread: pkt={'temperature.1:14.RubicsonTempPacket': 3.2, 'battery.1:14.RubicsonTempPacket': 0, 'usUnits': 17, 'dateTime': 1484634637}
Jan 17 07:30:39 raspberrypi weewx[20122]: sdr: MainThread: packet={'dateTime': 1484634637, 'usUnits': 17, 'extraTemp1': 3.2}
`

matthewwall, 2017-01-17:
run rtl_433 with the option '-F json'

does rtl_433 emit any json output for the rubicson?

sbsrouteur, 2017-01-17:
here you go the rubiscon -F json output : 

`{"time" : "2017-01-17 20:47:41", "model" : "Rubicson Temperature Sensor", "id" : 14, "channel" : 1, "battery" : "OK", "temperature_C" : -1.800, "crc" : "OK"}
`

matthewwall, 2017-01-17:
json support added at ff3230b

closed 2017-01-17 by matthewwall

sbsrouteur, 2017-01-20:
My bad, I did not notice, but the parsing regex does not handel sign. 
This line : 
`'Temperature': ['temperature', re.compile('([\d.]+) C'), lambda x: float(x)]}`

must be fxed and look like : 
`'Temperature': ['temperature', re.compile('([\d.-]+) C'), lambda x: float(x)]}`

Sorry for this. I do not think it affects the json part, since it direclty comes from rtl_433.

matthewwall, 2017-01-20:
fixed at commit 2b74212