matthewwall/weewx-sdr #9: Unable to get weewx and weewx-sdr to talk to each other

Big80s, 2017-01-21:
Hello,

I'm having trouble with weewx running weewx-sdr so I can see weewx's output.
I am able to run sdr.py successfully, however, when I try running weewx with the proper sensor_map, I get errors.  Please see attached, and I thank you for any insight into what I'm doing wrong.

[sdr.txt](https://github.com/matthewwall/weewx-sdr/files/721382/sdr.txt)

matthewwall, 2017-01-21:
> pi@raspberrypi2:~/weewx/weewx-3.4.0 $ sudo ./bin/weewxd weewx.conf
> Traceback (most recent call last):
>   File "./bin/weewxd", line 64, in <module>
>     weewx.engine.main(options, args)
>   File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 842, in main
>     config_dict = getConfiguration(config_path)
>   File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 935, in getConfiguration
>     config_dict = configobj.ConfigObj(config_path, file_error=True)
>   File "/usr/lib/python2.7/dist-packages/configobj.py", line 1229, in __init__
>     self._load(infile, configspec)
>   File "/usr/lib/python2.7/dist-packages/configobj.py", line 1318, in _load
>     raise error
> configobj.ConfigObjError: Parsing failed with several errors.
> First error at line 87.

there is a problem with your config file at line 87

> pi@raspberrypi2:~/weewx/weewx-3.4.0 $ sudo ./bin/weewxd weewx.conf
> Traceback (most recent call last):
>   File "./bin/weewxd", line 64, in <module>
>     weewx.engine.main(options, args)
>   File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 855, in main
>     engine = EngineClass(config_dict)
>   File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 69, in __init__
>     self.setupStation(config_dict)
>   File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 92, in setupStation
>     __import__(driver)
> ImportError: No module named sdr

too hard to say without seeing the code.  it looks like you have modified sdr.py and perhaps other parts of weewx?

Big80s, 2017-01-21:
Yes, I did.  I will attach both the sdr.py and the weewx.conf to this string.  I should have done that first.
The weewx.conf is currently configured to run my AcuRite.  I have commented out the relevant Ambient Weather Sensors driver section (right below the Station, AcuRite section) so weewx can continue to collect AcuRite data.  
[weewx.conf.txt](https://github.com/matthewwall/weewx-sdr/files/721423/weewx.conf.txt)
[sdr_local.txt](https://github.com/matthewwall/weewx-sdr/files/721424/sdr_local.txt)

matthewwall, 2017-01-21:
another problem: you posted non-json output, but you requested json output (-F json) when you ran with weewx.

if you post json output from the sensor, i'll add that to sdr.py

matthewwall, 2017-01-21:
please try the latest sdr.py (0.20rc1 at commit 8d008d2)

i added parsing for the ambient th sensor.

also, please post the json output you see when you run rtl_433 directly with the -F json option.  then i can include json support for that sensor as well as the multi-line support.

Big80s, 2017-01-21:
Here's the json output:
{"time" : "2017-01-21 13:01:30", "model" : "Ambient Weather F007TH Thermo-Hygrometer", "device" : 80, "channel" : 1, "temperature_F" : 61.800, "humidity" : 10}

Big80s, 2017-01-21:
And, thanks for adding the parsing.

matthewwall, 2017-01-21:
json parsing included at v0.20 (commit d8fc41e)

please let me know whether that works, as i do not have that sensor for testing.

matthewwall, 2017-01-21:
i got F/C wrong for multi-line (but ok for json).

fixed at 3d1df1a

Big80s, 2017-01-21:
I assume that I just copy your new commit code to my sdr.py version, is that correct?

matthewwall, 2017-01-21:
yes - just overwrite your sdr.py with the latest from github

Big80s, 2017-01-21:
Will do.

matthewwall, 2017-01-21:
btw, you can only run the sdr driver OR the acurite driver, not both

matthewwall, 2017-01-21:
if you want to run both drivers at the same time, then you need one weewx.conf for acurite, and another weewx.conf for sdr.  if you do that, then each conf should have its own database and reports.

Big80s, 2017-01-21:
Thanks.  That said, it still doesn't work.  I made sure to comment out the AcuRite station stuff.
Here's the SDR section in the weewx.conf followed by the error messages:
[SDR]
    driver = user.sdr
    cmd = rtl_433 -q -U -F json -R 20
    path = /home/pi/rtl_433-master/bin
    ld_library_path = /opt/libusb-1.0.20/lib:/home/pi/rtl-sdr/lib
       [[sensor_map]]        
           inTemp = temperature.1:80.AmbientF007THPacket
    	   inHumidity = humidity.1:80.AmbientF007THPacket
#          inTemp = temperature.5:114.AmbientF007THPacket
#  	   inHumidity = humidity.5:114.AmbientF007THPacket
#  	   inTemp = temperature.3:56.AmbientF007THPacket
#  	   inHumidity = humidity.3:56.AmbientF007THPacket
pi@raspberrypi2:~/weewx/weewx-3.4.0 $ sudo ./bin/weewxd weewx.conf
Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 855, in main
    engine = EngineClass(config_dict)
  File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 69, in __init__
    self.setupStation(config_dict)
  File "/home/pi/weewx/weewx-3.4.0/bin/weewx/engine.py", line 92, in setupStation
    __import__(driver)
ImportError: No module named sdr

matthewwall, 2017-01-21:
what does the log say?

Big80s, 2017-01-21:
Here you go: 

[log.txt](https://github.com/matthewwall/weewx-sdr/files/721455/log.txt)

matthewwall, 2017-01-21:
what modifications have you made to weewx?

Big80s, 2017-01-21:
The only modifications I've made were to the weewx.conf.  If you are thinking of something(s) in particular, I can check.  I installed weewx early last year, and can't remember if I made any other changes to it at that time.

matthewwall, 2017-01-21:
it looks like your PYTHONPATH is messed up, probably in the shell in which you are invoking weewxd

Big80s, 2017-01-21:
This is what's in the sdr.py file now: PYTHONPATH='/usr/share/weewx'
Should it point to here instead: /home/pi/weewx/weewx-3.4.0?

matthewwall, 2017-01-21:
don't modify sdr.py

Big80s, 2017-01-21:
Ok, I'll take out that line, and see what happens.

matthewwall, 2017-01-21:
take out any changes you made, not just that one line

matthewwall, 2017-01-21:
to run it directly you just do this:

```
cd /home/pi/weewx/weewx-3.4.0
PYTHONPATH=bin python bin/user/sdr.py

```
assuming that weewx is installed in /home/pi/weewx/weewx-3.4.0

Big80s, 2017-01-21:
Yes, weewx is installed in that directory.  
Here's what I just got:

pi@raspberrypi2:~/weewx/weewx-3.4.0 $ PYTHONPATH=bin python bin/user/sdr.py
python: can't open file 'bin/user/sdr.py': [Errno 2] No such file or directory

matthewwall, 2017-01-21:
so where is sdr.py?  you need to put it in the user directory to use it with weewx.

Big80s, 2017-01-21:
It's currently here: /usr/share/weewx/user  I'll move it to the /weewx-3.4.0 directory and see if that works.

Big80s, 2017-01-21:
I moved it, and reran the command...and this is what I got:
pi@raspberrypi2:~/weewx/weewx-3.4.0 $ PYTHONPATH=bin python /home/pi/weewx/weewx-3.4.0/sdr.py
err: rtl_433: invalid option -- 'G'

err: rtl_433, an ISM band generic data receiver for RTL2832 based DVB-T receivers

err: 

err: Usage:	= Tuner options =

err: 	[-d <device index>] (default: 0)

err: 	[-g <gain>] (default: 0 for auto)

err: 	[-f <frequency>] [-f...] Receive frequency(s) (default: 433920000 Hz)

err: 	[-p <ppm_error] Correct rtl-sdr tuner frequency offset error (default: 0)

err: 	[-s <sample rate>] Set sample rate (default: 250000 Hz)

err: 	[-S] Force sync output (default: async)

err: 	= Demodulator options =

err: 	[-R <device>] Listen only for the specified remote device (can be used multiple times)

err: 	[-l <level>] Change detection level used to determine pulses [0-32767] (0 = auto) (default: 8000)

err: 	[-z <value>] Override short value in data decoder

err: 	[-x <value>] Override long value in data decoder

err: 	[-n <value>]  Specify number of samples to take (each sample is 2 bytes: 1 each of I & Q)

err: 	= Analyze/Debug options =

err: 	[-a] Analyze mode. Print a textual description of the signal. Disables decoding

err: 	[-A] Pulse Analyzer. Enable pulse analyzis and decode attempt

err: 	[-D] Print debug info on event (repeat for more info)

err: 	[-q] Quiet mode, suppress non-data messages

err: 	[-W] Overwrite mode, disable checks to prevent files from being overwritten

err: 	= File I/O options =

err: 	[-t] Test signal auto save. Use it together with analyze mode (-a -t). Creates one file per signal

err: 		 Note: Saves raw I/Q samples (uint8 pcm, 2 channel). Preferred mode for generating test files

err: 	[-r <filename>] Read data from input file instead of a receiver

err: 	[-m <mode>] Data file mode for input / output file (default: 0)

err: 		 0 = Raw I/Q samples (uint8, 2 channel)

err: 		 1 = AM demodulated samples (int16 pcm, 1 channel)

err: 		 2 = FM demodulated samples (int16) (experimental)

err: 		 3 = Raw I/Q samples (cf32, 2 channel)

err: 		 Note: If output file is specified, input will always be I/Q

err: 	[-F] kv|json|csv Produce decoded output in given format. Not yet supported by all drivers.

err: 	[-C] native|si|customary Convert units in decoded output.

err: 	[-T] specify number of seconds to run

err: 	[-U] Print timestamps in UTC (this may also be accomplished by invocation with TZ environment variable set).

err: 	[<filename>] Save data stream to output file (a '-' dumps samples to stdout)

err: 

err: Supported devices:

err: 	[01] Silvercrest Remote Control

err: 	[02] Rubicson Temperature Sensor

err: 	[03] Prologue Temperature Sensor

err: 	[04] Waveman Switch Transmitter

err: 	[05] Steffen Switch Transmitter

err: 	[06] ELV EM 1000

err: 	[07] ELV WS 2000

err: 	[08] LaCrosse TX Temperature / Humidity Sensor

err: 	[09] Acurite 5n1 Weather Station

err: 	[10] Acurite 896 Rain Gauge

err: 	[11] Acurite Temperature and Humidity Sensor

err: 	[12] Oregon Scientific Weather Sensor

err: 	[13] Mebus 433

err: 	[14] Intertechno 433

err: 	[15] KlikAanKlikUit Wireless Switch

err: 	[16] AlectoV1 Weather Sensor (Alecto WS3500 WS4500 Ventus W155/W044 Oregon)

err: 	[17] Cardin S466-TX2

err: 	[18] Fine Offset Electronics, WH-2 Sensor

err: 	[19] Nexus Temperature & Humidity Sensor

err: 	[20] Ambient Weather Temperature Sensor

err: 	[21] Calibeur RF-104 Sensor

err: 	[22] X10 RF

err: 	[23] DSC Security Contact

err: 	[24] Brennstuhl RCS 2044

err: 	[25] GT-WT-02 Sensor

err: 	[26] Danfoss CFR Thermostat

err: 	[27] Energy Count 3000 (868.3 MHz)

err: 	[28] Valeo Car Key

err: 	[29] Chuango Security Technology

err: 	[30] Generic Remote SC226x EV1527

err: 	[31] TFA-Twin-Plus-30.3049 and Ea2 BL999

err: 	[32] Fine Offset WH1080 Weather Station

err: 	[33] WT450

err: 	[34] LaCrosse WS-2310 Weather Station

err: 	[35] Esperanza EWS

err: 	[36] Efergy e2 classic

err: 	[37] Inovalley kw9015b rain and Temperature weather station

err: 	[38] Generic temperature sensor 1

err: 	[39] Acurite 592TXR Temperature/Humidity Sensor and 5n1 Weather Station

err: 	[40] Acurite 986 Refrigerator / Freezer Thermometer

err: 	[41] HIDEKI TS04 Temperature and Humidity Sensor

err: 	[42] Watchman Sonic / Apollo Ultrasonic / Beckett Rocket oil tank monitor

err: 	[43] CurrentCost Current Sensor

err: 	[44] emonTx OpenEnergyMonitor

err: 	[45] HT680 Remote control

err: 	[46] S3318P Temperature & Humidity Sensor

err: 	[47] Akhan 100F14 remote keyless entry

err: 	[48] Quhwa

err: 	[49] OSv1 Temperature Sensor

err: 	[50] Proove

err: 	[51] Bresser Thermo-/Hygro-Sensor 3CH

err: 	[52] Springfield Temperature and Soil Moisture

err: 	[53] Oregon Scientific SL109H Remote Thermal Hygro Sensor

err: 	[54] Acurite 606TX Temperature Sensor

err: 	[55] TFA pool temperature sensor

err: 	[56] Kedsum Temperature & Humidity Sensor

err: 	[57] blyss DC5-UK-WH (433.92 MHz)

err: 	[58] Steelmate TPMS

err: 	[59] Schraeder TPMS

err: 	[60] LightwaveRF

err: 	[61] Elro DB286A Doorbell

err: 	[62] Efergy Optical

err: 	[63] Honda Car Key

err: 	[64] Template decoder

err: 

pi@raspberrypi2:~/weewx/weewx-3.4.0 $

matthewwall, 2017-01-21:
do you run weewx from /usr/share/weewx, /home/weewx, /home/pi/weewx/weewx-3.4.0, or some other location?

which version of weewx is installed at /usr/share/weewx?

normally one would either install weewx from a .deb (resulting in /usr/share/weewx), or from tarball using setup.py (resulting in /home/weewx)

if you installed from deb and run from the deb install, then you have no need for /home/pi/weewx/weewx-3.4.0

the rtl_433 output looks like old rtl_433.  how and when did you install that?

Big80s, 2017-01-21:
I run weewx from here: pi@raspberrypi2:~/weewx/weewx-3.4.0 $ sudo ./bin/weewxd weewx.conf
This is the output from the /user/share/weewx directory: pi@raspberrypi2:/usr/share/weewx $ weewxd --version
3.4.0

And, this is how I installed rtl_433 (not sure how to find out what version this is):
cd rtl_433/
mkdir build
cd build
cmake ../
make
make install

Big80s, 2017-01-21:
Forgot to add: I used the tarball method using setup.py to install weewx.

matthewwall, 2017-01-21:
at some point you installed weewx from a .deb package.  you might want to remove that:

sudo dpkg --purge weewx

the install the init.d script so you don't have to start weewx manually every time.  (see the user guide for details)

as for rtl_433, do a 'git pull' then do the 'make' and 'make install' again.  whatever you have installed right now does not recognize the -G option.

Big80s, 2017-01-21:
Thanks for the help.  I actually have weewx start in the init.d already.  And, I've got .sh script running to my website to display data from weewx.  I will follow your advice re: purge weewx and git pull.

Big80s, 2017-01-21:
I got down to the last 'make install' command, and got this:
pi@raspberrypi2:~/weewx/weewx-3.4.0/rtl_433/build $ make install
[  1%] Built target data
[ 98%] Built target rtl_433
[100%] Built target data-test
Install the project...
-- Install configuration: "Release"
-- Installing: /usr/local/include/rtl_433.h
CMake Error at include/cmake_install.cmake:36 (file):
  file INSTALL cannot copy file
  "/home/pi/weewx/weewx-3.4.0/rtl_433/include/rtl_433.h" to
  "/usr/local/include/rtl_433.h".
Call Stack (most recent call first):
  cmake_install.cmake:37 (include)
Makefile:66: recipe for target 'install' failed
make: *** [install] Error 1
pi@raspberrypi2:~/weewx/weewx-3.4.0/rtl_433/build $

Big80s, 2017-01-22:
My bad, I forgot to sudo make install.  That worked.  And this line also works now, and collecting data:PYTHONPATH=bin python /home/pi/weewx/weewx-3.4.0/sdr.py

Big80s, 2017-01-22:
Success!!  Everything works now.  However, the public_html and the weewx.sdb aren't updating even though
the log says that the images are being generated and the database was updated.  I haven't changed any of the settings. When I don't use the user.sdr driver, and use the acurite driver instead, images in the public_html are current as well as the database.  I'm sure I'm missing something simple.

Big80s, 2017-01-23:
Sorry for that.  Any additional help regarding the images and database is appreciated.  I'll close this issue after any feedback you have.  Thanks so much for your help and for creating this program...huge help!

matthewwall, 2017-01-27:
hard to say without seeing the config and the log.

run weewx directly and watch the loop packets - is your sensor_map configured properly?

topher2880, 2019-02-17:
Sorry for re-opening this thread, but ive tried all the other options for getting this to work.

I have an output on both:
sudo rtl_433 -M utc -F json -G
sudo rtl_433 -G

as well as:
sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -M utc -F json -G"

However when I try to:
sudo wee_config --reconfigure

I get:
Traceback (most recent call last):
  File "/usr/bin/wee_config", line 121, in <module>
    main()
  File "/usr/bin/wee_config", line 115, in main
    config_mgr.run(args, options)
  File "/usr/share/weewx/weecfg/config.py", line 103, in run
    self.modify_config(config_dict, options)
  File "/usr/share/weewx/weecfg/config.py", line 119, in modify_config
    stn_info = self.get_stn_info(config_dict, options)
  File "/usr/share/weewx/weecfg/config.py", line 128, in get_stn_info
    stn_info = weecfg.get_station_info(config_dict)
  File "/usr/share/weewx/weecfg/__init__.py", line 1220, in get_station_info
    stn_info['driver'] = config_dict[stn_info['station_type']]['driver']
  File "/usr/lib/python2.7/dist-packages/configobj.py", line 554, in __getitem__
    val = dict.__getitem__(self, key)
KeyError: 'driver'

Looking through other forums, and this thread, ive tried numerous edits to try and prompt WeeWx into running, but all I get from **sudo tail -f /var/log/syslog**
Feb 17 19:59:04 localhost weewx[14798]:     ****    File "/usr/share/weewx/weewx/engine.py", line 884, in main
Feb 17 19:59:04 localhost weewx[14798]:     ****      engine = engine_class(config_dict)
Feb 17 19:59:04 localhost weewx[14798]:     ****    File "/usr/share/weewx/weewx/engine.py", line 72, in __init__
Feb 17 19:59:04 localhost weewx[14798]:     ****      self.setupStation(config_dict)
Feb 17 19:59:04 localhost weewx[14798]:     ****    File "/usr/share/weewx/weewx/engine.py", line 90, in setupStation
Feb 17 19:59:04 localhost weewx[14798]:     ****      driver = config_dict[stationType]['driver']
Feb 17 19:59:04 localhost weewx[14798]:     ****    File "/usr/lib/python2.7/dist-packages/configobj.py", line 554, in __getitem__
Feb 17 19:59:04 localhost weewx[14798]:     ****      val = dict.__getitem__(self, key)
Feb 17 19:59:04 localhost weewx[14798]:     ****  KeyError: 'driver'
Feb 17 19:59:04 localhost weewx[14798]:     ****  Exiting.

My SDR config reads (some edits have been removed, such as **PATH and LD_LIB_path**
[SDR]
    # This section is for the software-defined radio driver.

     [[sensor_map]]
        outTemp = temperature.*.FOWH65BPacket
        outHumidity = humidity.*.FOWH65BPacket
        extraTemp1 = temperature.*.FOWH2Packet
        extraTemp2 = temperature.*.FOWH2Packet
        windDir = wind_dir.*.FOWH65BPacket
        windSpeed = wind_speed.3.FOWH65BPacket
        windGust = wind_gust.3.FOWH65BPacket
        rain = rain_total.3.FOWH65BPacket
        UV = uv.3.FOWH65BPacket
        radiation = light.3.FOWH65BPacket
        outTempBatteryStatus = battery.3.FOWH65BPacket

    # The driver to use
        driver = user.sdr
        cmd = rtl_433 -q -U -F json -R 32

I hope this isnt too annoying and all.... sorry for re-opening, but I have really spent the entire day on Dr Google looking through forums and anything else that can help me.

Regards

matthewwall, 2019-02-17:
instead of re-opening, you should just create a new issue.  but even better, just post to the weewx user group so others can see and learn.

this is not actually a bug.

you just need to put the 'driver' and 'cmd' parameters *before* the 'sensor_map'

as you have it written, 'driver' and 'cmd' are inside the sensor map, so they are just being ignored