matthewwall/weewx-sdr #10: New OS UV800 parser

sbsrouteur, 2017-01-30:
Hi, 

I got today a UVN800 from oregon scientific.

Below the code to parse it, and sample data from the sensor : 

`
#code for the UVN800 Class
class OSUV800Packet(Packet):
    # 2017-01-30 22:00:12     :       OS :    UV800
    # House Code:      207
    # Channel:         1
    # Battery:         OK
    # UV Index:        0

    IDENTIFIER = "UV800"
    PARSEINFO = {
        'House Code': ['house_code', None, lambda x: int(x)],
        'Channel': ['channel', None, lambda x: int(x)],
        'Battery': ['battery', None, lambda x: 0 if x == 'OK' else 1],
        'UV Index':
            ['UVIndex', re.compile('([\d.]+)'), lambda x: float(x)]}

    @staticmethod
    def parse_text(ts, payload, lines):
        pkt = dict()
        pkt['dateTime'] = ts
        pkt['usUnits'] = weewx.METRICWX
        pkt.update(Packet.parse_lines(lines, OSUV800Packet.PARSEINFO))
        channel = pkt.pop('channel', 0)
        code = pkt.pop('house_code', 0)
        sensor_id = "%s:%s" % (channel, code)
        pkt = Packet.add_identifiers(pkt, sensor_id, OSWGR800Packet.__name__)
        return pkt

`

sbsrouteur, 2017-01-30:
Sample json data for the sensor : 
{"time" : "2017-01-30 22:19:40", "brand" : "OS", "model" : "UV800", "id" : 207, "channel" : 1, "battery" : "OK", "uv" : 0}

matthewwall, 2017-01-30:
thanks for that!  added at commit 534c693

i used the keyword uv_index rather than UVIndex

closed 2017-01-30 by matthewwall

sbsrouteur, 2017-01-31:
This is a small copy paste error in the code : 

`pkt = Packet.add_identifiers(pkt, sensor_id, OSWGR800Packet.__name__)`

should be 

`pkt = Packet.add_identifiers(pkt, sensor_id, OSUV800Packet.__name__)`