matthewwall/weewx-sdr #13: Version 0.21 seems to suffer from a mysterious memory leak

kathode, 2017-02-09:
Hi Mathew,

This is a preliminary report pending more information from the logs of the affected installation. It seems since a few days ago weewx is using insane amounts of memory. I had to restart weewx yesterday early morning and could see the RPi running out of both physical memory and swap space. 

I now noticed again that the weather station was no longer reporting. Quickly logged in and top produced the following top lines:

`top - 00:30:34 up 31 days,  7:13,  1 user,  load average: 4,61, 5,79, 6,13
Tasks: 123 total,   1 running, 122 sleeping,   0 stopped,   0 zombie
%Cpu(s): 32,1 us,  5,1 sy,  0,0 ni, 62,7 id,  0,0 wa,  0,0 hi,  0,1 si,  0,0 st
KiB Mem:    882772 total,   866692 used,    16080 free,    34164 buffers
KiB Swap:   102396 total,   102336 used,       60 free.   108372 cached Mem

  PID USER      PR  NI    VIRT    RES    SHR S  %CPU %MEM     TIME+ COMMAND                                                                                  
20098 root      20   0 1269928 555880   3424 S 116,1 63,0   1037:09 python                                                                                   
19906 mysql     20   0  364808  65220   4160 S  17,4  7,4 140:36.68 mysqld                                                                                   
20102 root      20   0   37312   9688    700 S   8,2  1,1  86:55.60 rtl_433    `

The only change compared to previous weeks is that I switched to MySQL as a backend and updated weewx-sdr.

I cannot confirm yet the exact source of the problem, but I will be updating this issue later tomorrow when I get time to read the log.

Thanks

kathode, 2017-02-09:
A first look seems to suggest that there are way too many MySQL connections, exceeding the server's capacity and that genimages is taking way too long to run (several hundred seconds in certain cases).

I'm adding an option to report_timing so as to lengthen the interval at which it compiles reports. Is this sensible?

matthewwall, 2017-02-09:
yes - report_timing is what you should use to make reports happen less frequently.

the log should indicate if a report is starting before a previous report has finished.

kathode, 2017-02-12:
I've decreased the period for generating reports (specifically cmon reports only run every 20 mins now), and it seems to run stable. I've also tweaked the MySQL server to use less memory, thus avoiding more swap being used and running close to the limit of the RPi2.

Many thanks again for the guidance! I shall be closing this issue for now.

closed 2017-02-12 by kathode