matthewwall/weewx-sdr #16: Driver user.sdr failed to load

manyhatz, 2017-05-17:
I am a noob for sure, but I am enjoying working through the issues and learning something new. 
When installing your driver I get the following error:

``Driver user.sdr failed to load: Non-ASCII character '\xc2' in file /usr/share/weewx/user/sdr.py on line 727, but no encoding declared; see http://python.org/dev/peps/pep-0263/ for details (sdr.py, line 727)``

As I researched the error it looked to me, could be wrong, that the encoding is expected in the diver it self and not something I can fix or did wrong.
Any help would be greatly appreciated. Not sure what other info I can give that might be a help. I am setting up a Raspberry Pi with weewx.

Thanks
Craig

andylittle, 2017-05-17:
Temperature: 23.9 ° C   delete the degree symbol from line 727
Andy

manyhatz, 2017-05-17:
Keep it simple is something I need to remember. 
Thank you
Craig

closed 2017-05-17 by manyhatz

matthewwall, 2017-05-18:
fixed at commit aca23f3

sorry about that