matthewwall/weewx-sdr #17: Parser for WH1080/WH3080 not working

waded, 2017-06-08:
Running 

`PYTHONPATH=bin python bin/user/sdr.py --cmd "rtl_433 -q -U -F json -R 32"`

with an Ambient Weather WS2080 in the area I'm seeing JSON output as follows:

```
out: ['{"time" : "2017-06-08 06:27:13", 
"model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", 
"msg_type" : 0, "id" : 38, "temperature_C" : 20.400, "humidity" : 66, 
"direction_str" : "N", "direction_deg" : "0", "speed" : 0.000, "gust" : 1.224, 
"rain" : 407.400, "battery" : "OK"}\n']
```
(and occasionally a second msg_type:1 that seems to be WWVB, but I focused on he above)

If I understand correctly it's intended weewx-sdr supports this device (class FOWHx080Packet) but it looks like the IDENTIFIER value is not an exact match (extra spaces), if not other parts of the mapping like id.

I'd be happy to fix support if you agree that's a starting point. This is the first day I've laid eyes on weewx, rtl-sdr, sdr_433, and weewx-sdr and don't want to take device support down the wrong path. I'd like to get this device working.

waded, 2017-06-16:
It appears @eyesnz perhaps diagnosed a similar issue?

matthewwall, 2017-08-04:
this is addressed in commit 7fe0e56

not a great solution, but hopefully robust enough until rtl_433 gets some proper versioning and/or better consistency in the packet identification and structure

closed 2017-08-04 by matthewwall