matthewwall/weewx-sdr #21: Parser for non-SDR broadcast packets

captain-coredump, 2017-08-13:
WeatherFlow is finally about to ship their crowd-funded PWS.  Their basic architecture is a wireless network hub device, with Bluetooth BLE links to one or more "Air" and "Sky" outdoor devices that each contain groups of sensors.  They have already published all kinds of cloud API's (yawn), and are promising to publish their BLE API in the near future.

TL;DR - They are also doing something that is interesting for easily capturing station data, namely broadcasting UDP packets (apparently as they arrive at their wireless network hub) onto the local network segment for any device to see.  No fake DNS and web servers, no pass-through network ports, just listen for the UDP packets and decode the JSON goodness!  I have simplified rtl_433 down to this:

#!/usr/bin/python

from socket import *
s=socket(AF_INET, SOCK_DGRAM)
s.bind(('255.255.255.255',50222))
while True:
   m=s.recvfrom(1024)
   print m[0]

This is what an output packet from an "Air" device should look like:

{"serial_number":"AR-00009876","type":"obs_air","obs":[[1502287861,1009.30,21.90,85,0,0,3.46,1]],"firmware_revision":12}

(All of their other status packets are pure JSON, but for some reason they chose to send out observations as a list in the "obs" field.  The format is completely documented in their API docs.)

Long story short, this shares so much in common with the SDR broadcast decoder that I'm wondering if this SDR station code should be renamed to a medium-agnostic broadcast decoder, and encourage development of external programs other than rtl_433 as input?

I'm not overly familiar with all of the available weewx station input types, but seeing the simplicity of grabbing broadcast packets off of the wire I think that I may actually start to send similar packets from my growing collection of DIY IoT sensors and use this same basic format.  Hence, thinking that the sensor-mapping feature in the SDR station code is way more flexible than my half-finished station driver with everything hard-coded.  If there's a more appropriate way of doing this, I'm all ears.....

matthewwall, 2017-08-15:
i'm leaning toward a generic weewx-wu driver that captures (sniff or listen) tcp/ip packets that are wu protocol (http GET).  that driver would work with acurite bridge, fine offset observer, and any other hardware that sends directly to weather underground.

there is already a working implementation in the weewx-interceptor driver.  we would push the generic tcp/ip components into the weewx 'driver' module, then distribute a simple weewx-wu driver.

the weewx-sdr driver would be much simpler if it only had to recognize the json output.  i'm happy to make that change sooner than later, and push the pressure for consistency and versioning down onto the rtl_433 development.  that could also facilitate the use of sdr dongle for capturing davis signals, not just 433Mhz signals.

but i still don't have a clear vision of the "one json-speaking driver to rule them all"

is it a weewx-json driver that accepts any json input?  then that driver has plugins/modules/api to feed it via tcp/ip (e.g., for the weatherflow broadcasts) or pipe (e.g., for rtl_433 or rtldavis)?

or do we just push the json-handling stuff into a simple class or two in the weewx drivers module.  then implement weewx-weatherflow driver that listens for udp broadcasts (using tcp/ip components in weewx.drivers) and handles them (using json components in weewx.drivers).  and we implement weewx-sdr that is just a simplification of the current weewx-sdr.

or something else?

matthewwall, 2017-08-15:
discussion moved to the weewx developer forum

closed 2017-08-15 by matthewwall