matthewwall/weewx-sdr #23: WT450 parsing issue

magazaki, 2017-09-17:
Hi
It seems like it's only parsing humidity for the WT450 sensor.
When running as a service I get "ignoring duplicate" packet for humidity

Sep 18 00:22:52 raspberrypi weewx[1395]: sdr: MainThread: lines=['{"time" : "2017-09-17 22:22:49", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 26.880, "humidity" : 46}\n', '{"time" : "2017-09-17 22:22:49", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 26.900, "humidity" : 46}\n', '{"time" : "2017-09-17 22:22:49", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 26.930, "humidity" : 46}\n']
Sep 18 00:22:52 raspberrypi rsyslogd-2007: action 'action 18' suspended, next retry is Mon Sep 18 00:23:52 2017 [try http://www.rsyslog.com/e/2007 ]
Sep 18 00:22:52 raspberrypi weewx[1395]: sdr: MainThread: packet={'outHumidity': 46.0, 'usUnits': 16, 'dateTime': 1505686969}
Sep 18 00:22:52 raspberrypi weewx[1395]: sdr: MainThread: ignoring duplicate packet {'outHumidity': 46.0, 'usUnits': 16, 'dateTime': 1505686969}
Sep 18 00:22:52 raspberrypi weewx[1395]: sdr: MainThread: ignoring duplicate packet {'outHumidity': 46.0, 'usUnits': 16, 'dateTime': 1505686969}

Humidity is parsed and database is updated with this but not temperature.

When running the command manually, it states that it will parse both, so this part seems to be correct:
 sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -q -U -F json -R 33"
out: ['{"time" : "2017-09-17 22:34:49", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.400, "humidity" : 45}\n', '{"time" : "2017-09-17 22:34:49", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.420, "humidity" : 45}\n', '{"time" : "2017-09-17 22:34:49", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.450, "humidity" : 45}\n']
parsed: {'battery.1:2.AcuriteWT450Packet': 0, 'dateTime': 1505687689, 'channel.1:2.AcuriteWT450Packet': 2, 'humidity.1:2.AcuriteWT450Packet': 45.0, 'usUnits': 16, 'sensor_id.1:2.AcuriteWT450Packet': 1, 'temperature.1:2.AcuriteWT450Packet': 25.4}
parsed: {'battery.1:2.AcuriteWT450Packet': 0, 'dateTime': 1505687689, 'channel.1:2.AcuriteWT450Packet': 2, 'humidity.1:2.AcuriteWT450Packet': 45.0, 'usUnits': 16, 'sensor_id.1:2.AcuriteWT450Packet': 1, 'temperature.1:2.AcuriteWT450Packet': 25.42}
parsed: {'battery.1:2.AcuriteWT450Packet': 0, 'dateTime': 1505687689, 'channel.1:2.AcuriteWT450Packet': 2, 'humidity.1:2.AcuriteWT450Packet': 45.0, 'usUnits': 16, 'sensor_id.1:2.AcuriteWT450Packet': 1, 'temperature.1:2.AcuriteWT450Packet': 25.45}

brg
Magnus

magazaki, 2017-09-18:
OK, I finally got it working now, the temperatures and humidity are being parsed correctly.
As I am only a noob, I did some trial and error and snipped some code from other sensors in your code.
I think what made it work was to get rid of the "underscores" in the parser for WT450.
I added the "PARSEINFO" part and adjusted the "sensorid"

class AcuriteWT450Packet(Packet):
    IDENTIFIER = "WT450 sensor"
    PARSEINFO = {
        'Sensorid': ['id', None, lambda x: int(x)],
        'Channel': ['channel', None, lambda x: int(x)],
        'Temperature': ['temperature', re.compile('([\d.-]+) C'), lambda x: float(x)],}

    # {"time" : "2017-09-14 20:24:43", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.090, "humidity" : 49}
    # {"time" : "2017-09-14 20:24:44", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.110, "humidity" : 49}
    # {"time" : "2017-09-14 20:24:44", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.120, "humidity" : 49}

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRIC
        pkt['Sensorid'] = Packet.get_int(obj, 'id')
        pkt['channel'] = Packet.get_int(obj, 'channel')
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        _id = "%s:%s" % (pkt['Sensorid'], pkt['channel'])
        return Packet.add_identifiers(pkt, _id, AcuriteWT450Packet.__name__)

I am unsure about the "_id" part, there might be some redundant code,  but the parser is able to differentiate the sensors.
Please let me know if this looks correct from a developers perspective too,
Thanks!

brg
Magnus

matthewwall, 2017-09-18:
the PARSEINFO is only for plain-text parsing, i.e., non-json output.

i do not understand why it would have *any* effect, since the WT450 code does not have a parse_text method that uses PARSEINFO.

the change from Sensorid to _id *should* make no difference.

try removing the PARSEINFO, and change 'Sensorid' to 'sid' in the parse_json method.

magazaki, 2017-09-18:
Thanks, you're correct, your suggested changes is all that is needed.
I was just taking a guess, as this is really not field at all.
However happy to have it working!

So the the parser now reads:

class AcuriteWT450Packet(Packet):
    IDENTIFIER = "WT450 sensor"

    # {"time" : "2017-09-14 20:24:43", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.090, "humidity" : 49}
    # {"time" : "2017-09-14 20:24:44", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.110, "humidity" : 49}
    # {"time" : "2017-09-14 20:24:44", "model" : "WT450 sensor", "id" : 1, "channel" : 2, "battery" : "OK", "temperature_C" : 25.120, "humidity" : 49}

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRIC
        pkt['sid'] = Packet.get_int(obj, 'id')
        pkt['channel'] = Packet.get_int(obj, 'channel')
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        _id = "%s:%s" % (pkt['sid'], pkt['channel'])
        return Packet.add_identifiers(pkt, _id, AcuriteWT450Packet.__name__)

Requesting commit update.
Thanks for all your help,
Best regards Magnus

matthewwall, 2017-09-18:
thank you for your patience and willingness to work through this.

i should have known better than to use 'id' as a dictionary key.

closed 2017-09-18 by matthewwall

magazaki, 2017-09-18:
Thanks the same! This is only a hobby project I just jumped into, but I was close to hitting a "system overload" and give up.
Really appreciate your great contributions and I don't blame your minor "id" flaw at all.
I guess the clue is to never give in and It's all in the details- all the way.
/Magnus