matthewwall/weewx-sdr #28: Fine Offset rain unit

millallo, 2018-05-16:
Hi,
I'm using a FineOffset WH3080 USB station to record values for a while but now it's broken and i switched to SDR. Now I'm having in the weewx db rain values in centimeters instead of millimeters:
This is the db query:
`select datetime(datetime,'unixepoch','localtime') dt, rain,rainRate from archive where dt>'2018-04-1 00:00' and rain>0 order by datetime;`

With USB station:
```
2018-04-10 07:48:01|0.0300000000000011|0.120000000000005
2018-04-11 22:47:51|0.029999999999994|0.119999999999976
2018-04-12 12:27:52|0.0300000000000011|0.120000000000005
2018-04-12 13:07:52|0.0300000000000011|0.120000000000005
2018-04-12 13:17:58|0.0300000000000011|0.240000000000009
```

With SDR:
```
2018-05-15 22:50:00|0.299999999999997|4.799999995104
2018-05-15 23:00:00|3.6|4.82608695159913
2018-05-15 23:10:00|2.7|23.0297872105522
2018-05-15 23:20:00|2.4|15.8869565055345
2018-05-15 23:30:00|3.9|16.1749999835015
2018-05-15 23:40:00|3.9|22.621276572671
2018-05-15 23:50:00|5.7|29.249999970165
2018-05-16 00:00:00|2.09999999999999|26.6347825815281
2018-05-16 03:20:00|0.600000000000009|1.17391304228089
2018-05-16 03:30:00|0.299999999999997|2.54693877291235
2018-05-16 03:40:00|0.299999999999997|1.42499999854649
2018-05-16 10:30:00|0.299999999999997|0.834782607844166
```

This is the weewx output:
```
LOOP:   2018-05-16 10:26:11 CEST (1526459171) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459171, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 96.847296, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:26:11 CEST (1526459171) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459171, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 96.847296, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:27:11 CEST (1526459231) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459231, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 97.248561, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:27:11 CEST (1526459231) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459231, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 97.248561, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:27:30 CEST (1526459250) altimeter: None, appTemp: 18.38178938, barometer: None, cloudbase: 791.20635095, dateTime: 1526459250, dewpoint: 13.394742377, heatindex: 19.7, humidex: 22.7168635756, inDewpoint: None, maxSolarRad: None, outBatteryStatus: 0, outHumidity: 67.0, outTemp: 19.7, pressure: None, rain: 0.0, rain_total: 66.6, rainRate: 1.2, usUnits: 16, windchill: 19.7, windDir: 180.0, windGust: 18.3600456336, windSpeed: 12.2400304224
LOOP:   2018-05-16 10:27:30 CEST (1526459250) altimeter: None, appTemp: 18.14378938, barometer: None, cloudbase: 791.20635095, dateTime: 1526459250, dewpoint: 13.394742377, heatindex: 19.7, humidex: 22.7168635756, inDewpoint: None, maxSolarRad: None, outBatteryStatus: 0, outHumidity: 67.0, outTemp: 19.7, pressure: None, rain: 0.0, rain_total: 66.6, rainRate: 1.2, usUnits: 16, windchill: 19.7, windDir: 180.0, windGust: 18.3600456336, windSpeed: 13.4640334647
LOOP:   2018-05-16 10:28:11 CEST (1526459291) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459291, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 98.651181, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:28:11 CEST (1526459291) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459291, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 98.651181, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:28:18 CEST (1526459298) altimeter: None, appTemp: 18.4573384752, barometer: None, cloudbase: 762.808509235, dateTime: 1526459298, dewpoint: 13.6224888303, heatindex: 19.7, humidex: 22.8465459062, inDewpoint: None, maxSolarRad: None, outBatteryStatus: 0, outHumidity: 68.0, outTemp: 19.7, pressure: None, rain: 0.0, rain_total: 66.6, rainRate: 1.2, usUnits: 16, windchill: 19.7, windDir: 90.0, windGust: 17.1360425914, windSpeed: 12.2400304224
LOOP:   2018-05-16 10:29:06 CEST (1526459346) altimeter: None, appTemp: 18.14378938, barometer: None, cloudbase: 791.20635095, dateTime: 1526459346, dewpoint: 13.394742377, heatindex: 19.7, humidex: 22.7168635756, inDewpoint: None, maxSolarRad: None, outBatteryStatus: 0, outHumidity: 67.0, outTemp: 19.7, pressure: None, rain: 0.0, rain_total: 66.6, rainRate: 1.2, usUnits: 16, windchill: 19.7, windDir: 135.0, windGust: 19.5840486759, windSpeed: 13.4640334647
LOOP:   2018-05-16 10:29:11 CEST (1526459351) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459351, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 100.411686, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:30:11 CEST (1526459411) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459411, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 100.557732, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:30:12 CEST (1526459412) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459412, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 100.557732, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:30:12 CEST (1526459412) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459412, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 100.557732, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
LOOP:   2018-05-16 10:31:11 CEST (1526459471) altimeter: None, appTemp: None, barometer: None, cloudbase: None, dateTime: 1526459471, dewpoint: None, heatindex: None, humidex: None, inDewpoint: None, maxSolarRad: None, pressure: None, radiation: 100.696548, rainRate: 1.2, usUnits: 16, uv: 5.0, windchill: None, windDir: None, windGustDir: None
REC:    2018-05-16 10:30:00 CEST (1526459400) altimeter: None, appTemp: 18.4517977758, barometer: None, boilerTemp: 17.1, cloudbase: 791.018840775, dateTime: 1526459400.0, dewpoint: 13.3587461807, ET: 0.00149506532808, heatindex: 19.6625, humidex: 22.6596625763, inDewpoint: None, interval: 10, maxSolarRad: None, outBatteryStatus: 0.0, outHumidity: 67.0, outTemp: 19.6625, pressure: None, radiation: 93.9994954, rain: 0.3, rain_total: 66.5625, rainRate: 0.834782607844, upsCharge: 100.0, upsLoad: 6.0, usUnits: 16, uv: 4.8, windchill: 19.6625, windDir: 152.193694786, windGust: 19.5840486759, windGustDir: None, windrun: 21.2319748416, windSpeed: 11.6280289013
LOOP:   2018-05-16 10:31:30 CEST (1526459490) altimeter: None, appTemp: 18.61978938, barometer: None, cloudbase: 791.20635095, dateTime: 1526459490, dewpoint: 13.394742377, heatindex: 19.7, humidex: 22.7168635756, inDewpoint: None, maxSolarRad: None, outBatteryStatus: 0, outHumidity: 67.0, outTemp: 19.7, pressure: None, rain: 0.0, rain_total: 66.6, rainRate: 1.2, usUnits: 16, windchill: 19.7, windDir: 135.0, windGust: 20.8080517181, windSpeed: 11.0160273802
LOOP:   2018-05-16 10:33:06 CEST (1526459586) altimeter: None, appTemp: 18.14378938, barometer: None, cloudbase: 791.20635095, dateTime: 1526459586, dewpoint: 13.394742377, heatindex: 19.7, humidex: 22.7168635756, inDewpoint: None, maxSolarRad: None, outBatteryStatus: 0, outHumidity: 67.0, outTemp: 19.7, pressure: None, rain: 0.0, rain_total: 66.6, rainRate: 1.2, usUnits: 16, windchill: 19.7, windDir: 180.0, windGust: 24.4800608448, windSpeed: 13.4640334647
```

At Line https://github.com/matthewwall/weewx-sdr/blob/1df6925033637e451278885186de4fa346a5da36/bin/user/sdr.py#L694
the rain total is supposed to be cm but i think it is in mm.

matthewwall, 2018-07-06:
fixed at commit ce5c677f1b314f3d601878b408f68c2a1a06d9ec

closed 2018-07-06 by matthewwall

millallo, 2018-08-03:
Hi,
this issue remains, rain_total still remains in cm but it is in mm.
In these log line, it rains 0.3 mm
```
Aug  3 19:04:52 servo weewx[17859]: sdr: MainThread: lines=['{"time" : "2018-08-03 17:04:48", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 38, "temperature_C" : 24.900, "humidity" : 86, "direction_str" : "NE", "direction_deg" : "45", "speed" : 4.896, "gust" : 7.344, "rain" : 37.800, "battery" : "LOW"}\n', '{"time" : "2018-08-03 17:04:49", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 38, "temperature_C" : 24.900, "humidity" : 86, "direction_str" : "NE", "direction_deg" : "45", "speed" : 4.896, "gust" : 7.344, "rain" : 37.800, "battery" : "LOW"}\n']
Aug  3 19:04:52 servo weewx[17859]: sdr: MainThread: packet={'outTemp': 24.9, 'outHumidity': 86.0, 'dateTime': 1533315888, 'windDir': 45.0, 'outBatteryStatus': 1, 'windSpeed': 4.896, 'windGust': 7.344, 'rain_total': 37.8, 'usUnits': 16}
Aug  3 19:04:52 servo weewx[17859]: sdr: MainThread: packet={'outTemp': 24.9, 'outHumidity': 86.0, 'dateTime': 1533315889, 'windDir': 45.0, 'outBatteryStatus': 1, 'windSpeed': 4.896, 'windGust': 7.344, 'rain_total': 37.8, 'usUnits': 16}
```
```
Aug  3 19:05:39 servo weewx[17859]: sdr: MainThread: lines=['{"time" : "2018-08-03 17:05:36", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 38, "temperature_C" : 24.900, "humidity" : 86, "direction_str" : "NNW", "direction_deg" : "338", "speed" : 4.896, "gust" : 7.344, "rain" : 38.100, "battery" : "LOW"}\n']
Aug  3 19:05:39 servo weewx[17859]: sdr: MainThread: packet={'outTemp': 24.9, 'outHumidity': 86.0, 'dateTime': 1533315936, 'windDir': 338.0, 'outBatteryStatus': 1, 'windSpeed': 4.896, 'windGust': 7.344, 'rain_total': 38.1, 'usUnits': 16}
```
```
Aug  3 19:06:28 servo weewx[17859]: sdr: MainThread: lines=['{"time" : "2018-08-03 17:06:24", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 38, "temperature_C" : 24.800, "humidity" : 86, "direction_str" : "NE", "direction_deg" : "45", "speed" : 3.672, "gust" : 4.896, "rain" : 38.100, "battery" : "LOW"}\n', '{"time" : "2018-08-03 17:06:25", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 38, "temperature_C" : 24.800, "humidity" : 86, "direction_str" : "NE", "direction_deg" : "45", "speed" : 3.672, "gust" : 4.896, "rain" : 38.100, "battery" : "LOW"}\n']
Aug  3 19:06:28 servo weewx[17859]: sdr: MainThread: packet={'outTemp': 24.8, 'outHumidity': 86.0, 'dateTime': 1533315984, 'windDir': 45.0, 'outBatteryStatus': 1, 'windSpeed': 3.672, 'windGust': 4.896, 'rain_total': 38.1, 'usUnits': 16}
Aug  3 19:06:28 servo weewx[17859]: sdr: MainThread: packet={'outTemp': 24.8, 'outHumidity': 86.0, 'dateTime': 1533315985, 'windDir': 45.0, 'outBatteryStatus': 1, 'windSpeed': 3.672, 'windGust': 4.896, 'rain_total': 38.1, 'usUnits': 16}
```