matthewwall/weewx-sdr #37: updates to rtl_433

andylittle, 2019-01-04:
5n1 parsing changed:

https://groups.google.com/d/msg/weewx-user/lwgSFCjMyEc/b1m6B1tkDgAJ

Command sysntax changed
root@KCAMEADO21:~# PYTHONPATH=/home/weewx/bin/ python sdr.py  --cmd="rtl_433 -q -U -F json -G" 
err: rtl_433 version 18.12-60-gccbc568 branch master at 201812311225
err: quiet option (-q) is default and deprecated. See -v to increase verbosity
err: UTC mode option (-U) is deprecated. Please use "-M utc"

This seems to work 
root@KCAMEADO21:~# PYTHONPATH=/home/weewx/bin/ python sdr.py  --cmd="rtl_433 -q -M utc -F json -G"

matthewwall, 2019-01-04:
hopefully fixed at commit cca58e8

any testing you could do would be appreciated

(i tried to accommodate old and new labels for acurite5n1)

closed 2019-01-04 by matthewwall

andylittle, 2019-01-04:
I can test in a couple days. Wind speed label changed to kph in rtl_433

matthewwall, 2019-01-04:
change to kph has been addressed at commit a072c97