matthewwall/weewx-sdr #38: Alecto v1 wind sensor not parsed

roberto1984, 2019-01-20:
Hi,

i have an alecto V1 wind sensor.
Since there is no parser for it i would like to implement it but if i launch the sdr.py i can see that the wind sensor is "tagged" as out and not unparsed.
What does it means that is tgged with "out"?is it possible to write a parser for it?
Thank you in advance!

matthewwall, 2019-01-20:
'out' is the output from rtl_433

if you post the 'out' output then i can easily write the parser.  or have a go at it yourself, if you prefer.

its best to use the 'out' from the json, if possible.

roberto1984, 2019-01-20:
Hi Mattewe,

thank you for your support.
the output is this belw

out: ['{"time" : "2019-01-20 11:14:00", "model" : "AlectoV1 Wind Sensor", "id" : 7, "channel" : 0, "battery" : "OK", "wind_speed" : 0.000, "wind_gust" : 0.000, "wind_direction" : 270, "mic" : "CHECKSUM"}\n', '{"time" : "2019-01-20 11:14:00", "model" : "Springfield Temperature & Moisture", "sid" : 224, "channel" : 3, "battery" : "OK", "transmit" : "MANUAL", "temperature_C" : -204.800, "moisture" : 0, "mic" : "CHECKSUM"}\n', '{"time" : "2019-01-20 11:14:00", "model" : "Springfield Temperature & Moisture", "sid" : 224, "channel" : 3, "battery" : "OK", "transmit" : "MANUAL", "temperature_C" : -204.800, "moisture" : 0, "mic" : "CHECKSUM"}\n', '{"time" : "2019-01-20 11:14:00", "model" : "Springfield Temperature & Moisture", "sid" : 224, "channel" : 3, "battery" : "OK", "transmit" : "MANUAL", "temperature_C" : -204.800, "moisture" : 0, "mic" : "CHECKSUM"}\n']

as you can see,everytime this packet is received, also a temperature packet is received, that is inside the wind sensor, so i don't know how to parse it correctly..

Any suggestion?
Thanks!

matthewwall, 2019-01-20:
implemented at commit 8eab34e

please let me know if that works

closed 2019-01-20 by matthewwall