matthewwall/weewx-sdr #44: Sensor Map confustion

phinnay, 2019-02-14:
I have weewx-sdr working with my Fine Offset WH65B. Everything looks good except for random crashes in weewx with "ZeroDivisionError" errors. 
example: 
```Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/home/weewx/bin/weewx/engine.py", line 890, in main
    engine.run()
  File "/home/weewx/bin/weewx/engine.py", line 191, in run
    self.dispatchEvent(weewx.Event(weewx.NEW_LOOP_PACKET, packet=packet))
  File "/home/weewx/bin/weewx/engine.py", line 224, in dispatchEvent
    callback(event)
  File "/home/weewx/bin/weewx/engine.py", line 540, in new_loop_packet
    self.accumulator = self._new_accumulator(event.packet['dateTime'])
  File "/home/weewx/bin/weewx/engine.py", line 650, in _new_accumulator
    self.archive_interval)
  File "/home/weewx/bin/weeutil/weeutil.py", line 220, in startOfInterval
    m = int(time_tt.tm_min // interval_m * interval_m)
ZeroDivisionError: integer division or modulo by zero
```
what stands out is the **"self._new_accumulator(event.packet['dateTime'])"** line. 

When I run the weewx-sdr driver directly, I see output from my sensor array as follows:
```
out: ['{"time" : "2019-02-14 01:10:14.653076", "protocol" : 78, "model" : "Fine Offset WH65B", "id" : 10, "temperature_C" : 1.500, "humidity" : 51, "wind_dir_deg" : 283, "wind_speed_ms" : 1.721, "gust_speed_ms" : 2.040, "rainfall_mm" : 407.416, "uv" : 2, "uvi" : 0, "light_lux" : 0.000, "battery" : "OK", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.983, "freq2" : 914.911, "rssi" : -12.775, "snr" : 14.586, "noise" : -27.362}]
parsed: {'rain_total.10.FOWH65BPacket': 407.416, 'uv.10.FOWH65BPacket': 2.0, 'wind_dir.10.FOWH65BPacket': 283.0, 'uv_index.10.FOWH65BPacket': 0.0, 'light.10.FOWH65BPacket': 0.0, 'battery.10.FOWH65BPacket': 0, 'wind_gust.10.FOWH65BPacket': 2.04, 'temperature.10.FOWH65BPacket': 1.5, 'dateTime': 1550106614, 'humidity.10.FOWH65BPacket': 51.0, 'usUnits': 17, 'wind_speed.10.FOWH65BPacket': 1.721}
```
my sensor map is: 
```
 [[sensor_map]]
        light = light.10.FOWH65BPacket
        uv = uv.10.FOWH65BPacket
        rain = rain_total.10.FOWH65BPacket
        windDir = wind_dir.10.FOWH65BPacket
        windSpeed = wind_speed.10.FOWH65BPacket
        windGust = wind_gust.10.FOWH65BPacket
        outHumidity = humidity.10.FOWH65BPacket
        outTemp = temperature.10.FOWH65BPacket
        txBatteryStatus = battery.10.FOWH65BPacket
```

There are a few items in the "parsed" stanza without sensor identifiers (usUnits, dateTime) I assume that since they do not have identifiers, I do not need to manually map them in the sensor_map? Did not see anything in the documentation to confirm or deny this, however the mention of dateTime in the error has me second guessing... 

Any ideas?

matthewwall, 2019-02-14:
> There are a few items in the "parsed" stanza without sensor identifiers (usUnits, dateTime) I assume that since they do not have identifiers, I do not need to manually map them in the sensor_map?

that is correct.

what archive interval did you specify in your weewx configuration?

phinnay, 2019-02-14:
my archive interval is set to "30"

matthewwall, 2019-02-14:
weewx supports archive interval of 60 or greater.  also, archive interval must be a multiple of 60, otherwise you will get unexpected behavior at midnight crossings.

300 (5 minutes) is a standard archive interval.  if you want to capture data at higher temporal resolution, then use weewx to feed your loop data into influx or some other timeseries database.

if you think this should be otherwise, please post to the weewx user group.

closed 2019-02-14 by matthewwall

phinnay, 2019-02-14:
thank you for the update, I set it to "60" and will look into your InfluxDB plugin if additional data is needed. 
After changing update_interval WeeWX ran for an hour or so before crashing again with the following error:

```
Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/home/weewx/bin/weewx/engine.py", line 890, in main
    engine.run()
  File "/home/weewx/bin/weewx/engine.py", line 188, in run
    for packet in self.console.genLoopPackets():
  File "/home/weewx/bin/user/sdr.py", line 2114, in genLoopPackets
    for packet in PacketFactory.create(lines):
  File "/home/weewx/bin/user/sdr.py", line 1985, in create
    pkt = PacketFactory.parse_json(lines)
  File "/home/weewx/bin/user/sdr.py", line 2001, in parse_json
    return parser.parse_json(obj)
  File "/home/weewx/bin/user/sdr.py", line 1143, in parse_json
    pkt['light'] = Packet.get_float(obj, 'light_lux') * 0.007893
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```
line 1143 in sdr.py states: 
```
        pkt['light'] = Packet.get_float(obj, 'light_lux') * 0.007893
``
nothing earth shattering there... the math adds up whenever I plug in the values, even if the packet read 0 it would still be a float so not sure why it's throwing an error.

matthewwall, 2019-02-14:
i fixed the lux-to-solarrad conversion by removing it.  the driver should not be doing those conversions, especially lux-to-solarrad, since there is no correct way to convert from lux to solar radiation.  commit cb40cea