matthewwall/weewx-sdr #45: light_lux unsupported operand type(s) for *: 'NoneType' and 'float'

phinnay, 2019-02-14:
upon further research this is unrelated to previous issue - I See @StephenR0 had the same problem in https://github.com/matthewwall/weewx-sdr/issues/34#issuecomment-449088536

 After changing update_interval WeeWX ran for an hour or so before crashing again with the following error:

```
Traceback (most recent call last):
  File "./bin/weewxd", line 64, in <module>
    weewx.engine.main(options, args)
  File "/home/weewx/bin/weewx/engine.py", line 890, in main
    engine.run()
  File "/home/weewx/bin/weewx/engine.py", line 188, in run
    for packet in self.console.genLoopPackets():
  File "/home/weewx/bin/user/sdr.py", line 2114, in genLoopPackets
    for packet in PacketFactory.create(lines):
  File "/home/weewx/bin/user/sdr.py", line 1985, in create
    pkt = PacketFactory.parse_json(lines)
  File "/home/weewx/bin/user/sdr.py", line 2001, in parse_json
    return parser.parse_json(obj)
  File "/home/weewx/bin/user/sdr.py", line 1143, in parse_json
    pkt['light'] = Packet.get_float(obj, 'light_lux') * 0.007893
TypeError: unsupported operand type(s) for *: 'NoneType' and 'float'
```

line 1143 in sdr.py states:

```
        pkt['light'] = Packet.get_float(obj, 'light_lux') * 0.007893
```

nothing earth shattering there... the math adds up whenever I plug in the values, even if the packet read 0 it would still be a float so not sure why it's throwing an error.

matthewwall, 2019-02-14:
fixed at commit cb40cea

the code was sloppy - it did not check for None before trying to multiply.  your sensors sometimes report None for light_lux.

but the driver should not do the conversion anyway

closed 2019-02-14 by matthewwall

phinnay, 2019-02-14:
thank you so much for looking into these issues!!