matthewwall/weewx-sdr #46: Using StdWXCalculate to convert lux to solar radiation

StephenR0, 2019-02-14:
Thank you for the recent commit that corrects my conversion from lux to solar radiation in the sdr driver.  I understand that there is no direct conversion from lux to solar radiation since they are different things.  The hardware in question is a Fine Offset WS-2902A.  The outdoor sensor array returns lux and the console converts that value to W/M^2, solar radiation.  Not ideal, but there it is.  So, I'm faced with accounting for this in Weewx.  So, I have two questions.

First, I'm trying to grapple with using the StdWXCalculate service to do this, but I'm not clear how that would be done.  From the documentation that service doesn't seem to handle radiation.  Can you give me a pointer or suggest a better way to do this?

Second, it's clear that rtl_433 sometimes returns no value for light from the WH65B.  Where ever this calculation is done, will I have to check for none before performing it?  Thanks.

matthewwall, 2019-02-14:
try this in your weewx configuration:

```
[StdCalibrate]
    [[Corrections]]
        radiation = light * 0.0079 if light is not None else None
```

StephenR0, 2019-02-14:
Thank you very much.  That looks great.

closed 2019-02-14 by StephenR0