matthewwall/weewx-sdr #47: Add support for Ambient Weather WH31E

phinnay, 2019-02-14:
Added a new class for the WH31 Outdoor temperature sensor. 

```
class AmbientWH31EPacket(Packet):
    # 2019-02-15 13:37:02 :   Ambient Weather WH31E
    #"protocol" : 113

    # {"time" : "2019-02-14 17:24:41.259441", "protocol" : 113, "model" : "AmbientWeather-WH31E", "id" : 24, "channel" : 1, 
	# "battery" : "OK", "temperature_C" : 6.000, "humidity" : 42, "data" :"2f00000000", "mic" : "CRC", "mod" : "FSK", "freq1" : 914.984, 
	# "freq2" : 914.906, "rssi" : -13.328, "snr" : 13.197, "noise" : -26.525}
    
	IDENTIFIER = "AmbientWeather-WH31E"

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRICWX
        pkt['station_id'] = obj.get('id')
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        return AmbientWH31EPacket.insert_ids(pkt)

    @staticmethod
    def insert_ids(pkt):
        station_id = pkt.pop('station_id', '0000')
        pkt = Packet.add_identifiers(pkt, station_id, AmbientWH31EPacket.__name__)
        return pkt
```
Appears to be working... I'm not big on python so adjust or fix any errors as needed!! 

Thank you again for your help and putting the time into this.

phinnay, 2019-02-14:
this is the sensor:
https://www.ambientweather.com/amwh31b.html

RTL_433 reads it as a WH31E for whatever reason.

matthewwall, 2019-02-14:
added at commit 4427be2

closed 2019-02-14 by matthewwall

cweakland, 2022-02-03:
Just a note, if your battery does not seem correct (always displaying low), you may need to change this line:

        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1

to:

        pkt['battery'] = 0 if obj.get('battery_ok') == 1 else 1

You can see why in the packet from RTL_433, not sure if this is related to my install or if something changed. Good luck! 

$ rtl_433 -M utc -F json -f 914980000 -s 250000
rtl_433 version 21.05-7-g78330501 branch master at 202106031747 inputs file rtl_tcp RTL-SDR
Use -h for usage help and see https://triq.org/ for documentation.
Trying conf file at "rtl_433.conf"...
Trying conf file at "/home/chris/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...

New defaults active, use "-Y classic -s 250k" for the old defaults!

Registered 157 out of 186 device decoding protocols [ 1-4 8 11-12 15-17 19-23 25-26 29-36 38-60 63 67-71 73-100 102-105 108-116 119 121 124-128 130-149 151-161 163-168 170-175 177-186 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 914.980MHz.
Allocating 15 zero-copy buffers
baseband_demod_FM: low pass filter for 250000 Hz at cutoff 50000 Hz, 20.0 us
{"time" : "2022-02-03 22:45:50", "model" : "AmbientWeather-WH31E", "id" : 221, "channel" : 1, "battery_ok" : 1, "temperature_C" : 20.100, "humidity" : 41, "data" : "6800000000", "mic" : "CRC"}