matthewwall/weewx-sdr #48: Anybody can help me writing a parser for the following Sensors: THGR968 and RGR968

chaoschris, 2019-02-15:
['{"time" : "2019-02-15 13:43:25", "brand" : "OS", "model" : "THGR968", "id" : 187, "channel" : 1, "battery" : "OK", "temperature_C" : 16.500, "humidity" : 11}\n', '{"time" : "2019-02-15 13:43:26", "brand" : "OS", "model" : "THGR968", "id" : 187, "channel" : 1, "battery" : "OK", "temperature_C" : 16.500, "humidity" : 11}\n']

['{"time" : "2019-02-15 14:32:51", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n', '{"time" : "2019-02-15 14:32:51", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n',

I was trying to modify an existing parser for OS, but without success....

chaoschris, 2019-02-15:
This was my first try:

class OSTHGR968Packet(Packet):

    IDENTIFIER = "THGR968"
    PARSEINFO = {
        'House Code': ['house_code', None, lambda x: int(x)],
        'Channel': ['channel', None, lambda x: int(x)],
        'Battery': ['battery', None, lambda x: 0 if x == 'OK' else 1],
        'Temperature': [
            'temperature', re.compile('([\d.-]+) C'), lambda x: float(x)],
        'Humidity': ['humidity', re.compile('([\d.]+) %'), lambda x: float(x)]}

    @staticmethod
    def parse_text(ts, payload, lines):
        pkt = dict()
        pkt['dateTime'] = ts
        pkt['usUnits'] = weewx.METRIC
        pkt.update(Packet.parse_lines(lines, THGR968Packet.PARSEINFO))
        return OS.insert_ids(pkt, THGR968Packet.__name__)

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRIC
        pkt['house_code'] = obj.get('id')
        pkt['channel'] = obj.get('channel')
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        return OS.insert_ids(pkt, THGR968Packet.__name__)

matthewwall, 2019-02-15:
please try the latest v0.58

support for these two sensors added at commit bcfce2a

chaoschris, 2019-02-15:
Okay, looks much better than my try.... ;-)
Thanks a lot!

root@raspberrypi:/usr/share/weewx# sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -R 12 -M utc -F json -G"
out: ['{"time" : "2019-02-15 15:52:42", "brand" : "OS", "model" : "BHTR968", "id" : 179, "channel" : 0, "battery" : "LOW", "temperature_C" : 19.900, "humidity" : 49, "pressure_hPa" : 999.000}\n', '{"time" : "2019-02-15 15:52:42", "brand" : "OS", "model" : "BHTR968", "id" : 179, "channel" : 0, "battery" : "LOW", "temperature_C" : 19.900, "humidity" : 49, "pressure_hPa" : 999.000}\n', '{"time" : "2019-02-15 15:52:42", "brand" : "OS", "model" : "BHTR968", "id" : 179, "channel" : 0, "battery" : "LOW", "temperature_C" : 19.900, "humidity" : 49, "pressure_hPa" : 999.000}\n', '{"time" : "2019-02-15 15:52:42", "brand" : "OS", "model" : "BHTR968", "id" : 179, "channel" : 0, "battery" : "LOW", "temperature_C" : 19.900, "humidity" : 49, "pressure_hPa" : 999.000}\n']
parsed: {'humidity.0:179.OSBTHR968Packet': 49.0, 'pressure.0:179.OSBTHR968Packet': None, 'battery.0:179.OSBTHR968Packet': 1, 'dateTime': 1550245962, 'temperature.0:179.OSBTHR968Packet': 19.9, 'usUnits': 16}
parsed: {'humidity.0:179.OSBTHR968Packet': 49.0, 'pressure.0:179.OSBTHR968Packet': None, 'battery.0:179.OSBTHR968Packet': 1, 'dateTime': 1550245962, 'temperature.0:179.OSBTHR968Packet': 19.9, 'usUnits': 16}
parsed: {'humidity.0:179.OSBTHR968Packet': 49.0, 'pressure.0:179.OSBTHR968Packet': None, 'battery.0:179.OSBTHR968Packet': 1, 'dateTime': 1550245962, 'temperature.0:179.OSBTHR968Packet': 19.9, 'usUnits': 16}
parsed: {'humidity.0:179.OSBTHR968Packet': 49.0, 'pressure.0:179.OSBTHR968Packet': None, 'battery.0:179.OSBTHR968Packet': 1, 'dateTime': 1550245962, 'temperature.0:179.OSBTHR968Packet': 19.9, 'usUnits': 16}
out: ['{"time" : "2019-02-15 15:52:45", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n', '{"time" : "2019-02-15 15:52:45", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n', '{"time" : "2019-02-15 15:52:45", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n', '{"time" : "2019-02-15 15:52:45", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n', '{"time" : "2019-02-15 15:52:45", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n', '{"time" : "2019-02-15 15:52:45", "brand" : "OS", "model" : "RGR968", "id" : 48, "channel" : 0, "battery" : "OK", "rain_rate" : 0.000, "total_rain" : 6935.100}\n']
parsed: {'total_rain.0:48.OSRGR968Packet': 6935.1, 'battery.0:48.OSRGR968Packet': 0, 'rain_rate.0:48.OSRGR968Packet': 0.0, 'usUnits': 16, 'dateTime': 1550245965}
parsed: {'total_rain.0:48.OSRGR968Packet': 6935.1, 'battery.0:48.OSRGR968Packet': 0, 'rain_rate.0:48.OSRGR968Packet': 0.0, 'usUnits': 16, 'dateTime': 1550245965}
parsed: {'total_rain.0:48.OSRGR968Packet': 6935.1, 'battery.0:48.OSRGR968Packet': 0, 'rain_rate.0:48.OSRGR968Packet': 0.0, 'usUnits': 16, 'dateTime': 1550245965}
parsed: {'total_rain.0:48.OSRGR968Packet': 6935.1, 'battery.0:48.OSRGR968Packet': 0, 'rain_rate.0:48.OSRGR968Packet': 0.0, 'usUnits': 16, 'dateTime': 1550245965}
parsed: {'total_rain.0:48.OSRGR968Packet': 6935.1, 'battery.0:48.OSRGR968Packet': 0, 'rain_rate.0:48.OSRGR968Packet': 0.0, 'usUnits': 16, 'dateTime': 1550245965}
parsed: {'total_rain.0:48.OSRGR968Packet': 6935.1, 'battery.0:48.OSRGR968Packet': 0, 'rain_rate.0:48.OSRGR968Packet': 0.0, 'usUnits': 16, 'dateTime': 1550245965}
out: ['{"time" : "2019-02-15 15:52:49", "model" : "Prologue sensor", "id" : 5, "rid" : 199, "channel" : 1, "battery" : "OK", "button" : 0, "temperature_C" : 11.400, "humidity" : 21}\n']
parsed: {'dateTime': 1550245969, 'humidity.199.ProloguePacket': 21.0, 'usUnits': 16, 'battery.199.ProloguePacket': 0, 'channel.199.ProloguePacket': 1, 'temperature.199.ProloguePacket': 11.4}
out: ['{"time" : "2019-02-15 15:52:55", "brand" : "OS", "model" : "THGR968", "id" : 187, "channel" : 1, "battery" : "OK", "temperature_C" : 12.400, "humidity" : 26}\n', '{"time" : "2019-02-15 15:52:55", "brand" : "OS", "model" : "THGR968", "id" : 187, "channel" : 1, "battery" : "OK", "temperature_C" : 12.400, "humidity" : 26}\n', '{"time" : "2019-02-15 15:52:56", "brand" : "OS", "model" : "THGR968", "id" : 187, "channel" : 1, "battery" : "OK", "temperature_C" : 12.400, "humidity" : 26}\n', '{"time" : "2019-02-15 15:52:56", "brand" : "OS", "model" : "THGR968", "id" : 187, "channel" : 1, "battery" : "OK", "temperature_C" : 12.400, "humidity" : 26}\n']
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550245975}
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550245975}
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550245976}
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550245976}
[... cut ...]
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550246012}
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550246013}
parsed: {'humidity.1:187.OSTHGR968Packet': 26.0, 'battery.1:187.OSTHGR968Packet': 0, 'temperature.1:187.OSTHGR968Packet': 12.4, 'usUnits': 16, 'dateTime': 1550246013}

closed 2019-02-17 by matthewwall