matthewwall/weewx-sdr #49: improve logging of unmapped

wjcarpenter, 2019-03-03:
Current logging of unmapped packets looks like this:

loginf("unmapped: %s (%s)" % (lines, packet))

But due to the surrounding logic, we already know that "packet" will always be empty. It would be more useful to log the original "packet" before it underwent map_to_fields() that makes it empty.

(This is especially useful if you want to make a fake sensor mapping entry for something that you know about but just don't care about, but you still want to log truly unmapped sensors. For example, I see some other weather stations on my radio, but I don't map them. Without the fake sensor mapping entries, they clutter my log file.)

wjcarpenter, 2019-03-03:
BTW, thanks for your work on weewx_sdr. Your write-up saved me a lot of figuring-out time.

matthewwall, 2019-03-03:
thank you for a thoughtful, very useful suggestion!

implemented at commit dc4ab90

closed 2019-03-03 by matthewwall