matthewwall/weewx-sdr #50: Parser for pressure doesn't work for OSBTHR968Packet. Other values from this sensor are parsed fine

chaoschris, 2019-03-06:
root@WEATHERCAM2:/usr/share/weewx# PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -M utc -F json -G" --hide out,empty

parsed: {'humidity.0:179.OSBTHR968Packet': 54.0, 'pressure.0:179.OSBTHR968Packet': None, 'battery.0:179.OSBTHR968Packet': 1, 'dateTime': 1551877367, 'temperature.0:179.OSBTHR968Packet': 19.9, 'usUnits': 16}
parsed: {'humidity.0:179.OSBTHR968Packet': 54.0, 'pressure.0:179.OSBTHR968Packet': None, 'battery.0:179.OSBTHR968Packet': 1, 'dateTime': 1551877367, 'temperature.0:179.OSBTHR968Packet': 19.9, 'usUnits': 16}

starting plain rtl_433 shows that pressure values are sent:

root@WEATHERCAM2:/usr/share/weewx# rtl_433 
rtl_433 version 18.12-91-gcab7ca9 branch master at 201902131710 inputs file rtl_tcp RTL-SDR
Trying conf file at "rtl_433.conf"...
Trying conf file at "/root/.config/rtl_433/rtl_433.conf"...
Trying conf file at "/usr/local/etc/rtl_433/rtl_433.conf"...
Trying conf file at "/etc/rtl_433/rtl_433.conf"...
Registered 96 out of 120 device decoding protocols [ 1-4 8 11-12 15-17 19-21 23 25-26 29-36 38-60 62-64 67-71 73-100 102-103 108-116 119 ]
Detached kernel driver
Found Rafael Micro R820T tuner
Exact sample rate is: 250000.000414 Hz
[R82XX] PLL not locked!
Sample rate set to 250000 S/s.
Tuner gain set to Auto.
Tuned to 433.920MHz.
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-03-06 14:05:19                    brand     : OS
model     : BHTR968      House Code: 179
Channel   : 0            Battery   : LOW           Celsius   : 19.80 C       Humidity  : 54 %          Pressure  : 974 hPa
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
time      : 2019-03-06 14:05:19                    brand     : OS
model     : BHTR968      House Code: 179
Channel   : 0            Battery   : LOW           Celsius   : 19.80 C       Humidity  : 54 %          Pressure  : 974 hPa

steepleian, 2019-03-06:
Not familiar with this kit but is the pressure sensor on a separate indoor
device?

matthewwall, 2019-03-06:
yet another change to rtl_433 output.  i really wish they would enforce some consistency in their naming.  sigh.  fixed at commit 39904df

closed 2019-03-06 by matthewwall