matthewwall/weewx-sdr #52: Wrong timezone since DST

mercurystorm, 2019-04-08:
Since DST in Europe, my weewx-sdr is reporting the incorrect timezone

```
Apr  8 10:38:09 raspberrypi weewx[705]: sdr: MainThread: lines=['{"time" : "2019-04-08 10:38:06", "model" : "Bresser-5in1", "id" : 144, "temperature_C" : 11.300, "humidity" : 83, "wind_gust" : 1.400, "wind_speed" : 1.500, "wind_dir_deg" : 90.000, "rain_mm" : 42.000, "mic" : "CHECKSUM"}\n']
Apr  8 10:38:09 raspberrypi weewx[705]: sdr: MainThread: packet={'outHumidity': 83.0, 'dateTime': 1554719886, 'windDir': 90.0, 'outTemp': 11.3, 'windSpeed': 1.5, 'rain_total': 42.0, 'usUnits': 17}
Apr  8 10:38:10 raspberrypi weewx[705]: extrasensors: found humidity value of 46.4000015259 %
Apr  8 10:38:10 raspberrypi weewx[705]: extrasensors: found temperature value of 20.8999996185 C
Apr  8 10:38:10 raspberrypi weewx[705]: restx: MQTT: Published record 2019-04-08 11:38:06 IST (1554719886)
```

I can see the time from the weather station is correct (10:38:06) but the EPOCH time on the sdr driver is 1554719886 which is 11:38:06

Any ideas?
```

root@raspberrypi:~# date
Mon  8 Apr 10:41:19 IST 2019
root@raspberrypi:~#
```

I have confirmed that my settings are correct on the Pi for timezone and python does print out the correct time 

```
root@raspberrypi:~# python -c "import time; print time.time()"
1554716534.58
```

```
root@raspberrypi:~# python -c "import time; print time.timezone"
0
```

matthewwall, 2019-05-01:
perhaps your time zone is not set correctly?  is IST the correct time zone for you?  what is your timezone offset supposed to be, and what is it actually?

this does not appear to be specific to weewx-sdr

if you still have timezone problems you might want to ask on weewx-user group at google groups

closed 2019-05-01 by matthewwall

mercurystorm, 2019-05-01:
IST is the correct timezone (Irish Standard Time) - The entire raspberry Pi is on the correct timezone.. its just the weewx-sdr packets that have not accounted for the daylight savings