matthewwall/weewx-sdr #53: pulling Celsius data as fahrenheit

slvrscoobie, 2019-04-12:
I have 8 wH31E that I use with WeeWx. 7 of them work as expected. 
Ive added them to my database schema but one of them is throwing celsius values in for fahrenheit. 
My 'freezer' is showing that its say -14F, when its actually -14 temp_C
From my log and weewx 
Apr 12 14:32:19 raspberrypi weewx[11956]: sdr: MainThread: lines=['{"time" : "2019-04-12 18:32:15", "model" : "AmbientWeather-WH31E", "id" : 94, "channel" : 8, "battery" : "OK", "temperature_C" : -11.800, "humidity" : 69, "data" : "9b00000000", "mic" : "CRC"}\n', '{"time" : "2019-04-12 18:32:15", "model" : "AmbientWeather-WH31E", "id" : 94, "channel" : 8, "battery" : "OK", "temperature_C" : -11.800, "humidity" : 69, "data" : "9b00000000", "mic" : "CRC"}\n']

LOOP:   2019-04-12 14:33:23 EDT (1555094003) dateTime: 1555094003, extraTemp8: -12.0, maxSolarRad: None, rainRate: 0, usUnits: 1

andylittle, 2019-04-23:
can you post an example of a working sensor?

slvrscoobie, 2019-04-23:
Matthew posted on my WeeWx Forum page that although I had extended the Schema, i didnt give the schema a unit. I havent had time to apply this fix, but I am sure thats probably whats happening. Ill close this for now.

closed 2019-04-23 by slvrscoobie