matthewwall/weewx-sdr #54: New sensor - WT0124 Pool Thermometer

gaz99gh, 2019-04-23:
I wish to add a WT0124 Pool Thermometer to my Weewx system, the sensor is now supported in rtl_433 but not in sdr.py.

The only output when manually running sdr.py is an 'out' line.

$ sudo PYTHONPATH=bin python bin/user/sdr.py --cmd="rtl_433 -M utc -F json -R 109"

out: ['{"time" : "2019-04-23 12:28:52", "model" : "WT0124 Pool Thermometer", "rid" : 122, "channel" : 1, "temperature_C" : 22.800, "mic" : "CHECKSUM", "data" : 172}\n']

$ sudo PYTHONPATH=bin python bin/user/sdr.py --cmd="rtl_433 -M utc -R 109"

out: ['_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ \n', 'time      : 2019-04-23 12:40:22\n', 'model     : WT0124 Pool Thermometer                Random ID : 122           Channel   : 1             Temperature: 23.0 C       Integrity : CHECKSUM\n', 'Data      : 172\n']

Any chance of adding this ??

Gary

andylittle, 2019-04-23:
Try this version of sdr.py version 0.48. https://groups.google.com/forum/#!topic/weewx-user/cJ0ozA0k8WQ

matthewwall, 2019-04-24:
please try sdr.py version 0.63 (commit 4cc1ac6)

gaz99gh, 2019-04-24:
Hi Matt,

V0.63 is working fine, thanks.

Gary

closed 2019-05-01 by matthewwall