matthewwall/weewx-sdr #57: Acurite Lightning 6045M Strikes Total - Issue Maybe

Stormtracker017, 2019-07-10:
I'm trying to add the strikes total for an archive period to the database, but is always getting a null value.  I don't see anything about the strikes_total in the packet.

`Jul  9 23:06:31 raspberrypi weewx[19366]: sdr: MainThread: lines=['{"time" : "2019-07-10 03:06:28", "model" : "Acurite Lightning 6045M", "id" : 14052, "channel" : "B", "temperature_F" : 66.500, "humidity" : 78, "strike_count" : 18, "storm_dist" : 15, "active" : 0, "rfi" : 0, "ussb1" : 1, "battery" : "OK", "exception" : 0, "raw_msg" : "b6e46f4e90f512cfbd"}\n', '{"time" : "2019-07-10 03:06:28", "model" : "Acurite Lightning 6045M", "id" : 14052, "channel" : "B", "temperature_F" : 66.500, "humidity" : 78, "strike_count" : 18, "storm_dist" : 15, "active" : 0, "rfi" : 0, "ussb1" : 1, "battery" : "OK", "exception" : 0, "raw_msg" : "b6e46f4e90f512cfbd"}\n'

I could add strike_count to the database, but i think this value is the counter that gets reset after 127. I don't know if this a issue or I have something wrong in my configuration.

Stormtracker017, 2019-07-17:
I attached a graphic showing how the strike_count just wraps around from 0 to 127 but the total lightning strikes for the archive period is not being updated for the 06045M. The AS3935 module seems to be updating correctly.

[image]

matthewwall, 2019-12-01:
the sensor reports `strike_count`, but the weewx-sdr driver should convert that to `strikes_total`

then the difference between the `strikes_total` in each period should be reported in the `strikes` field.  so `strikes` is the field you want in the database.  it is analogous to `rain` - it is a delta value, not a cumulative value.

the fact that the acurite sensor wraps every 128 counts could cause some problems, especially if you get more than 128 strikes in a time interval.  we might have to add some logic to deal with wraparounds in general, and the acurite sensor in particular. 

what are you using for the accumulated total, and how are you doing the accumulating?

Stormtracker017, 2019-12-02:
Here is my mapping for the 06045M  sensors:

[[sensor_map]]
        # 5n1 Sensor
        
        windDir = wind_dir.0C37.Acurite5n1Packet
        windSpeed = wind_speed.0C37.Acurite5n1Packet
        outTemp = temperature.0C37.Acurite5n1Packet
        outHumidity = humidity.0C37.Acurite5n1Packet
        rain_total = rain_total.0C37.Acurite5n1Packet
        
        # 06045M 
        
        LS_1_active = active.36E4.AcuriteLightningPacket
        LS_1_distance = distance.36E4.AcuriteLightningPacket
        LS_1_lightning_strikes = strikes_total.36E4.AcuriteLightningPacket
        LS_1_rfi = rfi.36E4.AcuriteLightningPacket
        LS_1_temperature = temperature.36E4.AcuriteLightningPacket
        LS_1_humidity = humidity.36E4.AcuriteLightningPacket
        LS_1_battery = battery.36E4.AcuriteLightningPacket
        
        LS_2_active = active.00E4.AcuriteLightningPacket
        LS_2_distance = distance.00E4.AcuriteLightningPacket
        LS_2_lightning_strikes = strikes_total.00E4.AcuriteLightningPacket
        LS_2_rfi = rfi.00E4.AcuriteLightningPacket
        LS_2_temperature = temperature.00E4.AcuriteLightningPacket
        LS_2_humidity = humidity.00E4.AcuriteLightningPacket
        LS_2_battery = battery.00E4.AcuriteLightningPacket

Here is what I see:
 
Dec  1 21:59:13 raspberrypi weewx[582]: sdr: MainThread: lines=['{"time" : "2019-12-02 02:59:10", "model" : "Acurite Lightning 6045M", "id" : 228, "channel" : "C", "temperature_F" : 30.000, "humidity" : 99, "strike_count" : 91, "storm_dist" : 5, "active" : 0, "rfi" : 0, "ussb1" : 1, "battery" : "OK", "exception" : 0, "raw_msg" : "00e46f638e88dbc56c"}\n', '{"time" : "2019-12-02 02:59:10", "model" : "Acurite Lightning 6045M", "id" : 228, "channel" : "C", "temperature_F" : 30.000, "humidity" : 99, "strike_count" : 91, "storm_dist" : 5, "active" : 0, "rfi" : 0, "ussb1" : 1, "battery" : "OK", "exception" : 0, "raw_msg" : "00e46f638e88dbc56c"}\n', '{"time" : "2019-12-02 02:59:10", "model" : "Acurite Lightning 6045M", "id" : 228, "channel" : "C", "temperature_F" : 30.000, "humidity" : 99, "strike_count" : 91, "storm_dist" : 5, "active" : 0, "rfi" : 0, "ussb1" : 1, "battery" : "OK", "exception" : 0, "raw_msg" : "00e46f638e88dbc56c"}\n', '{"time" : "2019-12-02 02:59:10", "model" : "Acurite tower sensor", "id" : 4079, "sensor_id" : 4079, "channel" : "A", "temperature_C" : 23.800, "humidity" : 26, "battery_low" : 1}\n', '{"time" : "2019-12-02 02:59:10", "model" : "Acurite tower sensor", "id" : 4079, "sensor_id" : 4079, "channel" : "A", "temperature_C" : 23.800, "humidity" : 26, "battery_low" : 1}\n', '{"time" : "2019-12-02 02:59:10", "model" : "Acurite tower sensor", "id" : 4079, "sensor_id" : 4079, "channel" : "A", "temperature_C" : 23.800, "humidity" : 26, "battery_low" : 1}\n']

Dec  1 21:59:14 raspberrypi weewx[582]: bme280: {'LS_2_distance': 5, 'LS_2_humidity': 99, 'LS_2_active': 0, 'dateTime': 1575255550, 'LS_2_battery': 0, 'inHumidity': 27.88299552542264, 'inTemp': 80.35782372527291, 'LS_2_temperature': 30.0, 'LS_2_rfi': 0, 'LS_2_lightning_strikes': 91, 'usUnits': 1}

I'm using sum for the accumulated total and taking the last three archive periods (900 seconds) and summing them together.

matthewwall, 2019-12-02:
try this in your weewx config:

```
    [[sensor_map]]
        ...
        LS_1_lightning_strikes_total = strikes_total.36E4.AcuriteLightningPacket
        ...
        LS_2_lightning_strikes_total = strikes_total.00E4.AcuriteLightningPacket
        ...
    [[deltas]]
        rain = rain_total
        LS_1_lightning_strikes = LS_1_lightning_strikes_total
        LS_2_lightning_strikes = LS_1_lightning_strikes_total
```

Stormtracker017, 2019-12-02:
Made the changes in the weewx config file, the Lightning Total equals zero.  Just need to wait for a Thunderstorm or two.
[image]

closed 2019-12-03 by Stormtracker017

jrhelbert, 2020-04-08:
Working to help round out the Atlas code in rtl_433 (https://groups.google.com/d/msg/rtl_433/gvTUFWJlFxE/oDidDIx9AAAJ) and running into the lightning wrap around issue.

Seems like the best option to deal with the wraparound is for weewx-sdr to be aware of the field size. ie the Atlas code should be aware of how big fields are (at least fields that are cumulative and we might want to run deltas against.) Then we can use that value to calculate the delta whenever there is a wraparound.

rct, 2020-04-13:
In case anyone finds this closed issue - see #81 - The decoding of the 6045 strike counter changed. It was incorrectly reported as 7 bits in rtl_433 before April 13 2020.