matthewwall/weewx-sdr #59: Unsupported format for AcuriteTowerPacket and Acurite5n1Packet added

networkname, 2019-07-29:
Modified sdr.py to support the a different sensor format. I called themAcuriteTowerPacketV2 and Acurite5n1PacketV2. Adding new classes seemed cleaner than continuing to add to the existing formats. Also, I included all the fields. I find the SNR and such helpful. Anyway hope this might help someone else. I would check this into the repository but do not think I have access.

[sdr.py.gz](https://github.com/matthewwall/weewx-sdr/files/3442951/sdr.py.gz)

matthewwall, 2019-12-01:
what arguments do you feed to rtl_433 to get the json output for these?

closed 2019-12-12 by matthewwall

wjcarpenter, 2020-05-25:
This caused a head-scratching moment for me. Well, more than a moment.
I'm moving my weewx from one RPi to another and also upgrading a zillion things along the way. When I copied over my working SDR config to the new environment, nothing was being reported. The weewx main packet loop just seemed dead. Everything else seemed fine, including running `sdr.py` standalone. After an embarrassingly long time, I finally noticed the `V2` suffix. Doh!

I'm not sure what makes my station be seen as `Acurite5n1V2` instead of `Acurite5n1` as it was before. It's the same transmitter. I'm mentioning this here in case it helps some other unobservant person