matthewwall/weewx-sdr #65: Fine Offset WH5 sensor not parsed

starvald, 2019-10-27:
Please can you add support for this.

out: ['{"time" : "2019-10-27 10:54:33", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 8.800, "humidity" : 72, "mic" : "CRC"}\n']

not sure if it matters but it alternates between a single signal and a double one.

out: ['{"time" : "2019-10-27 10:55:21", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 8.800, "humidity" : 72, "mic" : "CRC"}\n', '{"time" : "2019-10-27 10:55:21", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 8.800, "humidity" : 72, "mic" : "CRC"}\n']

thanks.

starvald, 2019-10-27:
ok, gave myself a break to mentally recover and had another look at exmaples in sdr.py.
i think i got this.

`class FOWH5Packet(Packet):
    # {"time" : "2019-10-27 14:51:21", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 11.700, "humidity" : 62, "mic" : "CRC"}

    IDENTIFIER = "Fine Offset WH5 sensor"
    PARSEINFO = {
        'ID': ['station_id', None, lambda x: int(x)],
        'Temperature':
            ['temperature', re.compile('([\d.-]+) C'), lambda x: float(x)]
        }

    @staticmethod
    def parse_text(ts, payload, lines):
        pkt = dict()
        pkt['dateTime'] = ts
        pkt['usUnits'] = weewx.METRIC
        pkt.update(Packet.parse_lines(lines, FOWH5Packet.PARSEINFO))
        return FOWH5Packet.insert_ids(pkt)

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRIC
        pkt['station_id'] = obj.get('id')
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        return FOWH5Packet.insert_ids(pkt)

    @staticmethod
    def insert_ids(pkt):
        station_id = pkt.pop('station_id', '0000')
        return Packet.add_identifiers(pkt, station_id, FOWH5Packet.__name__)
`

FOWH5Packet added to KNOWN.PACKETS

starvald, 2019-10-27:
sudo PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -R 18 -M utc -F json" 

```
out: ['{"time" : "2019-10-27 15:40:09", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 11.300, "humidity" : 64, "mic" : "CRC"}\n', '{"time" : "2019-10-27 15:40:09", "model" : "Fine Offset WH5 sensor", "id" : 48, "temperature_C" : 11.300, "humidity" : 64, "mic" : "CRC"}\n']
parsed: {'humidity.48.FOWH5Packet': 64.0, 'temperature.48.FOWH5Packet': 11.3, 'usUnits': 16, 'dateTime': 1572190809}
parsed: {'humidity.48.FOWH5Packet': 64.0, 'temperature.48.FOWH5Packet': 11.3, 'usUnits': 16, 'dateTime': 1572190809}
```

matthewwall, 2019-12-01:
added at commit f6e1010

closed 2019-12-01 by matthewwall

starvald, 2022-11-02:
Hi, just started using this again after one of my Oregon scientific devices died and looks like rtl_433 has changed the device name.
it's now Fineoffset-WH5

out: ['{"time" : "2022-11-02 19:38:55", "model" : "Fineoffset-WH5", "id" : 59, "temperature_C" : 14.300, "humidity" : 72, "mic" : "CRC"}\n', '{"time" : "2022-11-02 19:38:55", "model" : "Fineoffset-WH5", "id" : 59, "temperature_C" : 14.300, "humidity" : 72, "mic" : "CRC"}\n']
parsed: {'dateTime': 1667417935, 'usUnits': 16, 'temperature.59.FOWH5Packet': 14.3, 'humidity.59.FOWH5Packet': 72.0}
parsed: {'dateTime': 1667417935, 'usUnits': 16, 'temperature.59.FOWH5Packet': 14.3, 'humidity.59.FOWH5Packet': 72.0}