matthewwall/weewx-sdr #66: Parser for Oregon Scientific UVR128

sirrated, 2019-11-06:
First off, thank you for making this WeeWx driver available!

Can some kind soul write a parser for the Oregon Scientific UVR128 UV Index sensor?  RTL_433 added support in 2018.  Here is the output:

    out: ['{"time" : "2019-11-05 07:07:07", "model" : "Oregon Scientific UVR128", "id" : 116, "uv" : 0, "battery" : "OK"}\n']

Thank you!

sirrated, 2019-11-19:
I made an attempt to write my own parser, based on similar parsers already in sdr.py:

	class OSUVR128Packet(Packet):
		# 2019-11-05 07:07:07 : Oregon Scientific UVR128
		# House Code: 116
		# UV Index: 0
		# Battery: OK

		IDENTIFIER = "Oregon Scientific UVR128"
		PARSEINFO = {
			'House Code': ['house_code', None, lambda x: int(x)],
			'UV Index':
				['uv_index', re.compile('([\d.-]+) C'), lambda x : float(x)],
			'Battery': ['battery', None, lambda x: 0 if x == 'OK' else 1]}

		@staticmethod
		def parse_text(ts, payload, lines):
			pkt = dict()
			pkt['dateTime'] = ts
			pkt['usUnits'] = weewx.METRIC
			pkt.update(Packet.parse_lines(lines, OSUVR128Packet.PARSEINFO))
			return OS.insert_ids(pkt, OSUVR128Packet.__name__)

			# {"time" : "2019-11-19 06:44:53", "model" : "Oregon Scientific UVR128", "id" : 116, "uv" : 0, "battery" : "OK"}

		@staticmethod
		def parse_json(obj):
			pkt = dict()
			pkt['dateTime'] = Packet.parse_time(obj.get('time'))
			pkt['usUnits'] = weewx.METRIC
			pkt['house_code'] = obj.get('id')
			pkt['uv_index'] = Packet.get_float(obj, 'uv')
			pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
			return OS.insert_ids(pkt, OSUVR128Packet.__name__)

It appears to be working:

	out: ['{"time" : "2019-11-19 07:21:23", "model" : "Oregon Scientific UVR128", "id" : 116, "uv" : 0, "battery" : "OK"}\n']
	parsed: {'uv_index.0:116.OSUVR128Packet': 0.0, 'battery.0:116.OSUVR128Packet': 0, 'usUnits': 16, 'dateTime': 1574148083}

If you attempt writing your own parser, don't forget to add the new class to the KNOWN_PACKETS list. :)  I'm still new to GitHub so I'm still trying to figure out how to contribute.

andylittle, 2019-11-26:
[tracked here also](https://groups.google.com/forum/#!topic/weewx-user/tGKy0wqMMfs)

sirrated, 2019-11-26:
I created a fork with changes to sdr.py to add the UVR128 parser.  I plan to give it another week of testing before creating a pull request.

matthewwall, 2019-12-01:
added at commit 27eadb9

closed 2019-12-01 by matthewwall