matthewwall/weewx-sdr #69: Acurite 899 Rain gauge not parsed

rknobbe, 2019-12-05:
I have a patch for sdr.py to add Acurite Rain899 parsing to sdr.py

`--- weewx-sdr-master/bin/user/sdr.py    2019-12-02 09:18:52.000000000 -0800
+++ bin/user/sdr.py     2019-12-05 09:06:44.962474631 -0800
@@ -358,6 +358,36 @@
         return Packet.add_identifiers(pkt, sensor_id, pkt_type)

+class AcuriteRain899Packet(Packet):
+    # Based on AcuriteTowerPacket type, but implemented for unsupported format
+    IDENTIFIER = "Acurite-Rain899"
+    # Sample data:
+    # '{"time" : "2019-12-05 16:32:20", "model" : "Acurite-Rain899", "id" : 1699, "channel" : 0, "battery_ok" : 0, "rain_mm" : 6.096}\n'
+    # '{"time" : "2019-12-05 16:32:20", "model" : "Acurite-Rain899", "id" : 1699, "channel" : 0, "battery_ok" : 0, "rain_mm" : 6.096}\n'
+    # '{"time" : "2019-12-05 16:32:20", "model" : "Acurite-Rain899", "id" : 1699, "channel" : 0, "battery_ok" : 0, "rain_mm" : 6.096}\n'
+
+    @staticmethod
+    def parse_json(obj):
+        pkt = dict()
+        pkt['usUnits'] = weewx.METRIC
+        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
+        #pkt['protocol'] = Packet.get_int(obj, 'protocol') # 40
+        pkt['model'] = obj.get('model') # model = Acurite-Rain899
+        pkt['hardware_id'] = "%04x" % obj.get('id', 0)
+        #pkt['sensor_id'] = "%04x" % obj.get('sensor_id', 0)
+        pkt['channel'] = obj.get('channel')
+        #pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
+        #pkt['humidity'] = Packet.get_float(obj, 'humidity')
+        pkt['battery'] = Packet.get_int(obj, 'battery_ok') # 1 means battery OK, 0 means not low apparently
+        #pkt['mod'] = obj.get('mod') # apparently mod = ASK
+        #pkt['freq'] = Packet.get_float(obj, 'freq')
+        #pkt['rssi'] = Packet.get_float(obj, 'rssi')
+        #pkt['snr'] = Packet.get_float(obj, 'snr')
+        #pkt['noise'] = Packet.get_float(obj, 'noise')
+        if 'rain_mm' in obj:
+            pkt['rain_total'] = Packet.get_float(obj, 'rain_mm')/25.4
+        return Acurite.insert_ids(pkt, AcuriteRain899Packet.__name__)
+
 class AcuriteTowerPacketV2(Packet):
     # Based on AcuriteTowerPacket type, but implemented for unsupported format
     IDENTIFIER = "Acurite-Tower"
@@ -2350,6 +2380,7 @@
         AcuriteTowerPacket,
         Acurite5n1Packet,
         AcuriteTowerPacketV2,
+        AcuriteRain899Packet,
         Acurite5n1PacketV2,
         Acurite606TXPacket,
         Acurite986Packet,
`

matthewwall, 2019-12-12:
added at commit dd35822

closed 2019-12-12 by matthewwall

rknobbe, 2019-12-13:
Hold off on using this.  Data must be getting misinterpreted because it hasn't rained for days but the sensor is still reporting rain_mm valued in every sample.  This may be a problem with rtl_433.

`out:[u'{"time" : "2019-12-13 13:43:08", "model" : "Acurite-Rain899", "id" : 1699, "channel" : 0, "battery_ok" : 0, "rain_mm" : 6.096}\n', u'{"time" : "2019-12-13 13:43:08", "model" : "Acurite-Rain899", "id" : 1699, "channel" : 0, "battery_ok" : 0, "rain_mm" : 6.096}\n', u'{"time" : "2019-12-13 13:43:08", "model" : "Acurite-Rain899", "id" : 1699, "channel" : 0, "battery_ok" : 0, "rain_mm" : 6.096}\n']
`
[image]