matthewwall/weewx-sdr #70: Parser for TS-FT002 433 mhz water level gauge for WeeWx

splobsterman, 2019-12-22:
This gadget uses ultrasound to provide liquid level data but needs a parser to use with WeeWx.

rtl_433 sees it using:
`rtl_433 -R 125`

Any pointers about how to do this?  I took a look at sdr.py and it's well above my pay grade.

rtl_433 output:

> time      : 2019-12-22 16:57:58
model     : TS-FT002     Id        : 127
Depth     : 186          Temperature: 20.9 C       Transmit Interval: 180    Battery Flag?: 8          MIC       : CHECKSUM

matthewwall, 2019-12-22:
please run rtl_433 but with the JSON output.  something like this:
```
rtl_433 -M utc -R 125 -F JSON
```
then post some of the resulting output.  5 or 10 samples should be enough.

splobsterman, 2019-12-23:
Matthew,

With thanks:

{"time" : "2019-12-22 22:54:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 22:54:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 22:57:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 22:57:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:00:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:00:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:03:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:03:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:06:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:06:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:09:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:09:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:12:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:12:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:15:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:15:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:18:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:18:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:21:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:21:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:24:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}
{"time" : "2019-12-22 23:24:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 186, "temperature_C" : 20.700, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}

matthewwall, 2019-12-23:
please try weewx-sdr as of commit fe08ccc

splobsterman, 2019-12-23:
Matthew,

The parser works but with an odd result:

Running sudo PYTHONPATH=bin python bin/user/sdr.py --cmd="rtl_433 -M utc -F json -G”:

out:[u'{"time" : "2019-12-23 17:25:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 62, "temperature_C" : 26.400, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}\n', u'{"time" : "2019-12-23 17:25:58", "model" : "TS-FT002", "id" : 127, "depth_cm" : 62, "temperature_C" : 26.400, "transmit_s" : 180, "flags" : 8, "mic" : "CHECKSUM"}\n']
parsed: {'transmit.0000.TSFT002Packet': 180.0, 'dateTime': 1577121958, 'flags.0000.TSFT002Packet': 180, 'temperature.0000.TSFT002Packet': 26.4, 'depth.0000.TSFT002Packet': 62.0, 'usUnits': 16}

So I added 'depth.0000.TSFT002Packet’ to my SDR sensor map:

[SDR]
    # This section is for the software-defined radio driver.
    
    # The driver to use
    driver = user.sdr
    log_unknown_sensors = True
    log_unmapped_sensors = True
    [[sensor_map]]
        extraTemp2 = temperature.1:215.AmbientF007THPacket   	 	# pool temperature
        extraTemp3 = temperature.001A.AcuriteLightningPacket    		# temp at lightning detector
        soilTemp1 = strikes_total.001A.AcuriteLightningPacket    		# total strikes
        soilTemp2 = distance.001A.AcuriteLightningPacket    			# strike distance
        leafTemp1 = depth.0000.TSFT002Packet					# depth in cm tank level

and when I run sir.conf, it transmits this to the sdr.sdb:

LOOP:   2019-12-23 13:49:58 EST (1577126998) dateTime: 1577126998, leafTemp1: 147.2, maxSolarRad: 574.533540052, rainRate: 0, usUnits: 1

I am unsure how the 62 CM (I’ve checked it with my metric tape) gets transformed to 147.2. Have I mis-configured something?

Regards, and best wishes for the holidays.

matthewwall, 2019-12-25:
you need to tell weewx what units to use.  in its default configuration, leaftTemp1 is temperature, not length.

put this into the file `user/extensions.py` then restart weewx.  also, be sure to clear the old data.

```
import weewx.units
weewx.units.USUnits['group_range'] = 'inch'
weewx.units.MetricUnits['group_range'] = 'cm'
weewx.units.MetricWXUnits['group_range'] = 'cm'
weewx.units.obs_group_dict['leafTemp1'] = 'group_range’
```

closed 2019-12-25 by matthewwall

splobsterman, 2019-12-28:
Thank you Matthew, most helpful.