matthewwall/weewx-sdr #71: Changing packet definition in Ambient Weather F007TH Thermo-Hygrometer

splobsterman, 2019-12-24:
I'm running an Ambient Weather Thermo-Hygrometer (actually a floating thermometer for a pool). WeeWx-sdr picks it up and parses it just fine, and entering the temperature packet in the Sensor Map works fine. But after a few hours or a day, it stops being recognized. Going back and running the sdr.py parser I find that the temperature packet has changed.

For example

> out:[u'{"time" : "2019-12-23 15:48:34", "model" : "Ambient Weather F007TH Thermo-Hygrometer", "device" : 215, "channel" : 1, "battery" : "OK", "temperature_F" : 81.100, "humidity" : 43, "mic" : "CRC"}\n']
parsed: {'humidity.1:215.AmbientF007THPacket': 43.0, 'temperature.1:215.AmbientF007THPacket': 81.1, 'usUnits': 1, 'dateTime': 1577116114}

The "215" digits in the temperature packet change to something else. The little remote display which comes with the unit continues to display the temp properly.

Is this a bug in that unit? If not, can I wildcard/regex those digits in the sensor map to trap the changes?

matthewwall, 2019-12-25:
this is not unusual - some sensors reset their identifier whenever the power is reset, or sometimes just randomly.

if you only have one sensor of that type, use a wildcard to match it in your sensor map:

```
humidity = humidity.*.AmbientF007THPacket
extraTemp2 = temperature.*.AmbientF007THPacket
```

if you have multiple devices, you can configure one to each channel, then distinguish based on channel.

in the future, please post your questions to the weewx user group, so others can benefit from what you learn.  this space is for reporting bugs and some feature requests.

closed 2019-12-25 by matthewwall

splobsterman, 2019-12-28:
Will do.