matthewwall/weewx-sdr #75: Need help adding Ecowitt WH40 rain gauge sensor to weewx

Helmut63, 2020-02-05:
As per https://github.com/merbanan/rtl_433/issues/1275  we just finished the process of adding this new Ecowitt WH40 rain gauge sensor to rtl_433 driver. 
It now runs fine after updating the driver to the newest version, but I still cannot see the data in weewx.

Here are the details:
1) Running the driver apart from weewx.
rtl_433 -M utc -F json
{"time" : "2020-02-05 12:37:05", "model" : "EcoWitt-WH40", "id" : 52591, "rain_mm" : 0.800, "data" : "0002ed0000", "mic" : "CRC"}

rtl_433 -M utc
time      : 2020-02-05 12:35:27
model     : EcoWitt-WH40 id        : 52591
Total Rain: 0.8 mm       Extra Data: 00027b0000    Integrity : CRC

(The 0.8mm just represents some manual tips of the rain bucket) 

2) Then I added a line to the sensor map (where I have already other sensors reporting)

"rain_total = rain_total.52591.EcoWittWH40Packet"

Running weewx in debug mode and reporting unmapped and unrecognized sensors, I get this message:
Feb  5 10:13:05 raspberrypi weewx[14245]: sdr: MainThread: parse_json: unknown model EcoWitt-WH40
Feb  5 10:13:05 raspberrypi weewx[14245]: sdr: MainThread: punt unrecognized line '{"time" : "2020-02-05 13:13:01", "model" : "EcoWitt-WH40", "id" : 52591, "rain_mm" : 0.800, "data" : "0002de0000", "mic" : "CRC"}#012'

I'd appreciate your help  in understanding what else needs to be done for Weewx to recognize this new sensor (how to overcome the "unrecognized model..." and "unrecognized line..." messages), and/or any modification to the sensor map line, but surely once we see an "unparsed..." message we will be able to use the right naming for the sensor map definition.

Thanks in advance!
Best regards,
Helmut

matthewwall, 2020-02-06:
please try weewx-sdr 0.77 (as of commit 836591a)

closed 2020-02-06 by matthewwall

Helmut63, 2020-02-06:
Thanks a lot, will try it in a minute... 
Just unsure on how the sersor map line in weewx.conf should look like:
either "rain_total = rain_mm.52591.EcoWittWH40Packet"
or "rain_total = rain_total.52591.EcoWittWH40Packet"
Anything else than above?

Thks & BR

matthewwall, 2020-02-06:
```
[SDR]
    [[sensor_map]]
        rain_total = rain_total.52591.EcoWittWH40Packet
```

Helmut63, 2020-02-06:
Hi, I updated weewx-sdr and tried all kind of stuff, but for some reason after updating the sw I lost my other sensors, except EcoWitt WH31. Now looking at the weewx-sdr 077 (as of commit [836591a](https://github.com/matthewwall/weewx-sdr/commit/836591a11a537a62d598fab25edf8fdd9657d009) , it seems strange seeing at line 1024 "return FOWH32BPacket.insert_ids(pkt)" in the middle of the EcoWitt WH40 section.
Maybe that is ok, but just would appreciate you confirmation before staring a whole new install based on instructions from https://github.com/matthewwall/weewx-sdr
Thanks again! BR
[image]

matthewwall, 2020-02-06:
sorry about that.  fixed at commit 0d09a22

Helmut63, 2020-02-07:
Great, no problem! Have it up & running now, with a minor change in sensor map entry, as the "unmapped" message seen while debugging showed "({'rain.52591.EcoWittWH40Packet': 3.1, 'usUnits': 17, 'dateTime': 1581029709})
After changing it to this line below I started receiving updates from rain gauge for every manual bucket tip.
So just for the record, this is the line that worked for me:
[SDR]
    [[sensor_map]]
        rain_total = rain.52591.EcoWittWH40Packet

matthewwall, 2020-02-07:
helmut,

does the WH40 report a rain total (cumulative rain total)  or a rain delta (amount of rain since last reading)?

Helmut63, 2020-02-08:
As the rain bucket tips over, the counter increases, and the rtl_433 driver initially reported numbers from the counter. As we then identified the representing amount in mm (0.1mm for each tip) it now reports a cumulative rain total, and it is weewx that afterwards shows the amount of rain (Rain Today) since last midnight. 
Not sure what will happen if I reset the battery at the sensor, as the counter will reset to zero, what will weewx do? Take from there and start over as well? Probably I'd try to reset the battery on a sunny day, so any day ahead with some rain, weewx will read the new delta from last midnight...

Helmut63, 2020-02-08:
As mentioned earlier, as part of the upgrades of weewx-sdr, I had lost some other sensors I had running before. Now that I have the WH40 and WH31 running, I started seeking for the lost ones, and I´m happy that I finally had some progress and just wanted to let you know what I found.

Somehow the reporting of the previous sensors changed as part of the weewx-sdr update and I then had more "unmapped sensors" in the list of the debugging output. As I initially only focused on the WH40 entries, I didn´t realize the changes for the other sensors. 
For instance, all my "Acurite5n1Packet" previous mappings needed to be changed to "Acureite5n1PacketV2", "AcuriteTowerPacket" changed to "AcuriteTowerPacketV2". Still need to find some other changes that may have occurred as I also lost other ones, i.e. "FOWH24Packet" and "AlectoV1TemperaturePacket".
Is there any list or publication I may have missed? Otherwise I´ll continue seeking in the debug reports for these changes.
Thanks again and hopefully this might help others running into similar issues...

matthewwall, 2020-02-08:
please create a new issue if you have a problem, and please post log output, as well as an explanation of what version of each software (rtl-sdr, rtl_433, weewx-sdr) you are running.  otherwise i have to spend a lot of time guessing about what actually happened, and i will probably guess wrong.

matthewwall, 2020-02-08:
> As the rain bucket tips over, the counter increases, and the rtl_433 driver initially reported numbers from the counter. As we then identified the representing amount in mm (0.1mm for each tip) it [...]

it sounds like the device reports a cumulative total, not an incremental measurement.  i have modified the weewx-sdr driver accoringly at commit fb3f9fe

but if you would post the logs then i would not have to guess :)