matthewwall/weewx-sdr #79: New model names in rtl_433 cause "unknown model" issues with sdr.py

benlye, 2020-03-30:
It looks like rt_433 has changed the model names it uses when parsing data, see https://github.com/merbanan/rtl_433/pull/986

For now it is possible to use the old names by appending `-M oldmodel` but this generates a warning at startup:

```
 Warning: Using deprecated old model keys ("-M oldmodel"). This will be removed shortly.
```

It seems like sdr.py needs updating to use the new model names, because without the `-M oldmodel`  switch on rtl_433 the output is not parsed by the SDR driver.

For example, trying to read from a WH1080 weather station produced this output in the log:

```
Mar 30 10:37:12 pi4 weewx[6819]: sdr: MainThread: lines=[u'{"time" : "2020-03-30 09:37:08", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 109, "battery_ok" : 1, "temperature_C" : 5.600, "humidity" : 82, "wind_dir_deg" : 180, "wind_avg_km_h" : 6.120, "wind_max_km_h" : 9.792, "rain_mm" : 1038.900, "mic" : "CRC"}\n', u'{"time" : "2020-03-30 09:37:08", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 109, "battery_ok" : 1, "temperature_C" : 5.600, "humidity" : 82, "wind_dir_deg" : 180, "wind_avg_km_h" : 6.120, "wind_max_km_h" : 9.792, "rain_mm" : 1038.900, "mic" : "CRC"}\n']
Mar 30 10:37:12 pi4 weewx[6819]: sdr: MainThread: parse_json: unknown model Fineoffset-WHx080
Mar 30 10:37:12 pi4 weewx[6819]: sdr: MainThread: punt unrecognized line '{"time" : "2020-03-30 09:37:08", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 109, "battery_ok" : 1, "temperature_C" : 5.600, "humidity" : 82, "wind_dir_deg" : 180, "wind_avg_km_h" : 6.120, "wind_max_km_h" : 9.792, "rain_mm" : 1038.900, "mic" : "CRC"}#012'
Mar 30 10:37:12 pi4 weewx[6819]: sdr: MainThread: parse_json: unknown model Fineoffset-WHx080
Mar 30 10:37:12 pi4 weewx[6819]: sdr: MainThread: punt unrecognized line '{"time" : "2020-03-30 09:37:08", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 109, "battery_ok" : 1, "temperature_C" : 5.600, "humidity" : 82, "wind_dir_deg" : 180, "wind_avg_km_h" : 6.120, "wind_max_km_h" : 9.792, "rain_mm" : 1038.900, "mic" : "CRC"}#012'
```

mitchins, 2020-06-20:
I'm fighting this exact same issue right now.
Temporarily fixed in my instance by:

`sed -i 's/Fine Offset WH1080 weather station/Fineoffset-WHx080/g' /usr/share/weewx/user/sdr.py
`

mitchins, 2020-06-22:
Follow-up info:
https://github.com/merbanan/rtl_433/commit/12b50e5370ac153ee7848aa545d40c1f1669ea5b
This was done over a year ago.

mitchins, 2020-06-22:
Furthermore it looks like some of the key names have changed in this version too:

[image]

I'm made a PR locally, I wasn't sure if it was just my variant of the machine, but now it truly looks like the following diff:
[image]

I've already got this ready @matthewwall  what's the strategy here, RTL-433 has been changed, has nobody else been hit by this?

I've got a ready to go fix for the WH-1080 including model name and wind key fixes which I've tested extensively. I've also been tested on RTL-433 v19 and it indeed behaves as your whew plugin expected.

Presumably anyone who takes the latest RTL-433 available in ubuntu-focal's repo will have this problem also, which I imagine if they intend to use RTL-433 v19, they should use a stable/tagged version not master right?

With that in mind I'll create a PR to master, we'll have to get a move on updating... but I can't own models outside of WH1080, let me know in the PR if you prefer a new branch for RTL-433 v20

vittorio88, 2020-07-17:
Issue confirmed on Ubuntu Focal and **Fine Offset WH65B** or **Fineoffset-WH65B** 

I think since this code is based on sdr_433, it should be updated to match the newest names in sdr_433. Anybody who updates their copy of weewx-sdr will probably also obtain a recent sdr_433 that will break weewx-sdr, and people who don't update won't be affected, so I think it's *mostly* safe to make this change.

If nobody objects, @mitchins  would you mind making a PR for this?

mitchins, 2020-07-17:
@vittorio88 the names are updated in #88

scopey, 2021-01-19:
I had the same issue with the Fine Offset WH24.   OLD = IDENTIFIER = "Fine Offset WH24 "  NEW = IDENTIFIER = "Fineoffset-WH24