matthewwall/weewx-sdr #80: 5n1 rain_total

thespitz, 2020-04-06:
After updating weewxd, rtl_433 and weewx-sdr for the first time in ~6mo, weewx-sdr no longer appears to be parsing rain_total.  But it does show in rtl_433 as rain_in.

$ PYTHONPATH=/usr/share/weewx python /usr/share/weewx/user/sdr.py --cmd="rtl_433 -M utc -F json -R 40"

```
out:[u'{"time" : "2020-04-06 20:46:07", "model" : "Acurite-5n1", "subtype" : 49, "id" : 3287, "channel" : "B", "sequence_num" : 0, "battery_ok" : 1, "wind_avg_km_h" : 0.000, "wind_dir_deg" : 67.500, "rain_in" : 14.130, "mic" : "CHECKSUM"}\n', u'{"time" : "2020-04-06 20:46:07", "model" : "Acurite-5n1", "subtype" : 49, "id" : 3287, "channel" : "B", "sequence_num" : 1, "battery_ok" : 1, "wind_avg_km_h" : 0.000, "wind_dir_deg" : 67.500, "rain_in" : 14.130, "mic" : "CHECKSUM"}\n', u'{"time" : "2020-04-06 20:46:07", "model" : "Acurite-5n1", "subtype" : 49, "id" : 3287, "channel" : "B", "sequence_num" : 2, "battery_ok" : 1, "wind_avg_km_h" : 0.000, "wind_dir_deg" : 67.500, "rain_in" : 14.130, "mic" : "CHECKSUM"}\n']
parsed: {'wind_dir.0CD7.Acurite5n1PacketV2': 67.5, 'channel.0CD7.Acurite5n1PacketV2': u'B', 'snr.0CD7.Acurite5n1PacketV2': None, 'model.0CD7.Acurite5n1PacketV2': u'Acurite-5n1', 'wind_speed.0CD7.Acurite5n1PacketV2': 0.0, 'msg_type.0CD7.Acurite5n1PacketV2': 49, 'noise.0CD7.Acurite5n1PacketV2': None, 'rssi.0CD7.Acurite5n1PacketV2': None, 'dateTime': 1586205967, 'mod.0CD7.Acurite5n1PacketV2': None, 'sequence_num.0CD7.Acurite5n1PacketV2': 0, 'battery.0CD7.Acurite5n1PacketV2': 1, 'protocol.0CD7.Acurite5n1PacketV2': None, 'usUnits': 1, 'freq.0CD7.Acurite5n1PacketV2': None}
```

$ rtl_433 -M utc -F json -R 40
`{"time" : "2020-04-06 20:57:59", "model" : "Acurite-5n1", "subtype" : 49, "id" : 3287, "channel" : "B", "sequence_num" : 0, "battery_ok" : 1, "wind_avg_km_h" : 0.000, "wind_dir_deg" : 67.500, "rain_in" : 14.130, "mic" : "CHECKSUM"}`

Thank you!

thespitz, 2020-04-06:
Looks like the parser was only looking for rain_mm.  Adding this to Acurite5n1PacketV2 seemed to do the trick

`        if 'rain_in' in obj:
            pkt['rain_total'] = Packet.get_float(obj, 'rain_in')
`

MksRasp, 2020-04-21:
thank you thespitz
I followed your instruction and fixed my sdr issue also the acurite lightning 6045M have similar issue and fixed it by changing the identifier from "Acurite Lightning 6045M" to "Acurite-6045M"
and removed this line "        pkt['ussb1'] = obj.get('ussb1')" which is not needed after the latest change of rtl_433

bdwilson, 2020-05-21:
Fixed the issue for me as well.

bdwilson, 2020-07-24:
Did this change really fix things for you all? I've been noticing that the amount of rain reported by weewx is much less than AcuRite directly. I haven't been able to determine the issue. Off by .30 tonight seems kind of a big deal. 

[image]
[image]
[image]