matthewwall/weewx-sdr #81: Acurite 6045 decoding changes

rct, 2020-04-13:
FYI -

As April 13, the decoding for the Acurite 6045 lightning detector changed via [PR #1348](https://github.com/merbanan/rtl_433/pull/1348)

* The strike counter is now 8 bits instead of 7.
* The `ussb1` field has been removed (it was the missing bit).

This change will affect consumers when you start running the new code.  The wrap now occurs at 255 instead of 127.  

Be aware there will likely be a big change in your data when you run the new code  The value is now shifted left by one bit.  So if the old code was decoding the counter as 50, the new code will give you a new count of 100. So your consumer would register 50 new strikes.  

Here's the new message
```
{"time" : "2020-04-13 12:40:52.578064", "model" : "Acurite-6045M", "id" : 213, "channel" : "A", "battery_ok" : 1, "temperature_F" : 53.600, "humidity" : 96, "strike_count" : 100, "storm_dist" : 17, "active" : 0, "rfi" : 0, "exception" : 0, "raw_msg" : "c0d56f600f74b211aa", "mod" : "ASK", "freq" : 433.956, "rssi" : -0.095, "snr" : 11.631, "noise" : -11.726}
```

n7qnm, 2020-07-12:
The identifier of the probe has also changed to "Acurite-6045M'

andylittle, 2020-10-20:
replace your sdr.py file in bin/user/ of your weewx install dir with [this](https://github.com/andylittle/weewx-sdr/tree/master/bin/user) one. Then restart weewx.