matthewwall/weewx-sdr #83: Atlas & 6045 Lightning strike / Storm Distance Calculations

ChrisM510, 2020-04-24:
Has anyone managed to define in SDY.PY or WEEWX.CONFIG how to correctly record and display the strike distance/storm distance data from either the Atlas or 6045?  Because both devices continually transmit the last strike/storm distance, I am having trouble trying to figure out how to get WeeWX to calculate what the true distance is, i.e. display (0) when there are no lightning strikes, but report the distance when there IS a strike.  (My apologies if I am not using this forum/page correctly - I am really a newbie to so much of this still.)

andylittle, 2020-04-25:
This is more for bugs, try the [forum.](https://groups.google.com/forum/m/#!search/Acurite$206045/weewx-user/xaRRCx1QrgU)

ChrisM510, 2020-04-25:
Ok, thank you!

closed 2020-04-25 by ChrisM510