matthewwall/weewx-sdr #94: Oregon Sensor Update

luchmhor, 2020-08-20:
Hello There.

It seems that I've got a version of the WGR800 which is slightly different from the one before:

```{"time" : "2020-08-19 19:31:12", "brand" : "OS", "model" : "Oregon-WGR800", "id" : 177, "channel" : 0, "battery_ok" : 1, "wind_max_m_s" : 2.300, "wind_avg_m_s" : 2.700, "wind_dir_deg" : 315.000}```

below is the output for the rain sensor:

```{"time" : "2020-08-19 19:31:13", "brand" : "OS", "model" : "Oregon-PCR800", "id" : 80, "channel" : 0, "battery_ok" : 1, "rain_rate_in_h" : 0.000, "rain_in" : 27.741}```

andylittle, 2020-10-20:
replace your sdr.py file in bin/user/ of your weewx install dir with [this](https://github.com/andylittle/weewx-sdr/tree/master/bin/user) one. Then restart weewx.

Lester-Burnham, 2020-11-30:
Hi,
I'm posting to this same thread due to having a similar problem with the same OS devices from a WMR200. This is running on a Raspberry Pi  with SDR and rtl_433 installed from git, but using weewx installed from debian buster repos.
I'm currently running the version of sdr.py posted above. The only difference I see in the output now is complaining about rain being outside limits.

 WARNING weewx.qc: 2020-11-30 11:55:47 AEDT (1606697747) LOOP value 'rain' 10.7877 outside limits (0.0, 10.0)

SDR section of weewx.conf
------------------------------------------
[SDR]
    # This section is for the software-defined radio driver.
    
    # The driver to use
    driver = user.sdr
    log_unknown_sensors = False
    log_unmapped_sensors = False
    path = /usr/local/bin
    ld_library_path = /usr/local/lib

	[[sensor_map]]
	    outTemp = temperature.1:229.OSTHGR810Packet
	    outHumidity = humidity.1:229.OSTHGR810Packet
	    windDir = wind_dir.0:147.OSWGR800Packet
	    windSpeed = wind_speed.0:147.OSWGR800Packet
	    windGust = wind_gust.0:147.OSWGR800Packet
	    rainRate = rain_rate.0:74.OSPCR800Packet
	    rain = rain_total.0:74.OSPCR800Packet
	    UV = uv_index.1:70.OSUV800Packet
	    outTempBatteryStatus = battery.1:229.OSTHGR810Packet
	    windBatteryStatus = battery.0:147.OSWGR800Packet
	    rainBatteryStatus = battery.0:74.OSPCR800Packet
	    uvBatteryStatus = battery.1:70.OSUV800Packet
--------------------------------------------------------

This is the output of my sensors from the previous version of the file when setting up the SDR section of the weewx.conf
{"time" : "2020-11-27 01:03:02", "brand" : "OS", "model" : "Oregon-UV800", "id" : 70, "channel" : 1, "battery_ok" : 1, "uv" : 10}
{"time" : "2020-11-27 01:03:38", "brand" : "OS", "model" : "Oregon-PCR800", "id" : 74, "channel" : 0, "battery_ok" : 1, "rain_rate_in_h" : 0.000, "rain_in" : 50.214}
{"time" : "2020-11-27 01:03:39", "brand" : "OS", "model" : "Oregon-WGR800", "id" : 147, "channel" : 0, "battery_ok" : 1, "wind_max_m_s" : 2.600, "wind_avg_m_s" : 2.500, "wind_dir_deg" : 337.500}
{"time" : "2020-11-27 01:03:53", "brand" : "OS", "model" : "Oregon-THGR810", "id" : 229, "channel" : 1, "battery_ok" : 1, "temperature_C" : 23.400, "humidity" : 59}

Nov 27 21:31:37 weewx weewx[2538] INFO user.sdr: unmapped: ['{"time" : "2020-11-27 10:31:33", "brand" : "OS", "model" : "Oregon-WGR800", "id" : 147, "channel" : 0, "battery_ok" : 1, "wind_max_m_s" : 1.700, "wind_avg_m_s" : 1.500, "wind_dir_deg" : 202.500}\n'] ({'dateTime': 1606473093, 'usUnits': 1, 'battery.0:74.OSPCR800Packet': 1, 'rain_rate.0:74.OSPCR800Packet': None, 'rain_total.0:74.OSPCR800Packet': None})

 ({'dateTime': 1606473093, 'usUnits': 17, 'battery.0:147.OSWGR800Packet': 1, 'wind_gust.0:147.OSWGR800Packet': None, 'wind_speed.0:147.OSWGR800Packet': None, 'wind_dir.0:147.OSWGR800Packet': None})

Nov 27 21:31:51 weewx weewx[2538] INFO user.sdr: unmapped: [] ({'dateTime': 1606473106, 'usUnits': 17, 'battery.0:147.OSWGR800Packet': 1, 'wind_gust.0:147.OSWGR800Packet': None, 'wind_speed.0:147.OSWGR800Packet': None, 'wind_dir.0:147.OSWGR800Packet': None})

Nov 27 21:31:55 weewx weewx[2538] INFO user.sdr: unmapped: [] ({'dateTime': 1606473111, 'usUnits': 16, 'battery.1:229.OSTHGR810Packet': 1, 'temperature.1:229.OSTHGR810Packet': 20.2, 'humidity.1:229.OSTHGR810Packet': 66.0})

Nov 27 21:32:19 weewx weewx[2538] INFO user.sdr: unmapped: [] ({'dateTime': 1606473135, 'usUnits': 17, 'battery.0:147.OSWGR800Packet': 1, 'wind_gust.0:147.OSWGR800Packet': None, 'wind_speed.0:147.OSWGR800Packet': None, 'wind_dir.0:147.OSWGR800Packet': None})

Nov 27 21:32:48 weewx weewx[2538] INFO user.sdr: unmapped: [] ({'dateTime': 1606473165, 'usUnits': 16, 'battery.1:229.OSTHGR810Packet': 1, 'temperature.1:229.OSTHGR810Packet': 20.2, 'humidity.1:229.OSTHGR810Packet': 66.0})

Nov 27 21:33:42 weewx weewx[2538] INFO user.sdr: unmapped: ['{"time" : "2020-11-27 10:33:39", "brand" : "OS", "model" : "Oregon-UV800", "id" : 70, "channel" : 1, "battery_ok" : 1, "uv" : 0}\n'] ({'dateTime': 1606473218, 'usUnits': 16, 'battery.1:229.OSTHGR810Packet': 1, 'temperature.1:229.OSTHGR810Packet': 20.2, 'humidity.1:229.OSTHGR810Packet': 66.0})

Nov 27 21:33:42 weewx weewx[2538] INFO user.sdr: unmapped: [] ({'dateTime': 1606473219, 'usUnits': 16, 'battery.1:70.OSUV800Packet': 1, 'uv_index.1:70.OSUV800Packet': 0.0})

-----------------------------

pi@weewx:/etc/weewx $ sudo weewxd /etc/weewx/weewx.conf
LOOP:   2020-11-30 10:48:58 AEDT (1606693738) dateTime: 1606693738, maxSolarRad: 884.1650489526253, usUnits: 1, windBatteryStatus: 1, windDir: None, windGust: None, windGustDir: None, windSpeed: None
LOOP:   2020-11-30 10:49:11 AEDT (1606693751) dateTime: 1606693751, maxSolarRad: 884.7100631503614, usUnits: 1, UV: 13.0, uvBatteryStatus: 1
LOOP:   2020-11-30 10:49:12 AEDT (1606693752) dateTime: 1606693752, maxSolarRad: 884.7519389148077, rain: None, rainBatteryStatus: 1, rainRate: None, usUnits: 1
LOOP:   2020-11-30 10:49:28 AEDT (1606693768) dateTime: 1606693768, maxSolarRad: 885.4216214900965, usUnits: 1, windBatteryStatus: 1, windDir: None, windGust: None, windGustDir: None, windSpeed: None
LOOP:   2020-11-30 10:49:56 AEDT (1606693796) dateTime: 1606693796, maxSolarRad: 886.5910176335652, usUnits: 1, windBatteryStatus: 1, windDir: None, windGust: None, windGustDir: None, windSpeed: None
[... cut ...]
output was:
Traceback (most recent call last):
  File "/usr/share/weewx/user/sdr.py", line 92, in <module>
    import weewx.units
  File "/usr/share/weewx/weewx/units.py", line 23, in <module>
    import weeutil.weeutil
  File "/usr/share/weewx/weeutil/weeutil.py", line 26, in <module>
    from weeutil.config import accumulateLeaves, search_up
  File "/usr/share/weewx/weeutil/config.py", line 11, in <module>
    import configobj
ImportError: No module named configobj

Much appreciated if you can help.