matthewwall/weewx-sdr #96: Bresser 5in1 under-reporting accumulated rain

ComradeTurtle, 2020-10-08:
I have been using weewx-sdr for quite a while now and I have noticed that **weewx-sdr under reports accumulated rain** by around 50%, sometimes way more. Other times **rain is not reported at all.**

I have checked - **rtl_433's total rain data is indeed correct.** The problem is somewhere between weewx and weewx_sdr. I am happy to provide further information if needed. TIA!

Weewx sensor map:
```
[SDR]
    driver = user.sdr
    cmd = /usr/local/bin/rtl_433 -M utc -F json -f 868.2M -R 119
    [[sensor_map]]
        windDir = wind_dir.8.Bresser5in1Packet
        windSpeed = wind_speed.8.Bresser5in1Packet
        windGust = gust_speed.8.Bresser5in1Packet
        outTemp = temperature.8.Bresser5in1Packet
        outHumidity = humidity.8.Bresser5in1Packet
        rain_total = rain_total.8.Bresser5in1Packet
        [[deltas]]
        rain = rain_total
```

andylittle, 2020-11-04:
Is the rain value from the sdr driver in mm?

ComradeTurtle, 2020-11-05:
Yes, the SDR returns the total rain the station has measured in mm.

stefkal7, 2020-12-07:
I have the same issue ..My station is Breeser 5in1 7002580.... I am using rpi4 with weewx 4.2.0 and noolec sdr .... The sdr delivered to weewx the rain_total packet but weewx in my case doesn't make the deltas calculations so the rain is always 0.00mm ... I have tried almost everything in weewx.conf (because i think that something have to trigger weewx to calculate the rain value with rain_total packet ) but no rain ... Only pain ... Both sdr.py and weewx.conf are set up correctly (i think) and i have in the [sdr] field
[[deltas ]]
rain = rain_total 
Thank you in advanced if any help provided

stefkal7, 2020-12-14:
[SOLVED] 
For Bresser 5in1 7002580 i leave it here if anyone needs it....
The Weewx conf file should be like 

[[sensor_map]]        
    windDir = wind_dir.411043787.Bresser5in1Packet
    windSpeed = wind_speed.411043787.Bresser5in1Packet
    windGust = wind_gust.411043787.Bresser5in1Packet
    outTemp = temperature.411043787.Bresser5in1Packet
    outHumidity = humidity.411043787.Bresser5in1Packet
    barometer = pressure.411043787.Bresser5in1Packet         
    rain_total = rain_total.411043787.Bresser5in1Packet 
 [[deltas]]
    rain = rain_total

and the sdr.py : 

class Bresser5in1Packet(Packet):
# {"time" : "2020-11-21 14:17:00", 
# "model" : "Bresser-6in1", 
# "id" : 411043787, 
# "channel" : 0, 
# "battery_ok" : 1, 
# "sensor_type" : 1, 
# "wind_max_m_s" : 2.000, 
# "wind_avg_m_s" : 1.700, 
# "wind_dir_deg" : 68, 
# "rain_mm" : 11.600, 
# "mic" : "CRC"}

    IDENTIFIER = "Bresser-6in1"

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        baro = open('/home/pi/baro/baro.txt').read()               
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRICWX
        pkt['station_id'] = obj.get('id')        
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')           
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        pkt['wind_dir'] = Packet.get_float(obj, 'wind_dir_deg')
        pkt['wind_speed'] = Packet.get_float(obj, 'wind_avg_m_s')
        pkt['wind_gust'] = Packet.get_float(obj, 'wind_max_m_s')                
        pkt['pressure'] = float(baro) 
        rain_total = Packet.get_float(obj, 'rain_mm')
        if rain_total is not None:
            pkt['rain_total'] = rain_total      
        return Bresser5in1Packet.insert_ids(pkt)

    @staticmethod
    def insert_ids(pkt):
        station_id = pkt.pop('station_id', '0000')
        pkt = Packet.add_identifiers(pkt, station_id, Bresser5in1Packet.__name__)
        return pkt

Greetings from Greece ..!!

ComradeTurtle, 2020-12-15:
The problem has been solved for me too; I'll go ahead and close the issue.

closed 2020-12-15 by ComradeTurtle