matthewwall/weewx-sdr #97: parser for "Fineoffset-WHx080"?

W11T, 2020-10-13:
Could somebody kindly write a parser for "Fineoffset-WHx080"?

Example:
{"time" : "2020-10-13 17:29:10", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 228, "battery_ok" : 1, "temperature_C" : 9.600, "humidity" : 82, "wind_dir_deg" : 0, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "rain_mm" : 1011.000, "mic" : "CRC"}

This is a WS1080 but the format is different from "Fine Offset WH1080 weather station" and "Fine Offset Electronics WH1080".

andylittle, 2020-10-13:
I can take a look at this one.

andylittle, 2020-10-13:
new and old messages

`{"time" : "2020-10-13 14:04:48", "model" : "Fine Offset Electronics WH1080/WH3080 Weather Station", "msg_type" : 0, "id" : 14, "battery" : "OK", "temperature_C" : 24.400, "humidity" : 35, "direction_deg" : 225, "speed" : 0.000, "gust" : 0.000, "rain" : 41.400, "mic" : "CRC"}`

`{"time" : "2020-10-13 14:04:48", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 14, "battery_ok" : 1, "temperature_C" : 24.400, "humidity" : 35, "wind_dir_deg" : 225, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "rain_mm" : 41.400, "mic" : "CRC"}`

andylittle, 2020-10-13:
@W11T replace your sdr.py file in bin/user/ of your weewx install dir with [this](https://github.com/andylittle/weewx-sdr/blob/master/bin/user/sdr.py) one. Then restart weewx.

```
out:['{"time" : "2020-10-13 21:51:12", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 14, "battery_ok" : 1, "temperature_C" : 25.000, "humidity" : 35, "wind_dir_deg" : 225, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 0.000, "rain_mm" : 41.400, "mic" : "CRC"}\n']
parsed: {'dateTime': 1602625872, 'usUnits': 16, 'msg_type.14.FOWHx080Packet': 0, 'temperature.14.FOWHx080Packet': 25.0, 'humidity.14.FOWHx080Packet': 35.0, 'wind_dir.14.FOWHx080Packet': None, 'wind_speed.14.FOWHx080Packet': 0.0, 'wind_gust.14.FOWHx080Packet': 0.0, 'rain_total.14.FOWHx080Packet': 4.14, 'battery.14.FOWHx080Packet': 1, 'signal_type.14.FOWHx080Packet': 0, 'hours.14.FOWHx080Packet': None, 'minutes.14.FOWHx080Packet': None, 'seconds.14.FOWHx080Packet': None, 'year.14.FOWHx080Packet': None, 'month.14.FOWHx080Packet': None, 'day.14.FOWHx080Packet': None}
```
Andy

W11T, 2020-10-14:
@andylittle Thanks for your immediate support!

W11T, 2020-10-17:
@andylittle Thanks for your help

closed 2020-10-17 by W11T