matthewwall/weewx-sdr #98: timeouts

W11T, 2020-10-16:
user.sdr suffers timeouts possibly as soon as results are being returned by rtl_433 for the first time.
At my site rtl_433 delivers results each 30 seconds or more.

Oct 16 22:22:12 <host> weewx[15688] INFO weewx.engine: Loading station type SDR (user.sdr)
Oct 16 22:22:12 <host> weewx[15688] INFO user.sdr: driver version is 0.77.1
Oct 16 22:22:12 <host> weewx[15688] INFO user.sdr: sensor map is {'outHumidity': 'humidity.228.FOWHx080Packet', 'outTemp': 'temperature.228.FOWHx080Packet', 'dateTime': "dateTime': 1602800188", 'windDir': 'wind_dir.228.FOWHx080Packet', 'rainTotal': 'rain_total.228.FOWHx080Packet', 'windSpeed': 'wind_speed.228.FOWHx080Packet', 'windGust': 'wind_gust.228.FOWHx080Packet', 'usUnits': ["usUnits': 16"], 'outTempBatteryStatus': 'battery.228.FOWHx080Packet'}
Oct 16 22:22:12 <host> weewx[15688] INFO user.sdr: deltas is {'rain': 'rain_total', 'strikes': 'strikes_total'}
Oct 16 22:22:12 <host> weewx[15688] INFO user.sdr: startup process '/usr/local/bin/rtl_433 -f 868300000 -s 250000 -R 32 -g 49.6 -M time -F json'
Oct 16 22:22:12 <host> weewx[15688] INFO weewx.engine: StdConvert target unit is 0x11
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.wxservices: The following values will be calculated: pressure=prefer_hardware, altimeter=prefer_hardware, appTemp=prefer_hardware, barometer=prefer_hardware, beaufort=prefer_hardware, cloudbase=prefer_hardware, dewpoint=prefer_hardware, ET=prefer_hardware, heatindex=prefer_hardware, humidex=prefer_hardware, inDewpoint=prefer_hardware, maxSolarRad=prefer_hardware, rainRate=prefer_hardware, windchill=prefer_hardware, windrun=prefer_hardware
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.wxservices: The following algorithms will be used for calculations: altimeter=aaASOS, maxSolarRad=RS
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.engine: Archive will use data binding wx_binding
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.engine: Record generation will be attempted in 'software'
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.engine: Using archive interval of 300 seconds (software record generation)
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.restx: StationRegistry: Registration not requested.
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.restx: Wunderground: Posting not enabled.
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.restx: PWSweather: Posting not enabled.
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.restx: CWOP: Posting not enabled.
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.restx: WOW: Posting not enabled.
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.restx: AWEKAS: Posting not enabled.
Oct 16 22:22:13 <host> weewx[15688] INFO __main__: Starting up weewx version 4.1.1
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.engine: Using binding 'wx_binding' to database 'weewx'
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.manager: Starting backfill of daily summaries
Oct 16 22:22:13 <host> weewx[15688] INFO weewx.engine: Starting main packet loop.
Oct 16 22:22:59 <host> weewx[15688] INFO weewx.engine: Main loop exiting. Shutting engine down.
Oct 16 22:22:59 <host> weewx[15688] INFO user.sdr: shutdown process /usr/local/bin/rtl_433 -f 868300000 -s 250000 -R 32 -g 49.6 -M time -F json
Oct 16 22:23:09 <host> weewx[15688] INFO user.sdr: timed out waiting for stdout-thread
Oct 16 22:23:19 <host> weewx[15688] INFO user.sdr: timed out waiting for stderr-thread
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__: Caught unrecoverable exception:
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****  unhashable type: 'list'
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****  Traceback (most recent call last):
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****    File "<weewx_root>/bin/weewxd", line 154, in main
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****      engine.run()
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****    File "<weewx_root>/bin/weewx/engine.py", line 188, in run
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****      for packet in self.console.genLoopPackets():
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****    File "<weewx_root>/bin/user/sdr.py", line 2693, in genLoopPackets
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****      pkt = self.map_to_fields(packet, self._sensor_map)
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****    File "<weewx_root>/bin/user/sdr.py", line 2738, in map_to_fields
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****      label = SDRDriver._find_match(sensor_map[n], pkt.keys())
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****    File "<weewx_root>/bin/user/sdr.py", line 2752, in _find_match
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****      if pattern in keylist:
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****  TypeError: unhashable type: 'list'
Oct 16 22:24:33 <host> weewx[15688] CRITICAL __main__:     ****  Exiting.

andylittle, 2020-10-19:
@W11T can you post your sensor map from weewx.conf please. Remove usUnits and dateTime from your map if it is there.

Should match this except your id is 228 instead of 14.
 
`        outHumidity = humidity.14.FOWHx080Packet
        outTemp = temperature.14.FOWHx080Packet
        windDir = wind_dir.14.FOWHx080Packet
        rainTotal = rain_total.14.FOWHx080Packet
        windSpeed = wind_speed.14.FOWHx080Packet
        windGust = wind_gust.14.FOWHx080Packet
        windTempBatteryStatus = battery.14.FOWHx080Packet`

W11T, 2020-10-20:
@andylittle Thanks, you pushed me in the right direction! I had typos in my sensor map!
This is my sensor map now:

`    log_unknown_sensors = True
    log_unmapped_sensors = True
    [[sensor_map]]
        outHumidity = humidity.201.FOWHx080Packet
        outTemp = temperature.201.FOWHx080Packet
        windDir = wind_dir.201.FOWHx080Packet
        rainTotal = rain_total.201.FOWHx080Packet
        windSpeed = wind_speed.201.FOWHx080Packet
        windGust = wind_gust.201.FOWHx080Packet
        outTempBatteryStatus = battery.201.FOWHx080Packet
`
(I restarted my station and its id changed to 201)

rtl_433 delivers records like this:

`{"time" : "2020-10-20 21:17:17", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 201, "battery_ok" : 1, "temperature_C" : 11.500, "humidity" : 98, "wind_dir_deg" : 0, "wind_avg_km_h" : 0.000, "wind_max_km_h" : 1.224, "rain_mm" : 3.900, "mic" : "CRC"}`

But the don't show up in weewx.

W11T, 2020-10-20:
@andylittle Thanks, you made my day! I had typos in my sensor map!

closed 2020-10-20 by W11T