matthewwall/weewx-sdr #117: inFactory parser

gleven, 2021-03-03:
I have a temp/humidity sensor that even though it's picked up , it's not parsed.
Not sure if it's a parser issue or it's coming from rtl_433.
Here is the output

`out:[u'{"time" : "2021-03-03 10:19:53", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 73.200, "humidity" : 55, "mic" : "CRC"}\n', u'{"time" : "2021-03-03 10:19:54", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 73.200, "humidity" : 55, "mic" : "CRC"}\n', u'{"time" : "2021-03-03 10:19:54", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 73.200, "humidity" : 55, "mic" : "CRC"}\n', u'{"time" : "2021-03-03 10:19:54", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 73.200, "humidity" : 55, "mic" : "CRC"}\n', u'{"time" : "2021-03-03 10:19:54", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 73.200, "humidity" : 55, "mic" : "CRC"}\n', u'{"time" : "2021-03-03 10:19:54", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 73.200, "humidity" : 55, "mic" : "CRC"}\n']`

When i removed the batteries and reinserted them, the id changed. It used to be 5 . Now it's 195.

RaSTuS26, 2021-03-03:
You can use a wildcard ("*") for the 'id' field so it matches all id's, not sure if this is helpful for you but could help:

From my own weewx.conf:
```
outTemp = temperature.*.FOWHx080Packet
```

gleven, 2021-03-03:
> You can use a wildcard ("*") for the 'id' field so it matches all id's, not sure if this is helpful for you but could help:
> 
> From my own weewx.conf:
> 
> ```
> outTemp = temperature.*.FOWHx080Packet
> ```

Well, this didn't work.
I'm getting this output in weewx
`Mar  3 12:51:01 weewx-sdr weewx[5588] DEBUG user.sdr: parse_json: unknown model inFactory-TH
Mar  3 12:51:01 weewx-sdr weewx[5588] DEBUG user.sdr: punt unrecognized line '{"time" : "2021-03-03 10:50:56", "model" : "inFactory-TH", "id" : 195, "channel" : 1, "battery_ok" : 1, "temperature_F" : 72.000, "humidity" : 53, "mic" : "CRC"}#012'`

RaSTuS26, 2021-03-03:
I'm by no means an expert in this, just thought I could help with the changing 'id' issue.

That log error is not much use without context, can you show your "[SDR]" section from weewx.conf to show your setup generating this log output?

gleven, 2021-03-03:
> I'm by no means an expert in this, just thought I could help with the changing 'id' issue.
> 
> That log error is not much use without context, can you show your "[SDR]" section from weewx.conf to show your setup generating this log output?

Here's the SDR section. I have an acurite 5-in-1 and wanted to add the additional sensor. I currenlty have it under inTemp. I have played with various versions of this but nothing changes.

`[SDR]
    # This section is for the software-defined radio driver.

    # The driver to use
    driver = user.sdr
    cmd = /usr/local/bin/rtl_433 -M utc -F json
    [[sensor_map]]
        windDir = wind_dir.051F.Acurite5n1PacketV2
        windSpeed = wind_speed.051F.Acurite5n1PacketV2
        outTemp = temperature.051F.Acurite5n1PacketV2
        outHumidity = humidity.051F.Acurite5n1PacketV2
        rain_total = rain_total.051F.Acurite5n1PacketV2
        txBatteryStatus = battery.051F.Acurite5n1PacketV2
        inTemp = temperature.*.inFactory-TH
        #inHumidity = humidity.3969.AcuriteTowerPacket
    # I have more sensors, so I added them below, you may not.
    #extraTemp1 = temperature_probe.0BF1.Acurite00275MPacket
    #extraTemp2 = temperature.0BF1.Acurite00275MPacket
    [[deltas]]
        rain = rain_total`

RaSTuS26, 2021-03-03:
I can't find a class that decodes "inFactory-TH" anywhere in "/usr/share/weewx/user/sdr.py", maybe it's not supported weewx-sdr even though it seems to be in rtl_433 as device 91.

https://github.com/merbanan/rtl_433/blob/master/README.md

gleven, 2021-03-03:
I guess i need a parser for this. Can someone help with this?

RaSTuS26, 2021-03-03:
Beyond me I'm afraid, good luck with it @gleven.

gleven, 2021-03-03:
I managed it. Wrote my own parser and now it's working fine

closed 2021-03-03 by gleven