matthewwall/weewx-sdr #118: Oregon-BHTR968

linux150, 2021-03-05:
Hi
Sorry my first post here on github.
I modified my sdr.py to get all sensor data exept the BHTR968. 
Sorry difficult to explain, but i dont get the "Packet line" from this sensor. 
The data from the syslog `Mar  5 17:24:54 weewx weewx[4380] DEBUG user.sdr: lines=['{"time" : "2021-03-05 16:24:50", "brand" : "OS", "model" : "Oregon-BHTR968", "id" : 23, "channel" : 0, "battery_ok" : 1, "temperature_C" : 23.100, "humidity" : 31, "pressure_hPa" : 915.000}\n', 
'{"time" : "2021-03-05 16:24:50", "brand" : "OS", "model" : "Oregon-BHTR968", "id" : 23, "channel" : 0, "battery_ok" : 1, "temperature_C" : 23.100, "humidity" : 31, "pressure_hPa" : 915.000}\n']
Mar  5 17:24:57 weewx weewx[4380] DEBUG user.sdr: lines=[]
Mar  5 17:25:00 weewx weewx[4380] DEBUG user.sdr: lines=[]
`
And the lines in the sdr.py
`# apparently rtl_433 uses BHTR968 when it should be BTHR968
class OSBTHR968Packet(Packet):
    # Added 2017-04-22 ALG
    # 2017-09-12 21:44:55     :       OS :    BHTR968
    # House Code:      111
    # Channel:         0
    # Battery:         OK
    # Celcius:         26.20 C
    # Fahrenheit:      79.16 F
    # Humidity:        36 %
    # Pressure:        1012 mbar

    IDENTIFIER = "BHTR968"
    PARSEINFO = {
        'House Code': ['house_code', None, lambda x: int(x)],
        'Channel': ['channel', None, lambda x: int(x)],
        'Battery': ['battery', None, lambda x: 0 if x == 'OK' else 1],
        'Temperature': ['temperature', re.compile('([\d.-]+) C'), lambda x: float(x)],
        'Humidity': ['humidity', re.compile('([\d.]+) %'), lambda x: float(x)],
        'Pressure': ['pressure', re.compile('([\d.]+) mbar'), lambda x: float(x)]}

    @staticmethod
    def parse_text(ts, payload, lines):
        pkt = dict()
        pkt['dateTime'] = ts
        pkt['usUnits'] = weewx.METRIC
        pkt.update(Packet.parse_lines(lines, OSBTHR968Packet.PARSEINFO))
        return OS.insert_ids(pkt, OSBTHR968Packet.__name__)

    # original rtl_433 output
    # {"time" : "2017-01-18 14:56:03", "brand" : "OS", "model" :"BHTR968", "id" : 111, "channel" : 0, "battery" : "OK", "temperature_C" : 27.200, "temperature_F" : 80.960,  "humidity" : 46, "pressure" : 1013}
    # by 06mar2019
    # {"time" : "2019-03-06 13:27:23", "brand" : "OS", "model" : "BHTR968", "id" : 179, "channel" : 0, "battery" : "LOW", "temperature_C" : 19.800, "humidity" : 54, "pressure_hPa" : 974.000}

    @staticmethod
    def parse_json(obj):
        pkt = dict()
        pkt['dateTime'] = Packet.parse_time(obj.get('time'))
        pkt['usUnits'] = weewx.METRIC
        pkt['house_code'] = obj.get('id')
        pkt['channel'] = obj.get('channel')
        pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
        pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
        pkt['humidity'] = Packet.get_float(obj, 'humidity')
        if 'pressure' in obj:
            pkt['pressure'] = Packet.get_float(obj, 'pressure_hPa')
        elif 'pressure_hPa' in obj:
            pkt['pressure'] = Packet.get_float(obj, 'pressure_hPa')
        return OS.insert_ids(pkt, OSBTHR968Packet.__name__)
`

Any help ?
lg
linux150

andylittle, 2021-03-05:
#change the identifier
IDENTIFIER = "BHTR968" should be Oregon-BHTR968
#fix battery_ok
 pkt['battery'] = 0 if obj.get('battery_ok') == 1 else 1

There may be more, I just looked quick

Andy

linux150, 2021-03-05:
tnx, i did change but no solution. 
The weewx installation is running for testing and backup in the moment. But when i configured to start, the sensor already worked. I did check the logs and he stopped working on 15.2. ? Maybe after changing batterys ? Or any update on rtl.
lg
linux150

andylittle, 2021-03-05:
id" : 23

id probably changes on power cycle, is id hard coded in your map?

linux150, 2021-03-05:
Sorry for newbie :-)
I´m not sure, where can i check this ?
LG

RaSTuS26, 2021-03-05:
You can use a wildcard '*' in the 'id' field, so for example the following would work for any 'id'.

```
[[sensor_map]]
pressure = pressure.0:*.OSBTHR968Packet
.....
```

linux150, 2021-03-05:
Wow, great support. Tnx all working fine again !
Always difficult to start something new, config all until it runs and then after month of stable running, to starts from beginning.
So thanks for your help, issue closed.
linux150

closed 2021-03-05 by linux150