matthewwall/weewx-sdr #127: Acurite Atlas - getting phantom rain data in weewx database

mrneutron42, 2021-04-29:
This week, I got my Acurite Atlas data coming into weewx for the first time.  I'm using rtl_433 and weewx-sdr. 
The Atlas is inside my house while I set up weewx and test all the pieces of the system, so there is no rain entering the unit.

To test the rain counter in the Atlas, I tipped its teeder-todder back and forth a few times and that did register a fraction of an inch of rain.  It amounted to about 0.17 inches of rain.  Which I saw reported in the data that is produced from running sdr.py:
sudo PYTHONPATH=/home/weewx/bin python3 /home/weewx/bin/user/sdr.py --cmd="rtl_433 -R 40 -M utc -F json"

But, I am seeing large amounts of rain being registered in the weewx webpage, and can't understand how!
[image]
[image]
When I look into the /home/weewx/bin/weewx.sdb file, I see rain and rain rate numbers!  How?
[image]
The [sensor_map] section of weewx.conf is sending the 

...still adding info

andylittle, 2021-04-29:
I think this my fix it, you will have to delete some erroneous rain data from your database. Look at sdr.py around line 32 for the explanation:
`[SDR]
    driver = user.sdr
    [[sensor_map]]
        inTemp = temperature.25A6.AcuriteTowerPacket
        outTemp = temperature.24A4.AcuriteTowerPacket
        rain_total = rain_total.A52B.Acurite5n1Packet

If no sensor_map is specified, no data will be collected.

The deltas stanza indicates which observations are cumulative measures and
how they should be split into delta measures.

[SDR]
    ...
    [[deltas]]
        rain = rain_total
`

change to rain_total = rain_total.0011.AcuriteAtlasPacket.

mrneutron42, 2021-04-29:
Thanks for the pointer!
Do I need to do this with lightning_strike_count also?
From what I've seen of the raw data coming from the Atlas, that lightning_strike_count number never resets to 0, either.

andylittle, 2021-04-29:
1. I need to revisit this one, mine is broke.
...
        lightning_distance = distance.3C47.AcuriteLightningPacket
        strikes_total = strikes_total.3C47.AcuriteLightningPacket
    [[deltas]]
        lightning_strike_count = strikes_total
'''

2. I use the sqlite comand line.
3. you can delete the db file and on a restart a new db will be created.

mrneutron42, 2021-04-29:
How do you identify the data/time from that processed date/time value? ('dateTime': 1619553347)
I can't tell what that number means.

mrneutron42, 2021-04-29:
Adding the [[deltas]] section fixed the phantom rain problem!  Thanks!
[image]

closed 2021-04-29 by mrneutron42