matthewwall/weewx-sdr #132: Fineoffset-WHx080

mfraser, 2021-06-15:
Only installed weewx-sdr and weewx yesterday using rtl_433 version 21.05-14-g1a9b05cb and it doesn't seem to be recognising my weather station.

Jun 15 18:31:18 weathercam weewx[860] DEBUG user.sdr: lines=['{"time" : "2021-06-15 17:31:15", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 168, "battery_ok" : 1, "temperature_C" : 22.600, "humidity" : 51, "wind_dir_deg" : 135, "wind_avg_km_h" : 1.224, "wind_max_km_h" : 3.672, "rain_mm" : 0.000, "mic" : "CRC"}\n', '{"time" : "2021-06-15 17:31:15", "model" : "Fineoffset-WHx080", "subtype" : 0, "id" : 168, "battery_ok" : 1, "temperature_C" : 22.600, "humidity" : 51, "wind_dir_deg" : 135, "wind_avg_km_h" : 1.224, "wind_max_km_h" : 3.672, "rain_mm" : 0.000, "mic" : "CRC"}\n']
Jun 15 18:31:18 weathercam weewx[860] DEBUG user.sdr: parse_json: unknown model Fineoffset-WHx080

andylittle, 2021-06-15:
Try this one: https://github.com/andylittle/weewx-sdr/blob/master/bin/user/sdr.py

mfraser, 2021-06-17:
On 15/06/2021 22:24, andylittle wrote:
> Try this one: 
> https://github.com/andylittle/weewx-sdr/blob/master/bin/user/sdr.py 

sudo PYTHONPATH=/usr/share/weewx python3 /usr/share/weewx/user/sdr.py 
--cmd="rtl_433 -R 32 -M utc -F json"

Now gives me:

out:['{"time" : "2021-06-17 14:13:35", "model" : "Fineoffset-WHx080", 
"subtype" : 0, "id" : 168, "battery_ok" : 1, "temperature_C" : 20.100, 
"humidity" : 70, "wind_dir_deg" : 135, "wind_avg_km_h" : 2.448, 
"wind_max_km_h" : 3.672, "rain_mm" : 0.000, "mic" : "CRC"}\n']
parsed: {'dateTime': 1623939215, 'usUnits': 16, 
'msg_type.168.FOWHx080Packet': 0, 'temperature.168.FOWHx080Packet': 
20.1, 'humidity.168.FOWHx080Packet': 70.0, 
'wind_dir.168.FOWHx080Packet': None, 'wind_speed.168.FOWHx080Packet': 
2.448, 'wind_gust.168.FOWHx080Packet': 3.672, 
'rain_total.168.FOWHx080Packet': 0.0, 'battery.168.FOWHx080Packet': 0, 
'signal_type.168.FOWHx080Packet': 0, 'hours.168.FOWHx080Packet': None, 
'minutes.168.FOWHx080Packet': None, 'seconds.168.FOWHx080Packet': None, 
'year.168.FOWHx080Packet': None, 'month.168.FOWHx080Packet': None, 
'day.168.FOWHx080Packet': None}

So I changed the sensor_map lines to:

     [[sensor_map]]
         windGust = wind_gust.168.FOWHx080Packet
         outBatteryStatus = battery.168.FOWHx080Packet
         rain_total = rain_total.168.FOWHx080Packet
         windSpeed = wind_speed.168.FOWHx080Packet
         windDir = wind_dir.168.FOWHx080Packet
         outHumidity = humidity.168.FOWHx080Packet
         outTemp = temperature.168.FOWHx080Packet

/var/log/syslog is showing:
Jun 17 15:48:50 weathercam weewx[9098] DEBUG user.sdr: lines=['{"time" : 
"2021-06-17 14:48:47", "model" : "Fineoffset-WHx080", "subtype" : 0, 
"id" : 168, "battery_ok" : 1, "temperature_C" : 21.600, "humidity" : 65, 
"wind_dir_deg" : 0, "wind_avg_km_h" : 1.224, "wind_max_km_h" : 2.448, 
"rain_mm" : 0.000, "mic" : "CRC"}\n']
Jun 17 15:48:50 weathercam weewx[9098] DEBUG user.sdr: 
packet={'windGust': 2.448, 'outBatteryStatus': 0, 'rain_total': 0.0, 
'windSpeed': 1.224, 'windDir': None, 'outHumidity': 65.0, 'outTemp': 
21.6, 'dateTime': 1623941327, 'usUnits': 16}

Does this mean it is working as /var/lib/weewx/weewx.sdb isn't updating?

mfraser, 2021-06-18:
Working now.

closed 2021-06-18 by mfraser