matthewwall/weewx-sdr #133: Support for new sensors WH31L

MksRasp, 2021-06-30:
can you help me in writing parser for this device (WH31L), it is lightning detector ,below is the output from running driver directly

out:['{"time" : "2021-06-30 20:37:11", "model" : "FineOffset-WH31L", "id" : 67016, "battery_ok" : 0, "state" : 8, "flags" : 56, "storm_dist_km" : 10, "strike_count" : 2, "mic" : "CRC"}\n']

andylittle, 2021-07-01:
try this one [here](https://github.com/andylittle/weewx-sdr/blob/master/bin/user/sdr.py).(untested)  What are state and flags?

MksRasp, 2021-07-01:
> try this one [here](https://github.com/andylittle/weewx-sdr/blob/master/bin/user/sdr.py).(untested) What are state and flags?

thank you for your support
recently got this device and found it is not supported by rtl_433 so I raised issue to get it supported. I wrote all information I got and this device is only for lightning not like Acurite lightning detector which include temperature and humidity.
they figured out "lightning count, distance and battery", and was told there are some additional information within the code could not be figured out.

MksRasp, 2021-07-01:
did test but failed for some reason

Traceback (most recent call last):
  File "/usr/share/weewx/user/sdr.py", line 2940, in <module>
    main()
  File "/usr/share/weewx/user/sdr.py", line 2927, in main
    for p in PacketFactory.create(lines):
  File "/usr/share/weewx/user/sdr.py", line 2637, in create
    pkt = PacketFactory.parse_json(lines)
  File "/usr/share/weewx/user/sdr.py", line 2653, in parse_json
    return parser.parse_json(obj)
  File "/usr/share/weewx/user/sdr.py", line 1502, in parse_json
    return FOWH31LPacket.insert_ids(pkt)
AttributeError: type object 'FOWH31LPacket' has no attribute 'insert_ids'

andylittle, 2021-07-01:
I left out part of the section, grab [this](https://github.com/andylittle/weewx-sdr/commit/b299bb63c2ee006695a7ebac0313a74cc5ec732e#diff-949ec6e148b7747dc03d6e6281d0d19597bfb93ec97cc51cbd5ed2fd303f67eb)

MksRasp, 2021-07-02:
> I left out part of the section, grab [this]

it worked this time, I truly appreciate your help on this issue

closed 2021-07-02 by MksRasp