matthewwall/weewx-sdr #138: Parser for knock-off of Prologue sensor.

stephen-hocking, 2021-09-03:
Hi all,

I bought one of these and attempted to hack this into sdr.py, under the original Prologue class. Here's the diff

2294,2314d2293
< class PrologueTHPacket(Packet):
<     # 2021-09-03 : Prologue-TH Temperature and Humidity Sensor
<     # out:[u'{"time" : "2021-09-02 23:47:40", "model" : "Prologue-TH", "subtype" : 5, "id" : 70, "channel" : 1, "battery_ok" : 1, "temperature_C" : 24.800, "humidity" : 49, "button" : 0}\n']
< 
<     IDENTIFIER = "Prologue-TH"
< 
<     @staticmethod
<     def parse_json(obj):
<         pkt = dict()
<         pkt['dateTime'] = Packet.parse_time(obj.get('time'))
<         pkt['usUnits'] = weewx.METRIC
<         pkt['model'] = obj.get('model')
<         sensor_id = obj.get('id')
<         pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
<         pkt['humidity'] = Packet.get_float(obj, 'humidity')
<         pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
<         pkt['channel'] = obj.get('channel')
<         pkt = Packet.add_identifiers(pkt, sensor_id, PrologueTHPacket.__name__)
<         return pkt
< 
< 

I don't seem to be receiving any output from this - any obvious things I should look at?

stephen-hocking, 2021-09-03:
Never mind - have found the table I needed to add the function name to.

2294,2314d2293
< class PrologueTHPacket(Packet):
<     # 2021-09-03 : Prologue-TH Temperature and Humidity Sensor
<     # out:[u'{"time" : "2021-09-02 23:47:40", "model" : "Prologue-TH", "subtype" : 5, "id" : 70, "channel" : 1, "battery_ok" : 1, "temperature_C" : 24.800, "humidity" : 49, "button" : 0}\n']
< 
<     IDENTIFIER = "Prologue-TH"
< 
<     @staticmethod
<     def parse_json(obj):
<         pkt = dict()
<         pkt['dateTime'] = Packet.parse_time(obj.get('time'))
<         pkt['usUnits'] = weewx.METRIC
<         pkt['model'] = obj.get('model')
<         sensor_id = obj.get('id')
<         pkt['temperature'] = Packet.get_float(obj, 'temperature_C')
<         pkt['humidity'] = Packet.get_float(obj, 'humidity')
<         pkt['battery'] = 0 if obj.get('battery') == 'OK' else 1
<         pkt['channel'] = obj.get('channel')
<         pkt = Packet.add_identifiers(pkt, sensor_id, PrologueTHPacket.__name__)
<         return pkt
< 
< 
2564d2542
<         PrologueTHPacket,

sirrated, 2023-09-23:
I have the same sensor.  rtl_433 crashed when it attempted to parse:

```
out: ['{"time" : "2023-09-23 19:58:57", "model" : "Prologue-TH", "subtype" : 9, "id" : 13, "channel" : 1, "battery_ok" : 1, "temperature_C" : 26.900, "humidity" : 39, "button" : 0}\n']
Traceback (most recent call last):
  File "/usr/share/weewx/user/sdr.py", line 3512, in <module>
    main()
  File "/usr/share/weewx/user/sdr.py", line 3490, in main
    for p in PacketFactory.create(lines):
  File "/usr/share/weewx/user/sdr.py", line 3154, in create
    pkt = PacketFactory.parse_json(lines)
  File "/usr/share/weewx/user/sdr.py", line 3170, in parse_json
    return parser.parse_json(obj)
  File "/usr/share/weewx/user/sdr.py", line 2819, in parse_json
    pkt = Packet.add_identifiers(pkt, sensor_id, PrologueTHPacket.name)
AttributeError: type object 'PrologueTHPacket' has no attribute 'name'
```

I resolved the issue locally by updating this line in sdr.py:

```
Previous:  pkt = Packet.add_identifiers(pkt, sensor_id, PrologueTHPacket.name)
Updated:  pkt = Packet.add_identifiers(pkt, sensor_id, PrologueTHPacket.__name__)
```

sirrated, 2023-09-23:
Issue was addressed in #147.