matthewwall/weewx-sdr #143: Acurite-3n1 - help with parser

zenon2cubed, 2021-12-27:
I tried to follow the examples but struggled.

My current weewx config contains:
`[SDR]`
`    # This section is for the software-defined radio driver.`
`    # The driver to use`
`    driver = user.sdr`
`    cmd = rtl_433 -M utc -F json -R 40 `

The following command:
`sudo rtl_433 -M utc -F json -R 40`

Results in the following output:
`{"time" : "2021-12-27 02:53:38", "model" : "Acurite-3n1", "subtype" : 32, "id" : 7220, "channel" : "B", "sequence_num" : 1, "battery_ok" : 1, "wind_avg_mi_h" : 5.000, "temperature_F" : 5.100, "humidity" : 65, "mic" : "CHECKSUM"}`

Please help with creating a parser for my weather station, thanks!

matthewwall, 2021-12-30:
you posted output for a '32' subtype.  are there other subtypes from the device?

matthewwall, 2021-12-30:
implemented at commit 0ccf340.  please let me know if there are other packet types for the 3n1.

closed 2021-12-30 by matthewwall

zenon2cubed, 2021-12-30:
There are no other subtypes being transmitted from this unit.  Thanks for the update!  I'll advise if I have any further issues.