matthewwall/weewx-sdr #146: Acurite 06045 Lighting detector

RedVox73, 2022-02-20:
I currently am trying to use the Acurite 06045 lightning detector but run into 2 problems. (1 might be weewx related)

- The sensor is being received however every pack is shown 3 times.
- In Weewx the strikes total of every data pack is being added to the total instead of the difference between the data packs. ( the sensor sens out the total, like a rain sensor) (The sensor temp and humidity are mapped to "inTemp and inHumidity and work fine)

[image]

Any advise on where / how to fix these items?

andylittle, 2022-03-03:
https://groups.google.com/g/weewx-user/c/NQIQ6KhG7To?pli=1

andylittle, 2022-03-03:
https://github.com/matthewwall/weewx-sdr/issues/57

closed 2024-12-30 by matthewwall