matthewwall/weewx-sdr #148: Cotech-367959 weather station

csharpengineer, 2022-03-01:
I need help figuring out how to map my weather station so that it can be mapped properly.

Here is the output as shown from sdr.py:
`out: ['{"time" : "2022-03-01 14:11:26", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 47.000, "humidity" : 61, "rain_mm" : 18.600, "wind_dir_deg" : 45, "wind_avg_m_s" : 0.600, "wind_max_m_s" : 1.000, "mic" : "CRC"}\n']`

Occasionally, it will get two packets at a time.  Will it be OK with this?
`out: ['{"time" : "2022-03-01 14:11:42", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 46.900, "humidity" : 62, "rain_mm" : 18.600, "wind_dir_deg" : 16, "wind_avg_m_s" : 0.600, "wind_max_m_s" : 0.700, "mic" : "CRC"}\n', '{"time" : "2022-03-01 14:11:42", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 46.900, "humidity" : 62, "rain_mm" : 18.600, "wind_dir_deg" : 16, "wind_avg_m_s" : 0.600, "wind_max_m_s" : 0.700, "mic" : "CRC"}\n']`

Thanks in advance!

andylittle, 2022-03-03:
Duplicates are fine. 
Is that the only packet it sends?

csharpengineer, 2022-03-03:
Yes, all packets look just like that one.

`pi@raspberrypi:~ $ rtl_433 -d driver=rtlsdr -t digital_agc=true -g TUNER=30 -R 153 -F json -M utc
Registered 1 out of 175 device decoding protocols [ 153 ]
...
Found Rafael Micro R820T tuner
...
Tuned to 433.920MHz.
{"time" : "2022-03-03 15:56:49", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.200, "humidity" : 35, "rain_mm" : 18.600, "wind_dir_deg" : 285, "wind_avg_m_s" : 0.000, "wind_max_m_s" : 0.000, "mic" : "CRC"}
{"time" : "2022-03-03 15:56:49", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.200, "humidity" : 35, "rain_mm" : 18.600, "wind_dir_deg" : 285, "wind_avg_m_s" : 0.000, "wind_max_m_s" : 0.000, "mic" : "CRC"}
{"time" : "2022-03-03 15:57:05", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.400, "humidity" : 36, "rain_mm" : 18.600, "wind_dir_deg" : 78, "wind_avg_m_s" : 0.200, "wind_max_m_s" : 0.300, "mic" : "CRC"}
{"time" : "2022-03-03 15:57:21", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.500, "humidity" : 36, "rain_mm" : 18.600, "wind_dir_deg" : 127, "wind_avg_m_s" : 0.300, "wind_max_m_s" : 0.700, "mic" : "CRC"}
{"time" : "2022-03-03 15:57:21", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.500, "humidity" : 36, "rain_mm" : 18.600, "wind_dir_deg" : 127, "wind_avg_m_s" : 0.300, "wind_max_m_s" : 0.700, "mic" : "CRC"}
{"time" : "2022-03-03 15:57:37", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.600, "humidity" : 36, "rain_mm" : 18.600, "wind_dir_deg" : 225, "wind_avg_m_s" : 0.400, "wind_max_m_s" : 0.700, "mic" : "CRC"}
{"time" : "2022-03-03 15:57:53", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.600, "humidity" : 35, "rain_mm" : 18.600, "wind_dir_deg" : 200, "wind_avg_m_s" : 0.500, "wind_max_m_s" : 0.700, "mic" : "CRC"}
{"time" : "2022-03-03 15:57:53", "model" : "Cotech-367959", "id" : 24, "battery_ok" : 1, "temperature_F" : 71.600, "humidity" : 35, "rain_mm" : 18.600, "wind_dir_deg" : 200, "wind_avg_m_s" : 0.500, "wind_max_m_s" : 0.700, "mic" : "CRC"}`

andylittle, 2022-03-03:
https://github.com/andylittle/weewx-sdr-1/blob/master/bin/user/sdr.py

Replace your sdr.py file with this one, and post back if it works and the units seem correct. 

depending on your install method, the sdr.py file will be in /home/weewx/bin/user or /usr/share/weewx/user

Andy

csharpengineer, 2022-03-03:
Thanks!  If I would have known it was that easy I might have done it 👍 
I now get these parsed packets:
```
parsed: {'dateTime': 1646329409, 'usUnits': 16, 'battery.24.WS2032Packet': 1, 'temperature.24.WS2032Packet': 24.499999999999996, 'humidity.24.WS2032Packet': 27.0, 'wind_gust.24.WS2032Packet': 1.7, 'wind_speed.24.WS2032Packet': 1.4, 'wind_dir.24.WS2032Packet': 117.0, 'total_rain.24.WS2032Packet': 18.6}
```

Sensor map:
```
   [[sensor_map]]  
        battery = battery.24.WS2032Packet
        windDir = wind_dir.24.WS2032Packet
        windSpeed = wind_speed.24.WS2032Packet
        windGust = wind_gust.24.WS2032Packet
        outTemp = temperature.24.WS2032Packet
        outHumidity = humidity.24.WS2032Packet
        rain_total = total_rain.24.WS2032Packet
```

I think we might have it!
[image]

csharpengineer, 2022-03-03:
I probably need some tweaking because I don't have the map 100% correct, but thanks again.

andylittle, 2022-03-03:
ugh I had a typo in there, grab the file from my repo again and the identifier should change to xxxxx.Cotech367959Packet
pkt = Packet.add_identifiers(pkt, sensor_id, Cotech367959Packet.__name__)

whtz, 2022-07-22:
New to all of this, so please bear with me. I know this is an old thread, but it seemed to fit my situation as I was getting the same "Cotech-367959" from rtl_433.  Is it ok to ask here, or should I just start a new topic? Thanks in advance!

 Long story short, I get parsed data when I run it thru sdr.py,

`parsed: {'dateTime': 1658503239, 'usUnits': 16, 'battery.31.Cotech367959Packet': 1, 'temperature.31.Cotech367959Packet': 29.499999999999996, 'humidity.31.Cotech367959Packet': 50.0, 'wind_gust.31.Cotech367959Packet': 1.0, 'wind_speed.31.Cotech367959Packet': 0.8, 'wind_dir.31.Cotech367959Packet': 175.0, 'total_rain.31.Cotech367959Packet': 20.1}`

But when I run weewx, it crashes:

`Original exception was:
Traceback (most recent call last):
  File "/usr/share/weewx/weewxd", line 248, in <module>
    main()
  File "/usr/share/weewx/weewxd", line 153, in main
    engine.run()
  File "/usr/share/weewx/weewx/engine.py", line 210, in run
    self.dispatchEvent(weewx.Event(weewx.NEW_LOOP_PACKET, packet=packet))
  File "/usr/share/weewx/weewx/engine.py", line 245, in dispatchEvent
    callback(event)
  File "/usr/share/weewx/weewx/wxservices.py", line 112, in new_loop_packet
    self.do_calculations(event.packet, self.loop_calc_dict)
  File "/usr/share/weewx/weewx/wxservices.py", line 137, in do_calculations
    new_value = weewx.xtypes.get_scalar(obs_type, data_dict, self.db_manager)
  File "/usr/share/weewx/weewx/xtypes.py", line 77, in get_scalar
    return xtype.get_scalar(obs_type, record, db_manager, **option_dict)
  File "/usr/share/weewx/weewx/wxxtypes.py", line 374, in get_scalar
    return self.altimeter(record)
  File "/usr/share/weewx/weewx/wxxtypes.py", line 423, in altimeter
    altitude = weewx.units.convertStd(self.altitude_vt, record['usUnits'])
  File "/usr/share/weewx/weewx/units.py", line 1444, in convertStd
    return StdUnitConverters[target_std_unit_system].convert(val_t)
  File "/usr/share/weewx/weewx/units.py", line 881, in convert
    new_val_t = convert(val_t, new_unit_type)
  File "/usr/share/weewx/weewx/units.py", line 1409, in convert
    conversion_func = conversionDict[val_t[1]][target_unit]
KeyError: 'feet'
`
`[[sensor_map]]
        outTemp = temperature.31.Cotech367959Packet
        outHumidity = humidity.31.Cotech367959Packet
        windDir = wind_dir.31.Cotech367959Packet
        windSpeed = wind_speed.31.Cotech367959Packet
        windGust = wind_gust.31.Cotech367959Packet
        rain_total = total_rain.31.Cotech367959Packet
        # uv = uv.16.FOWH65BAltPacket # Not displaying in web page
        # light = light.16.FOWH65BAltPacket # Not displaying in web page
        outTempBatteryStatus = battery.31.Cotech367959Packet`

whtz, 2022-07-22:
Nevermind. It was a typo. Thank you for getting me going in the right direction in the first place though!

matthewwall, 2024-12-30:
incorporated by merging andylittle's branch, followed by fixes to the rain_total label

closed 2024-12-30 by matthewwall